liquidctl/liquidctl #596: NZXT Kraken Elite 360

Ch40t1x, 2023-04-23:
### Device type

AIO liquid cooler

### Product page

https://nzxt.com/product/kraken-360

### First-party software

NZXT CAM

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of embedded LEDs

### Physical connection

USB header

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic

brokenmass, 2023-05-18:
kraken elist has the same interface of kraken Z, at least the one for temperature and pump/fan controls

LXXero, 2023-05-19:
I have a kraken 240, non-elite, and i get the following:
`Bus 005 Device 002: ID 1e71:300e NZXT NZXT Kraken Base
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x300e 
  bcdDevice            1.00
  iManufacturer           1 NZXT Inc.
  iProduct                2 NZXT Kraken Base
`

brokenmass, 2023-05-19:
@LXXero amazing thanks. If I may ask, what are you using liquidctl for ? Do you use it to set pump image or to control pump duty ? Do you use it in conjunction with fan control ?

aleksamagicka, 2023-05-19:
Does the non-Elite Kraken behave like the Kraken Z as well? These new IDs aren't in liquidctl right now, only `3008` is for Kraken Z.

Looks like low hanging fruit if they do behave the same (and for the kernel driver as well). Can you please test all the functionality?

Ch40t1x, 2023-05-19:
Hi, how would one add the IDs to liquidctl in order to test them?

brokenmass, 2023-05-19:
> Does the non-Elite Kraken behave like the Kraken Z as well? These new IDs aren't in liquidctl right now, only `3008` is for Kraken Z.
> 
> Looks like low hanging fruit if they do behave the same (and for the kernel driver as well). Can you please test all the functionality?

Yes new kraken have same protocol as kraken v3 with the following differences:
- no led control
- Bulk transfer doesn’t seems to be limited to 512bytes but can be arbitrarily big (on the elite a whole raw image can be transferred in a single call 640*640*4bytes)

PIDS are
- 0x300e new kraken
- 0x300c new kraken elite

I’ve raised pr to librehardwaremonitor to add support and added to my own Fancontrol plug-in ( https://github.com/brokenmass/Fancontrol.NzxtKraken ) and works perfectly. I’ll add the non elite version pod now that @LXXero shared with me

LXXero, 2023-05-19:
> @LXXero amazing thanks. If I may ask, what are you using liquidctl for ? Do you use it to set pump image or to control pump duty ? Do you use it in conjunction with fancontrol ?
> 
> finally, can you let me know what’s the firmware version reported in cam ?

I've not used cam, or well, anything yet, on this PC. It's a brand new build sitting on the floor next to me,  with no OS installed, and I was running memtest86 to ensure it was stable before I installed the OS, and that's when I noticed CPU temps creeping up, leading me here. I didn't realize how new this cooler was when I bought it, go figure! But I'm glad to help get it working any way I can.

I haven't tried to set the image, I'm not sure if the non-elite can do that? I think that's the main difference between elite vs non-elite?? the screen is more basic i think on the non-elite one, but they do both have screens....i honestly didn't even realize the difference at the time of purchase, I just didn't care about having RGB.

Right now, I'm just using it to increase the fan speed, I'm not really sure what I intend to do with it beyond fan/pump speeds. I replaced the stock fans with noctua a12x25s and they needed a slight speed bump at minimum setting, as they're capable of running quite a bit slower than the stock fans. I'm not really sure where I intend to go from here, though.

brokenmass, 2023-05-19:
Difference between elite and non elite is lcd size and resolution. The current code will not work as it’s written for krakenZ that have resolution of 320x320 while new ones are 640x640 (elite) or 240x240 (non elite).
Apart from that and the fact that bulk calls do not have to be splitted in chunks of 512bytes, the protocol is the same.

aleksamagicka, 2023-05-19:
> Apart from that and the fact that bulk calls do not have to be splitted in chunks of 512bytes, the protocol is the same.

Does it still accept the image when it's sent in chunks? The resolution seems easy enough to customize in code. (I worked on the driver, but not on the LCD stuff so I'm pretty much looking at it for the first time - trying to gather info for a PR.)

brokenmass, 2023-05-19:
I think it might still work but sending 1.7MB in chunks is gonna be slow.

aleksamagicka, 2023-05-19:
@brokenmass That works as well! I can take a look at it.

We also have a [kernel driver](https://github.com/liquidctl/liquidtux/blob/master/nzxt-kraken3.c) for the Krakens in the works (it's almost complete and only some synchronization code is left to be merged, not pertaining to this). I'll look into extending it to support these devices as well.

brokenmass, 2023-05-20:
@aleksamagicka  sorry got carried away by another hack (https://www.reddit.com/r/NZXT/comments/13mem4v/using_a_kraken_lcd_as_a_monitor_d_30fps/) and didn't have time to raise a pr

greyowl, 2023-07-25:
Hi, I was wondering if there was a way to control kraken 240, non-elite with the most recent version of `liquitctl` whilst #605 is being worked on

PR #605 (2023-10-12): Add support for nzxt krakens 2023
Adds support for new NZXT Kraken AIOs (2023 standard and elite). Protocol is pretty much identical to krakenZ with following differences:
- max bulk size write buffer is 2MB
- new devices do not have color channels
- new devices have different resolutions than kraken Z ( 640x60 for kraken elite, 240x240 for kraken standard)
- when setting fan speed byte 3 and 4 must be `0x1 0x1` instead of `0x0 0x0`
- when setting pump speed byte 3 and 4 must be `0x1 0x0` instead of `0x0 0x0` (but even `0x0 0x0`) works

Closes: #596 

---

Checklist:

- [x] Adhere to the [deve [... cut]
closed 2023-10-12 by jonasmalacofilho