liquidctl/liquidctl #597: PWM fan registered as DC and not running at full speed (Noctua NF-A20)

el-ef, 2023-04-29:
### Describe the bug

Hi,
I have a strange issue with a Noctua NF-A20 connected to a NZXT Grid+ V3.

The fan is 4-Pin/PWM but registered as DC when running the `status` command. Rebooting and reconnected the controller and fan, switching the ports etc. makes no difference. It is still being detected as 3-Pin only.

Using the `set fan* speed` option I can manipulate the RPM. But the speed is capped to roughly 350 RPM with instead of the specified 800 RPM. A speed value of 100 leads to 12.18V but strangely only 350 RPM when running the status command. Lowering the speed value to say 60 drops the voltage and RPM accordingly: 203 RPM and 7.39V.  When set to 50, the RPM is sometimes read with a value of 12000-16000 RPM where the sensor completely looses track of what is happening.

The fan as well as the NZXT controller are working as intended in Windows. The fan is also able to run at full speed when connected directly to the motherboard or an 12V-adapter. With the NZXT controller it also runs at full speed during boot and during liquidctl initialization. It drops from full speed to less than half of that as soon as the system is booted and the fan control service is starting. Hence, it is not a hardware- but a software-related issue.

Other fans such as the NF-A14 PWM are working correctly and being detected as PWM fans when running status. It is only the NF-A20 PWM that produces suchs issues, no matter what port I am using on the NZXT controller. I do also have a pair of NF-A20 DC which seem to work flawlessly on the same port.

Any idea why this is happening?

I would really appreciate your help!

Best wishes,
el-ef

### Commands executed

liquidctl --match nzxt set fan1 speed 100
liquidctl --match nzxt status

### Output of all relevant commands with `--debug` flag

[DEBUG] (cli) (_log_env_infos): python: 3.11.3 (main, Apr  5 2023, 00:00:00) [GCC 13.0.1 20230401 (Red Hat 13.0.1-0)]
-> System default is set to python 3.7 though. Could this make any difference for this case?

---

[DEBUG] (usb) (probe): SmartDevice identified: NZXT Grid+ V3
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 145f:026d (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5702 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (find_devices): HID device: 04d9:0171 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 04d9:0171 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d9:0171 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 04d9:0171 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d9:0171 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 04d9:0171 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d9:0171 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 04d9:0171
[DEBUG] (usb) (find_devices): USB device: 048d:5702
[DEBUG] (usb) (find_devices): USB device: 145f:026d
[DEBUG] (usb) (find_devices): USB device: 046d:086b
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1e71:1711
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT Grid+ V3
[INFO] (smart_device) (set_fixed_speed): setting fan1 duty to 100%
[DEBUG] (usb) (write): writing report 0x02 with 64 bytes: 4d:00:00:64:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

### Affected device

NZXT Grid+ V3, FW v1.4

### Operating system and version

Fedora 38 Workstation, 6.2.13-300.fc38.x86_64

### Installation method

RPM

### Version of liquidctl

v1.12.1 (Linux-6.2.13-300.fc38.x86_64-x86_64-with-glibc2.37)

jonasmalacofilho, 2023-05-11:
I agree with your later edit, that this might be an issue with the controller.

And, based on how the protocol works and you _not_ running a brand new firmware (a scenario where the protocol could have changed), I think this is very unlikely to be a liquidctl problem.

> **EDIT:** Might be an issue related to the NZTX-controller itself. I just ran FanControl on Windows and it has the same problems: only 350 RPM max. although the fan is set to 100%. Really weird!

I would also check if the fan behaves correctly connected to the motherboard _but also with PWM < 100%._ It might be an issue with the Grid controller, sure, but it could also be an issue with the fan, like a PWM or tachometer problem (that the controller might then not be able to properly identify).

For the same reason, I wouldn't trust tachometer readings alone (with either controller): what if the motherboard reports "full speed", but that's not accurate; or what if the NZXT controller reports just ~300 rpm, but the fan's actually at full speed? At the very least, double check things with your ear (you should be able to distinctly recognize "full speed", even on silent fans like Noctuas).

closed 2023-05-11 by jonasmalacofilho

el-ef, 2023-05-11:
@jonasmalacofilho 
Yes, it is indeed a hardware-related issue with the NZXT controller itself and its firmware which appears to be unable to control and monitor PWM fans correctly. I got a Commander Pro as replacement which works just fine.

Issue closed. :)