liquidctl/liquidctl #598: Lack of Fan Control with Corsair Commander XT

sylly015, 2023-05-06:
Hello,

I have been working with the creator of Argus Monitor to try and resolve this issue, as he uses the same protocol to control the Corsair Commander Core XT devices. Originally, I was able to control the device with Argus Monitor just fine, but then the fans started ramping up intermittently. When this happened, Argus appeared to lose control with the controller. The only way to resolve this issue was to go into iCUE and reinstall the firmware, as when the fans lost control it appeared that the firmware version was corrupted. After some time, I was no longer able to control the fans without iCUE, as Argus monitor was no longer able to take control of the fans. I tried disabling all iCUE processes and services, completely powering off the device, and then restarting it, but I was still not having luck. 

So, next we decided to try using the latest version of liquidctl to control the fans. However, I was still unable to properly control the fans.
I am running the latest version of Windows 11. Liquidctl is able to properly recognize the status of the fans and complete an initialization command. It is not able to change to fan speed, though.
[image]

I also generated a debug output:

> [DEBUG] (cli) (_log_env_infos): script: E:\Downloads\liquidctl.exe
> [DEBUG] (cli) (_log_env_infos): version: 0.0.0-unknown
> [DEBUG] (cli) (_log_env_infos): platform: Windows-10-10.0.22621-SP0
> [DEBUG] (cli) (_log_env_infos): python: 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)]
> [DEBUG] (cli) (_log_env_infos): encoding: cp1252 default, cp1252 preferred, utf8_mode 0
> [DEBUG] (cli) (_log_env_infos): with colorlog: version n/a (colorlog)
> [DEBUG] (cli) (_log_env_infos): with crcmod: version n/a (crcmod)
> [DEBUG] (cli) (_log_env_infos): with docopt: version n/a (docopt)
> [DEBUG] (cli) (_log_env_infos): with hidapi: version n/a (hidapi)
> [DEBUG] (cli) (_log_env_infos): with pyusb: version n/a (pyusb)
> [DEBUG] (cli) (_log_env_infos): with pillow: version n/a (pillow)
> [DEBUG] (cli) (_log_env_infos): with smbus: version n/a (smbus)
> [DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (winusbcdc)
> [DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (libusb-package)
> [WARNING] (cli) (main): python -m liquidctl.cli is deprecated, prefer python -m liquidctl
> [DEBUG] (usb) (find_devices): searching HidapiBus
> [DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0xff00 usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0xff43 usage=0x0602)
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0x0001 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0xff00 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0x000c usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 0b05:19af (usage_page=0xff72 usage=0x00a1)
> [DEBUG] (usb) (probe): AuraLed identified: ASUS Aura LED Controller (experimental)
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0xff00 usage=0x0004)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0x0001 usage=0x0080)
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0xffbc usage=0x0088)
> [DEBUG] (usb) (find_devices): HID device: 046d:c539 (usage_page=0x0001 usage=0x0006)
> [DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0xff43 usage=0x0604)
> [DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0x0001 usage=0x0006)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x000c usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:0aa7 (usage_page=0x000c usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:0aa7 (usage_page=0x000b usage=0x0005)
> [DEBUG] (usb) (find_devices): HID device: 046d:0aa7 (usage_page=0xff99 usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:0aa7 (usage_page=0xff43 usage=0x0202)
> [DEBUG] (usb) (find_devices): HID device: 046d:c232 (usage_page=0x0001 usage=0x0006)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:1bf0 (usage_page=0x0001 usage=0x0006)
> [DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0x0001 usage=0x0006)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0001)
> [DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core XT (experimental)
> [DEBUG] (cli) (main): device: Corsair Commander Core XT (experimental)
> [DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
> [DEBUG] (usb) (write): writing report 0x00 with 96 bytes: 08:01:03:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (write): wrote 385 total bytes, expected 97
> [DEBUG] (usb) (read): read 96 bytes: 00:01:00:10:00:02:01:00:00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
> [DEBUG] (usb) (write): writing report 0x00 with 96 bytes: 08:0d:00:60:6d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (write): wrote 385 total bytes, expected 97
> [DEBUG] (usb) (read): read 96 bytes: 00:0d:00:10:00:02:01:00:00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
> [DEBUG] (usb) (write): writing report 0x00 with 96 bytes: 08:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (write): wrote 385 total bytes, expected 97
> [DEBUG] (usb) (read): read 96 bytes: 00:08:00:03:00:06:00:00:00:00:00:00:42:d8:ef:2c:02:32:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
> [DEBUG] (usb) (write): writing report 0x00 with 96 bytes: 08:05:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (write): wrote 385 total bytes, expected 97
> [DEBUG] (usb) (read): read 96 bytes: 00:05:00:03:00:06:00:00:00:00:00:00:42:d8:ef:2c:02:32:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... 39 lines cut ...]
> [DEBUG] (usb) (read): read 96 bytes: 00:0d:00:04:00:06:64:00:64:00:64:00:64:00:64:00:64:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:0 [... cut ...] 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (write): wrote 385 total bytes, expected 97
> [DEBUG] (usb) (read): read 96 bytes: 00:01:00:04:00:06:64:00:64:00:64:00:64:00:64:00:64:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

ProjectSynchro, 2024-01-18:
I can also confirm on Linux that controlling the Commander Core XT's fans does not appear to work, is there some way I can help debug what is happening?

jluizpjr, 2024-01-18:
I'm using the following script to control my Commander Core XT's fans

#!/bin/bash
# Script used to set the Corsair AIO pump and fans mode.
# Fan speeds are taken from iCUE default presets.

while getopts m: flag
do
        case "${flag}" in
                m) mode=${OPTARG};;
        esac
done

if [[ $mode = "quiet" ]];
then
        liquidctl -m corsair set fans speed 20  --temperature-sensor 0
elif [[ $mode = "balanced" ]];
then
        liquidctl -m corsair set fans speed 50  --temperature-sensor 0
elif [[ $mode = "extreme" ]];
then
        liquidctl -m corsair set fans speed 100  --temperature-sensor 0
else
        echo "usage"
        echo "aio -m <mode> (quiet, balanced, extreme)"
fi

ProjectSynchro, 2024-01-27:
> I'm using the following script to control my Commander Core XT's fans #!/bin/bash # Script used to set the Corsair AIO pump and fans mode. # Fan speeds are taken from iCUE default presets. while getopts m: flag do case "${flag}" in m) mode=${OPTARG};; esac done if [[ $mode = "quiet" ]]; then liquidctl -m corsair set fans speed 20 --temperature-sensor 0 elif [[ $mode = "balanced" ]]; then liquidctl -m corsair set fans speed 50 --temperature-sensor 0 elif [[ $mode = "extreme" ]]; then liquidctl -m corsair set fans speed 100 --temperature-sensor 0 else echo "usage" echo "aio -m <mode> (quiet, balanced, extreme)" fi
> […](#)
> -- Joao L Pereira ***@***.***

That script works well, I had to change it on my end to be able to deal with multiple Core XTs ( I have two). 

I have noticed that using CoolerControl (which by extension uses liquidctl) will cause Core XTs to get into a corrupted state after a while, almost as if sending too many commands borks them, blacklisting the Core XTs in CoolerControl and forcing a static fan speed seems to at least work around the issue. 

One thing I have noticed is that the fans are reported to be working at 0% duty cycle in CoolerControl while other devices like the H150i are reporting the correct percentage.

I'm also fairly certain that this is a dupe of https://github.com/liquidctl/liquidctl/issues/583.

defaultsecurity, 2024-05-18:
I cannot control my fans either. It used work, but doesn't anymore. I'm not sure what changed.

defaultsecurity, 2024-05-22:
Here are my debug logs for the Corsair Commander XT for the following commands, under ubuntu 22.04:
- liquidctl status --debug
- liquidctl -m corsair set fans speed 5 --debug

[commander XT debug.log](https://github.com/liquidctl/liquidctl/files/15406122/commander.XT.debug.log)

jonasmalacofilho, 2024-05-22:
@defaultsecurity , can you please try the latest code in git?

defaultsecurity, 2024-05-22:
@jonasmalacofilho
Yes, of course, how can I install that version?

jonasmalacofilho, 2024-05-22:
You can follow the [Manual Installation](https://github.com/liquidctl/liquidctl?tab=readme-ov-file#manual-installation) section of the README, installing from GitHub using pip in a virtual environment.

Since you already have the ubuntu-supplied liquidctl installed, you should be able to skip the dependencies.

defaultsecurity, 2024-05-22:
@jonasmalacofilho 
Thanks for the help. Originally I have installed liquidctl along with CoolerControl and I didn't set up a virtual environment. Just now managed to update to the git version v1.14.0.dev80+gd343c0d using the command:
`pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl`

The result was not a success. I have restarted the system before running the commands. All the connected PWM fans keep runnning at 100% speed. I have attached my log for the following commands:
```
liquidctl --version
liquidctl initialize all --debug
liquidctl status --debug
liquidctl -m corsair set fans speed 50 --debug
liquidctl -d 1 set fans speed 50 --debug
liquidctl status --debug
```
[commander XT debug 2.log](https://github.com/liquidctl/liquidctl/files/15406784/commander.XT.debug.2.log)

CoolerControl Daemon is not running.

```
USER@SYSTEM:~$ sudo systemctl status coolercontrold
coolercontrold.service - CoolerControl Daemon
     Loaded: loaded (/lib/systemd/system/coolercontrold.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
```

**Can you spot anything in log file?**

jonasmalacofilho, 2024-05-22:
@defaultsecurity , not really, sorry.