liquidctl/liquidctl #601: Corsair Commander ST (Experimental) Could Not Open Error

IAmDarthMole, 2023-05-16:
### Describe the bug

When listing, or trying to get status, or change any settings to Corsair Commander ST I get the following: ERROR: Corsair Commander ST (experimental): could not open, possibly due to insufficient permissions

I'm using an iCUE H150i Elite LCD XT, which isn't on the supported device list (which may be the problem) but the Corsair Commander ST is. 

Nix package states the liquidctl version is 1.11.1, however, using the liquidctl --version command as instructed below shows 0.0.0-unknown.

### Commands executed

```
liquidctl list
Device #0: Corsair Commander Pro
Device #1: Corsair Commander ST (experimental)
```

```
liquidctl --match st status
ERROR: Corsair Commander ST (experimental): could not open, possibly due to insufficient permissions
```

### Output of all relevant commands with `--debug` flag

```
liquidctl --match st status --debug
[DEBUG] (cli) (_log_env_infos): script: /nix/store/q0aax7g74mhq6cjc9xdpcs70bsrfgl2i-python3.10-liquidctl-1.12.1/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 0.0.0-unknown
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.2.6-76060206-generic-x86_64-with-glibc2.37
[DEBUG] (cli) (_log_env_infos): python: 3.10.11 (main, Apr  4 2023, 22:10:32) [GCC 12.2.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 1
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.13.1
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0f08 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff43 usage=0x0202)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x000c)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0x0001 usage=0x003a)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0xff00 usage=0x000c)
[DEBUG] (usb) (find_devices): HID device: 1038:12ad (usage_page=0x0001 usage=0x003a)
[DEBUG] (usb) (find_devices): HID device: 1462:7c37 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1462:7c37 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c10 (usage_page=0x0084 usage=0x0052)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c32 (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander ST (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c32 (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c39 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 044f:b108 (usage_page=0x0001 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 0c45:5104 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0c45:5104 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0c45:5104 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 0c45:5104 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0c45:5104 (usage_page=0xff1c usage=0x0092)
[DEBUG] (usb) (find_devices): HID device: 1532:001f (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1532:001f (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:001f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1532:001f
[DEBUG] (usb) (find_devices): USB device: 0c45:5104
[DEBUG] (usb) (find_devices): USB device: 044f:b108
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c39
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c32
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c10
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1462:7c37
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 05e3:0620
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1038:12ad
[DEBUG] (usb) (find_devices): USB device: 8087:0aa7
[DEBUG] (usb) (find_devices): USB device: 1532:0f08
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 045e:0719
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair Commander ST (experimental)
[INFO] (cli) (log): detailed error: Corsair Commander ST (experimental): could not open, possibly due to insufficient permissions: OSError('open failed')
Traceback (most recent call last):
  File "/nix/store/q0aax7g74mhq6cjc9xdpcs70bsrfgl2i-python3.10-liquidctl-1.12.1/lib/python3.10/site-packages/liquidctl/cli.py", line 461, in main
    with dev.connect(**opts):
  File "/nix/store/q0aax7g74mhq6cjc9xdpcs70bsrfgl2i-python3.10-liquidctl-1.12.1/lib/python3.10/site-packages/liquidctl/driver/usb.py", line 130, in connect
    self.device.open()
  File "/nix/store/q0aax7g74mhq6cjc9xdpcs70bsrfgl2i-python3.10-liquidctl-1.12.1/lib/python3.10/site-packages/liquidctl/driver/usb.py", line 416, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hidraw.pyx", line 154, in hidraw.device.open_path
OSError: open failed
[ERROR] (cli) (log): Corsair Commander ST (experimental): could not open, possibly due to insufficient permissions
```

### Affected device

iCUE H150i Elite LCD XT

### Operating system and version

Pop!_OS 22.04 LTS

### Installation method

Nix Package (Single User)

### Version of liquidctl

liquidctl v0.0.0-unknown (Linux-6.2.6-76060206-generic-x86_64-with-glibc2.37)

9xbt, 2023-05-17:
Try running the comand as root, I had a similar issue and that fixed it for me.
```liquidctl --match st status```

IAmDarthMole, 2023-05-17:
> 

Thank you very much for the suggestion! Sadly it didn't work. Oddly enough when I run as root I get a command not found error.

```
root@pop-os:~# liquidctl --match st status
Command 'liquidctl' not found, but can be installed with:
apt install liquidctl

root@pop-os:~# liquidctl --match pro status
Command 'liquidctl' not found, but can be installed with:
apt install liquidctl

```

But if I run it as my regular account it gives me the details as before.

```
liquidctl --match st status
ERROR: Corsair Commander ST (experimental): could not open, possibly due to insufficient permissions

liquidctl --match pro status
Corsair Commander Pro
├── +12V rail     11.97  V
├── +5V rail       4.90  V
└── +3.3V rail     3.30  V
```

closed 2023-05-17 by IAmDarthMole