liquidctl/liquidctl #617: Permission error

CrO2Cl2, 2023-07-20:
### Describe the bug

I get 
```
PermissionError: [Errno 13] Permission denied: '/sys/bus/i2c/devices'

```
with nearly every liquidctl command, even if I added the rules, and am running it from a root console or with sudo

### Commands executed

I ran `$sudo liquidctl list` `$sudo liquidctl initialize` , all of them returned the same error 

### Output of all relevant commands with `--debug` flag

```~$ sudo liquidctl list --debug
[DEBUG] (cli) (_log_env_infos): script: /snap/liquidctl/36/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0.dev9+gc20fbb0
[DEBUG] (cli) (_log_env_infos): platform: Linux-5.15.0-76-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_env_infos): python: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.13.1
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1.post2
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
Traceback (most recent call last):
  File "/snap/liquidctl/36/bin/liquidctl", line 8, in <module>
    sys.exit(main())
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/cli.py", line 420, in main
    selected = list(find_liquidctl_devices(**opts))
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/driver/__init__.py", line 61, in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/driver/smbus.py", line 60, in find_devices
    for i2c_dev in devices.iterdir():
  File "/usr/lib/python3.10/pathlib.py", line 1017, in iterdir
    for name in self._accessor.listdir(self):
PermissionError: [Errno 13] Permission denied: '/sys/bus/i2c/devices'
```

### Operating system and version

Zorin os based on ubuntu 20.04 LTS

### Installation method

snap

### Version of liquidctl

v1.13.0.dev9+gc20fbb0

jonasmalacofilho, 2023-07-21:
Please follow https://github.com/degville/liquidctl-snap/issues/2 for updates and/or guidelines on this.

closed 2023-07-21 by jonasmalacofilho

ScorpiosCrux, 2023-10-12:
Just wanted to let you guys know that on the snap docs (https://snapcraft.io/liquidctl)
It says:

> To allow this snap access to your device(s), issue the following connection commands:
`sudo snap connect liquidctl:raw-usb
 sudo snap connect liquidctl:hardware-observe
`