liquidctl/liquidctl #618: No langid

Motzumoto, 2023-07-20:
### Describe the bug

I can't initialize anything, i cant do a damn thing.

### Commands executed

```
$ liquidctl initialize all
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
$ liquidctl status
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
```

### Output of all relevant commands with `--debug` flag

```
# liquidctl initialize all --debug
[DEBUG] (cli) (_log_env_infos): script: /home/linuxbrew/.linuxbrew/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.12.1
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.3.0-kali1-amd64-x86_64-with-glibc2.37
[DEBUG] (cli) (_log_env_infos): python: 3.11.4 (main, Jun  6 2023, 22:16:46) [GCC 11.3.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.13.1
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.0.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-20
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-19
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-17
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-15
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-21
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-18
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-16
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-22
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice                                                                                                   
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x0064)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x0065)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x0066)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x00c8)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x00c9)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x00ca)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x00d9)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x00cb)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x00d3)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xffc5 usage=0x00cc)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x000c)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0x0001 usage=0x003a)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0xff00 usage=0x000c)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a51 (usage_page=0x0001 usage=0x003a)
[DEBUG] (usb) (find_devices): HID device: 1e71:3008 (usage_page=0xff00 usage=0x0001)
Traceback (most recent call last):
  File "/home/linuxbrew/.linuxbrew/bin/liquidctl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/liquidctl/cli.py", line 420, in main
    selected = list(find_liquidctl_devices(**opts))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/liquidctl/driver/__init__.py", line 61, in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/liquidctl/driver/usb.py", line 601, in find_devices
    yield from drv.probe(handle, vendor=vendor, product=product, **kwargs)
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/liquidctl/driver/usb.py", line 120, in probe
    dev = cls(handle, desc, **consargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/liquidctl/driver/kraken3.py", line 568, in __init__
    self.bulk_device = next(
                       ^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/liquidctl/driver/kraken3.py", line 572, in <genexpr>
    if handle.serial_number == self.device.serial_number
       ^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/liquidctl/driver/usb.py", line 369, in serial_number
    return self.usbdev.serial_number
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/usb/core.py", line 864, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/linuxbrew/.linuxbrew/Cellar/liquidctl/1.12.1/libexec/lib/python3.11/site-packages/usb/util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
```

### Affected device

Everything

### Operating system and version

Kali Linux

### Installation method

Brew

### Version of liquidctl

liquidctl v1.12.1 (Linux-6.3.0-kali1-amd64-x86_64-with-glibc2.37)

jonasmalacofilho, 2023-07-21:
There (likely) are a few problems here:

1. The liquidctl driver needs to bail out from setting up the bulk device for LCD control if there aren't enough permissions to do (specifically, to read the serial number and pick the corresponding bulk device); we should use a similar mechanism than what we use for ASUS GPUs, allowing `list` (and possibly other commands independent from the bulk device) to continue to work.

2. The Linuxbrew formula probably doesn't install any udev rules since, IIRC, it hasn't been adjusted from the Mac one. _Please report this upstream, to Linuxbrew._

3. You need to fix your permissions. One way to do that is with the udev rules I mentioned in (1). In any case, check the [Allowing access to the devices](https://github.com/liquidctl/liquidctl/#allowing-access-to-the-devices) section from the README. Also be aware that AppArmor and other sandbox/confinement mechanisms may require additional, specific, configuration.

Motzumoto, 2023-07-21:
> There (likely) are a few problems here:
> 
>     1. The liquidctl driver needs to bail out from setting up the bulk device for LCD control if there aren't enough permissions to do (specifically, to read the serial number and pick the corresponding bulk device); we should use a similar mechanism than what we use for ASUS GPUs, allowing `list` (and possibly other commands independent from the bulk device) to continue to work.
> 
>     2. The Linuxbrew formula probably doesn't install any udev rules since, IIRC, it hasn't been adjusted from the Mac one. _Please verify and report this upstream, to Linuxbrew._
> 
>     3. You need to fix your permissions. One way to do that is with the udev rules I mentioned in (2). In any case, check the [Allowing access to the devices](https://github.com/liquidctl/liquidctl/#allowing-access-to-the-devices) section from the README. Also be aware that AppArmor and other sandbox/confinement mechanisms may require additional, specific, configuration.

Hey thanks for the reply. I took what you said and did some udev rules and i managed to get it fixed. I have a question thats out of scope for this thread though. I have a NXZT Kraken z63 aio and i cant figure out how to control my AIO. I am and to set the lcd screen to the liquid temperatures but nothing else.

jonasmalacofilho, 2023-07-22:
Have you looked at <https://github.com/liquidctl/liquidctl/blob/main/docs/kraken-x3-z3-guide.md>?

Motzumoto, 2023-07-23:
> Have you looked at https://github.com/liquidctl/liquidctl/blob/main/docs/kraken-x3-z3-guide.md?

I have not! Thank you for this, much appreciated!

jonasmalacofilho, 2023-07-23:
You're welcome.

---

Reopening the issue because

>  The liquidctl driver needs to bail out from setting up the bulk device for LCD control if there aren't enough permissions to do so [...]

CriticalPoint, 2023-08-11:
> 

'permissions-related issues setting up the bulk device' - this has peaked my interest; could this be a possible cause of discussion #621 ?