liquidctl/liquidctl #619: python 3.12  issue

edpil02, 2023-07-24:
My distro  just  switched to python 3.12  and  liquidctl fails to build because  python 3.12  denied to write null bytes in :

 /tests/test_keyval.py
line 80 :   key_file.write_bytes(b'\x00')

Change the line to  key_file.write_bytes(b'\xff')   solves the issue but i wonder if this is the right way.
Thanks.

jonasmalacofilho, 2023-07-24:
Can you provide the full exception and traceback, as well as the exact Python 3.12 [beta] version?

jonasmalacofilho, 2023-07-24:
> Change the line to key_file.write_bytes(b'\xff') solves the issue but i wonder if this is the right way.

That's not a viable solution to fix the test: its purpose is to make sure of predictable and (if possible) correct behavior in the event that the file contents somehow get overwritten (by liquidctl or another program) with a null byte.

edpil02, 2023-07-24:
~$ python --version
Python 3.12.0b4

Backtrace : 

______________ test_fs_backend_handles_values_corupted_with_nulls ______________

tmpdir = local('/tmp/pytest-of-mockbuild/pytest-0/test_fs_backend_handles_values0')
caplog = <_pytest.logging.LogCaptureFixture object at 0x7fb920192ea0>

    def test_fs_backend_handles_values_corupted_with_nulls(tmpdir, caplog):
        run_dir = tmpdir.mkdir('run_dir')
        store = _FilesystemBackend(key_prefixes=['prefix'], runtime_dirs=[run_dir])
    
        store.store('key', 42)
        key_file = Path(run_dir).joinpath('prefix', 'key')
        assert key_file.read_bytes() == b'42', 'unit test is unsound'
    
        key_file.write_bytes(b'\x00')
>       val = store.load('key')

tests/test_keyval.py:81: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
liquidctl/keyval.py:104: in load
    value = literal_eval(data)
/usr/lib64/python3.12/ast.py:66: in literal_eval
    node_or_string = parse(node_or_string.lstrip(" \t"), mode='eval')
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

source = '\x00', filename = '<unknown>', mode = 'eval'

    def parse(source, filename='<unknown>', mode='exec', *,
              type_comments=False, feature_version=None):
        """
        Parse the source into an AST node.
        Equivalent to compile(source, filename, mode, PyCF_ONLY_AST).
        Pass type_comments=True to get back type comments where the syntax allows.
        """
        flags = PyCF_ONLY_AST
        if type_comments:
            flags |= PyCF_TYPE_COMMENTS
        if feature_version is None:
            feature_version = -1
        elif isinstance(feature_version, tuple):
            major, minor = feature_version  # Should be a 2-tuple.
            if major != 3:
                raise ValueError(f"Unsupported major version: {major}")
            feature_version = minor
        # Else it should be an int giving the minor version for 3.x.
>       return compile(source, filename, mode, flags,
                       _feature_version=feature_version)
E       SyntaxError: source code string cannot contain null bytes

/usr/lib64/python3.12/ast.py:52: SyntaxError
=============================== warnings summary ===============================
tests/test_cli.py::test_json_list
tests/test_cli.py::test_json_initialize
tests/test_cli.py::test_json_status
  /builddir/build/BUILD/liquidctl-1.12.1/liquidctl/cli.py:312: DeprecationWarning: Use setlocale(), getencoding() and getlocale() instead
    locale.getdefaultlocale()[1], locale.getpreferredencoding(), sys.flags.utf8_mode)

tests/test_keyval.py: 10 warnings
  /usr/lib64/python3.12/multiprocessing/popen_fork.py:66: DeprecationWarning: This process (pid=739) is multi-threaded, use of fork() may lead to deadlocks in the child.
    self.pid = os.fork()

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
=========================== short test summary info ============================
FAILED tests/test_keyval.py::test_fs_backend_handles_values_corupted_with_nulls
================== 1 failed, 434 passed, 13 warnings in 3.89s ===============

closed 2023-07-24 by jonasmalacofilho