liquidctl/liquidctl #623: Stranger things with Corsair Commander Core

NateTheGreat001, 2023-08-13:
### Describe the bug

All sorts of misbehaviour including:

1. Failure to increase rmp to target fan speed percentage
2. Failure to decrease rmp to target fan speed percentage
3. Weird error messages on some set fan speed commands but not on others
4. Total failure to respond to any command whatsoever (that can only be helped by booting windows and forcing a firmware update to corsair commander core)

### Commands executed

We see that (1) the fans do not reach full speed when after set fans speed 100 (because fan groups 2 and 6, that have max rpm of 3000 are at 2771  and 2661 respectively). We see (2) errors on setting lower fan speeds. We do see (3) fan rpms getting "stuck" at high load even when there is no error. Finally (4) we see that if we repeat the command the fan speeds do come down.

```
╭─nate@OSPREY in ~ took 276ms
 ╰─λ liquidctl status
Corsair Commander Core (experimental)
├── Pump speed        0  rpm
├── Fan speed 1    2298  rpm
├── Fan speed 2    2219  rpm
├── Fan speed 3       0  rpm
├── Fan speed 4    1789  rpm
├── Fan speed 5    2373  rpm
└── Fan speed 6    2191  rpm

 ╭─nate@OSPREY in ~ took 162ms
 ╰─λ liquidctl set fans speed 100; sleep 60

 ╭─nate@OSPREY in ~ took 4s
[🧱] × liquidctl status
Corsair Commander Core (experimental)
├── Pump speed        0  rpm
├── Fan speed 1    2816  rpm
├── Fan speed 2    2771  rpm
├── Fan speed 3       0  rpm
├── Fan speed 4    2230  rpm
├── Fan speed 5    2921  rpm
└── Fan speed 6    2661  rpm

 ╭─nate@OSPREY in ~ took 146ms
 ╰─λ liquidctl set fans speed 10; sleep 60
ERROR: Corsair Commander Core (experimental): unexpected error: AssertionError('response does not match command')
^[[A^C

 ╭─nate@OSPREY in ~ took 6s
[🧱] × liquidctl set fans speed 10; sleep 60
ERROR: Corsair Commander Core (experimental): unexpected error: AssertionError('response does not match command')
^C

 ╭─nate@OSPREY in ~ took 6s
[🧱] × liquidctl set fans speed 20; sleep 60

 ╭─nate@OSPREY in ~ took 1m
 ╰─λ liquidctl status
Corsair Commander Core (experimental)
├── Pump speed        0  rpm
├── Fan speed 1    2298  rpm
├── Fan speed 2    2219  rpm
├── Fan speed 3       0  rpm
├── Fan speed 4    1784  rpm
├── Fan speed 5    2370  rpm
└── Fan speed 6    2186  rpm

___________________________________________________________
A FEW MINUTES LATER
___________________________________________________________

 ╭─nate@OSPREY in ~ took 149ms
 ╰─λ liquidctl set fans speed 20; sleep 60

 ╭─nate@OSPREY in ~ took 1m
 ╰─λ liquidctl status
Corsair Commander Core (experimental)
├── Pump speed       0  rpm
├── Fan speed 1    728  rpm
├── Fan speed 2    689  rpm
├── Fan speed 3      0  rpm
├── Fan speed 4    237  rpm
├── Fan speed 5    764  rpm
└── Fan speed 6    757  rpm
```

### Output of all relevant commands with `--debug` flag

[liquidctl_debug_commands.txt](https://github.com/liquidctl/liquidctl/files/12330097/liquidctl_debug_commands.txt)

### Affected device

Corsair Commander Core

### Operating system and version

Garuda Linux latest

### Installation method

Pacman

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.4.9-zen1-1-zen-x86_64-with-glibc2.38)

Serphentas, 2023-10-19:
Have you tried with `--direct-access` ? What about `--fan-mode NUM:MODE --direct-access` (ex. `3:pwm`) ?

jonasmalacofilho, 2024-12-16:
Merged with other similar issues into #753.

closed 2024-12-16 by jonasmalacofilho