liquidctl/liquidctl #624: Please add support for Kraken 240

marcomistretta, 2023-08-18:
### Device type

AIO liquid cooler, Kraken 240 (no rgb version)

### Product page

https://nzxt.com/product/kraken-240

### First-party software

NZXT CAM

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of embedded LEDs, lighting of accessories like RGB fans or LED strips

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

aleksamagicka, 2023-11-24:
You didn't list the USB product ID, so I'll guess that it's `0x300E`. Support for that landed in https://github.com/liquidctl/liquidctl/pull/605. Can you please try and see if it works for you?

Silverwolf-3D, 2023-12-14:
Does this help I did a lsusb 
`Bus 001 Device 002: ID 1e71:2012 NZXT NZXT RGB Controller`
`Bus 003 Device 002: ID 1e71:300e NZXT NZXT Kraken Base`
I got one also, but I am a linux newbie so hope this helps.

lanceofwhichwedream, 2024-01-28:
Going to +1 this thread, looks like on windows the Kraken 240 presents itself like this

> USB\VID_1E71&PID_300E&MI_00
Vendor id 1e71
Product id 300e

I'd be super down to test on windows and I can provide any additional information you might need @aleksamagicka

aleksamagicka, 2024-01-28:
Support for that Kraken in liquidctl has been merged in [October](https://github.com/liquidctl/liquidctl/commit/846ac5f5aa9d5484e8ed1d549fd2478bbf51fdd8). Please retry with the [Git version](https://github.com/liquidctl/liquidctl#installing-from-pypi-or-github) for the time being.

lanceofwhichwedream, 2024-01-28:
Will do, thank you for the info. I'll report back if it was successful.

renhiyama, 2024-02-24:
@lanceofwhichwedream can you report back whether you got it working successfully or not? I need to buy a water cooler, and this cooler costs $30 less than kraken x53, so if its successful, I could be getting a better product with more features (lcd) and $30 cheaper 😅

naggie, 2024-03-31:
I can confirm it works on fb697e05c00a9ed8818d88a25d6202f46782d811 (latest master as of yesterday)

...however, after overnight using the python library to control pump and fan speed I get a "device has no longid" error

[image]

Same with liquidctl status: 
[image]

naggie, 2024-04-01:
It crashed again overnight, more than 12 hours after re-plugging it.

There's something suspicious in teh 
[image]

The LCD on the Kraken is also stuck.

aleksamagicka, 2024-04-02:
> Note also in windows I had to replug the device to get CAM to recognise it. Perhaps I have a dud?

From what you're describing, it seems plausible to me (but I've never had this happen to concur). Maybe check that the cables are snug? Perhaps the connector isn't soldered correctly or is otherwise damaged, and the firmware disconnects after failed packets.

I'm guessing that other devices on internal USB work fine?

naggie, 2024-04-03:
I actually spoke to soon -- I forgot the sudo (or udev rules) when I checked it before going to sleep :facepalm: 

It's been running for 15h after the firmware update without a problem. I'll report back tomorrow, if it's still working then the firmware update may have fixed it. 

Otherwise yes indeed the internal USB is fine with other devices and the cables seem ok. Thanks for the reply

naggie, 2024-04-04:
OK it's been nearly 24h, I can confidently say the firmware update fixed the problem. I've made a PR to update the readme.

robzombie91, 2024-04-18:
just to add to this, my kraken 240 doesnt display anything when doing liquid list, and the pip version shows this error ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
lsusb does show it as Bus 001 Device 006: ID 1e71:300e NZXT NZXT Kraken Base

naggie, 2024-04-18:
Have you tried updating the firmware using NZXT Cam? Also, I presume you're using udev rules or sudo.

robzombie91, 2024-04-18:
Yeah I figured out that sudoing would let me run the script successfully

closed 2024-12-16 by jonasmalacofilho