liquidctl/liquidctl #626: Thermaltake TOUGHLIQUID Ultra 360

KenjiTakahashi-Rial, 2023-08-24:
### Device type

AIO liquid cooler

### Product page

https://www.thermaltake.com/toughliquid-ultra-360-all-in-one-liquid-cooler.html

### First-party software

TT RGB PLUS 2.0

### What monitoring functionality does the device support?

temperatures

### What configuration functionality does the device support?

lighting of accessories like RGB fans or LED strips

### Physical connection

USB header

### Connection protocol

HID/USB

### Additional information about the device

This is the non-RGB version, so basically the only functionality is setting the LCD screen
```
> lsusb -v
Bus 001 Device 013: ID 264a:2328 Thermaltake AX
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x264a Thermaltake
  idProduct          0x2328 
  bcdDevice            2.00
  iManufacturer           1 Thermaltake
  iProduct                2 AX
  iSerial                 3 A247315PD002084
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              288mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     279
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic, analyzing traffic data, implementing the changes

KenjiTakahashi-Rial, 2023-08-29:
I am new to USB sniffing, so please let me know if this is not the correct traffic
Here are two pcapngs, one with all `usbhid.data` packets and one with all non-`USB_INTERRUPT` packets
[pcapngs.tar.gz](https://github.com/liquidctl/liquidctl/files/12466053/pcapngs.tar.gz)

aleksamagicka, 2023-08-29:
Thanks, what was reported in the official software?

daloki83, 2023-09-20:
+1
Same data for "TOUGHLIQUID Ultra 420 All-In-One Liquid Cooler"

`Bus 001 Device 012: ID 264a:2328 Thermaltake AX
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x264a Thermaltake
  idProduct          0x2328 
  bcdDevice            2.00
  iManufacturer           1 Thermaltake
  iProduct                2 AX
  iSerial                 3 A247315SL000693
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              288mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     279
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
`

Please let me know how i can support

dave-ct, 2024-08-26:
@KenjiTakahashi-Rial -  Did you manage to get this working for the TOUGHLIQUID Ultra 360 All-In-One Liquid Cooler?

KenjiTakahashi-Rial, 2024-08-27:
No, I ended up switching back to Windows because one of my favorite games performed very badly on Linux (and I found out about [WSL](https://learn.microsoft.com/en-us/windows/wsl/about)). So I didn't get very far with this 😅

klatooine, 2025-08-04:
Anyone working on this? i've been capturing some USB traffic this weekend, I've identified the image sent, but I don't know how to reverse engineer the headers and the ack packets

So far what i've seen is that animations are streamed to lcd as jpg frames in packets of 1024 bytes (1020 + command and counter) and a header on top of that (don't remember exactly, like 20 or 30 bytes)

Each jpg occupies N packets, for example the default rocketman is 22 packet per JPG frame.

Any guidance on how to continue? I am afraid that the ACKs and the headers might include a checksum or a counter that I can't decode. Also, can it be a case of PTP usb standard?

Finally, do you think that maybe the same 2.1inch screens that are on aliexpress are the same as this one?

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.