liquidctl/liquidctl #627: Support for NZXT Kraken Elite RGB 360 - RL-KR36E-W1

CharlesCage, 2023-08-25:
### Device type

AIO Liquid Cooler

### Product page

https://nzxt.com/product/kraken-360-elite-rgb

### First-party software

NZXT CAM

### What monitoring functionality does the device support?

fan/pump speeds or duty cycles

### Physical connection

USB Header

### Additional information about the device

`
Bus 001 Device 020: ID 1e71:300c NZXT NZXT Kraken Elite
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x300c 
  bcdDevice            1.02
  iManufacturer           1 NZXT, Inc.
  iProduct                2 NZXT Kraken Elite
  iSerial                 3 843892820EA01920
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 In Application
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     360
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
Device Status:     0x0001
  Self Powered

`

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

aleksamagicka, 2023-08-25:
Related: #596 (for non-RGB), #605

Rubsad, 2023-10-23:
Hello, I have the same problem with Kraken Elite but black (RL-KR36E-B1) and I'm willing to help debug to get it working properly.

I am using Arch Linux

> liquidctl status
ASUS Aura LED Controller
├── ARGB channels: 3
└──  RGB channels: 1

aleksamagicka, 2023-11-24:
Initial support for the listed ID (1e71:300c) landed in https://github.com/liquidctl/liquidctl/pull/605. Can you please try and see if it works for you?

danielfsbarreto, 2024-02-17:
I can confirm it's not recognizing my AIO.

> ➜  ~ liquidctl status   
ASUS Aura LED Controller
├── ARGB channels: 3      
└──  RGB channels: 1

aleksamagicka, 2024-02-17:
Support for that Kraken in liquidctl has been merged in [October](https://github.com/liquidctl/liquidctl/commit/846ac5f5aa9d5484e8ed1d549fd2478bbf51fdd8). Please retry with the [Git version](https://github.com/liquidctl/liquidctl#installing-from-pypi-or-github) for the time being and follow the instructions for setting up permissions in the README.

santi224m, 2024-07-22:
> Support for that Kraken in liquidctl has been merged in [October](https://github.com/liquidctl/liquidctl/commit/846ac5f5aa9d5484e8ed1d549fd2478bbf51fdd8). Please retry with the [Git [...]

Using the Git version worked for me, thanks! 

Setting GIFs and static images on the screen both work.

```bash
$ lsusb | grep Kraken
Bus 001 Device 004: ID 1e71:300c NZXT NZXT Kraken Elite
```

```bash
$ liquidctl status
ASUS Aura LED Controller
├── ARGB channels: 2      
└──  RGB channels: 1      

NZXT Kraken 2023 Elite (broken)
├── Liquid temperature    33.5  °C
├── Pump speed            1357  rpm
├── Pump duty               30  %
├── Fan speed                0  rpm
└── Fan duty                30  %
```

jonasmalacofilho, 2024-07-22:
Thanks for the feedback, @santi224m.

closed 2024-07-22 by jonasmalacofilho

jmginer, 2024-12-15:
Is not working for me :(

`$ lsusb | grep Kraken
Bus 003 Device 007: ID 1e71:300c NZXT NZXT Kraken Elite

$ liquidctl --version
liquidctl v1.13.0 (Linux-6.8.0-50-generic-x86_64-with-glibc2.39)

$ liquidctl status
ERROR: no device matches available drivers and selection criteria
`

aleksamagicka, 2024-12-15:
Please [install the Git version](https://github.com/liquidctl/liquidctl/issues/627#issuecomment-1950274961).

jmginer, 2024-12-15:
> Please [install the Git version](https://github.com/liquidctl/liquidctl/issues/627#issuecomment-1950274961).

This is what I have do:
pipx install git+https://github.com/liquidctl/liquidctl.git@v1.13.0

aleksamagicka, 2024-12-15:
This should be the command you need:

```
python -m pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl
```

You need the latest snapshot, not `v1.13.0`.

jmginer, 2024-12-15:
Now another error related langid, but same...

`xxx:~$ python3 -m venv venv
source venv/bin/activate
(venv) xxx:~$ python3 -m pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl
Collecting liquidctl
  Cloning https://github.com/liquidctl/liquidctl to /tmp/pip-install-jubuo0aa/liquidctl_6a813a61f497447ca039e0e0b42dc4f0
  Running command git clone --filter=blob:none --quiet https://github.com/liquidctl/liquidctl /tmp/pip-install-jubuo0aa/liquidctl_6a813a61f497447ca039e0e0b42dc4f0
  Resolved https://github.com/liquidctl/liquidctl to commit 702bf32f2b9e4ac57f509714d70185852359b602
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting colorlog (from liquidctl)
  Using cached colorlog-6.9.0-py3-none-any.whl.metadata (10 kB)
Collecting crcmod==1.7 (from liquidctl)
  Using cached crcmod-1.7-cp312-cp312-linux_x86_64.whl
Collecting docopt (from liquidctl)
  Using cached docopt-0.6.2-py2.py3-none-any.whl
Collecting hidapi (from liquidctl)
  Using cached hidapi-0.14.0.post4-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.6 kB)
Collecting pyusb (from liquidctl)
  Using cached pyusb-1.2.1-py3-none-any.whl.metadata (2.2 kB)
Collecting pillow (from liquidctl)
  Using cached pillow-11.0.0-cp312-cp312-manylinux_2_28_x86_64.whl.metadata (9.1 kB)
Collecting smbus (from liquidctl)
  Using cached smbus-1.1.post2-cp312-cp312-linux_x86_64.whl
Collecting setuptools>=19.0 (from hidapi->liquidctl)
  Using cached setuptools-75.6.0-py3-none-any.whl.metadata (6.7 kB)
Using cached colorlog-6.9.0-py3-none-any.whl (11 kB)
Using cached hidapi-0.14.0.post4-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (1.1 MB)
Using cached pillow-11.0.0-cp312-cp312-manylinux_2_28_x86_64.whl (4.4 MB)
Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Using cached setuptools-75.6.0-py3-none-any.whl (1.2 MB)
Building wheels for collected packages: liquidctl
  Building wheel for liquidctl (pyproject.toml) ... done
  Created wheel for liquidctl: filename=liquidctl-1.14.0.dev92+g702bf32-py3-none-any.whl size=140440 sha256=cf5fe174e1fc8979427f62ffcbca0c54f966f135f54e672e514031353ca1b49b
  Stored in directory: /tmp/pip-ephem-wheel-cache-e14gebe7/wheels/c3/fb/ef/8ec0332051b2ae6b8fbbb0a8176d7f24c0b7e87f5ae4200d7a
Successfully built liquidctl
Installing collected packages: smbus, docopt, crcmod, setuptools, pyusb, pillow, colorlog, hidapi, liquidctl
Successfully installed colorlog-6.9.0 crcmod-1.7 docopt-0.6.2 hidapi-0.14.0.post4 liquidctl-1.14.0.dev92+g702bf32 pillow-11.0.0 pyusb-1.2.1 setuptools-75.6.0 smbus-1.1.post2

(venv) xxx:~$ liquidctl --version
liquidctl v1.14.0.dev92+g702bf32 (Linux-6.8.0-50-generic-x86_64-with-glibc2.39)

(venv) xxx:~$ liquidctl status
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)

(venv) xxx:~$ sudo liquidctl status
[sudo] password for xxx: 
ERROR: no device matches available drivers and selection criteria`

`xxx:~$ python3 -m pipx install git+https://github.com/liquidctl/liquidctl#egg=liquidctl
⚠  Note: liquidctl was already on your PATH at /usr/bin/liquidctl
  installed package liquidctl 1.14.0.dev92+g702bf32, installed using Python 3.12.3
  These apps are now globally available
    - liquidctl
done! ✨ 🌟 ✨
xxx:~$ liquidctl status
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
xxx:~$ 
xxx:~$ liquidctl list
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
xxx:~$ sudo liquidctl list
xxx:~$ sudo liquidctl status
ERROR: no device matches available drivers and selection criteria
xxx:~$ `

aleksamagicka, 2024-12-15:
Have you set up permissions as per the section below in the instructions linked above?

Permissions are covered in the [Allowing access to devices](https://github.com/liquidctl/liquidctl#allowing-access-to-the-devices) section of the README, after the Git install step. A package manager would set that up for you, but here you'd have to do it yourself. Additional instructions are available in the linked udev rules file, without it you'll indeed need root.