liquidctl/liquidctl #629: Support for Lian-Li Uni Fan SL-INFINITY fan hub

CUexter, 2023-09-14:
### Device type

Fan hub/controller

### Product page

https://lian-li.com/product/uni-fan-sl-infinity/

### First-party software

L-Connect 3

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of accessories like RGB fans or LED strips

### Connection protocol

USB

### Additional information about the device

[It seems like this can be used to help ?](https://github.com/EightB1ts/uni-sync)

LuNeder, 2023-10-07:
Not sure if I should open a new issue for this, but please do the V2 too! (the linked program seems to also support it so I assume the method should be quite similar?)

ThimbleFire, 2023-10-14:
This would be great. The only software I'm familiar with that can run these operates only on Windows, and is Rem0o's FanControl. My fans will likely be stuck at 650 RPM until this gets added since they can't be controlled by the bios.

LuNeder, 2023-10-14:
> This would be great. The only software I'm familiar with that can run these operates only on Windows, and is Rem0o's FanControl. My fans will likely be stuck at 650 RPM until this gets added since they can't be controlled by the bios.

@ThimbleFire the [software op mentioned in their issue](https://github.com/EightB1ts/uni-sync) seem to have worked for me to set my fans to 100%, had to try it after my cpu was overheating when no software was controlling the fans and it seems to have done the job bc that didn’t happen anymore (and it’s a little bit more noisy too)

Tho yeah liquidctl support would be really really great since then we’d have better control and the ability to use stuff like coolero

Serphentas, 2023-11-08:
The utility is licensed with MIT so it should not pose any problem with respect to code usage. I will work on an initial draft but I need someone who can test it, any volunteers ?

LuNeder, 2023-11-08:
> The utility is licensed with MIT so it should not pose any problem with respect to code usage. I will work on an initial draft but I need someone who can test it, any volunteers ?

I could definitely test it!

Serphentas, 2023-11-13:
OK I've got the initial port from Rust done, see https://github.com/Serphentas/liquidctl/commit/3d018aba76e8af81616564b5ae52d276a0a7b791. I have to check a few things with Jonas and then I will ask you to kindly test it. Thank you for your patience.

Serphentas, 2023-11-20:
I think the code is ready to be tested now, you may pull [this branch](https://github.com/Serphentas/liquidctl/tree/feat-lianli-uni). Let me know how things go.

LuNeder, 2023-11-24:
> I think the code is ready to be tested now, you may pull [this branch](https://github.com/Serphentas/liquidctl/tree/feat-lianli-uni). Let me know how things go.

I'll try it this weekend and let you know, thanks!

LuNeder, 2023-11-25:
Hey, got to test it!

The device seems to be listed correctly
````
python3 -m liquidctl list                     
Device #0: LianLi-UNI SL
````
but it errors when trying to get the status:
````
sudo python3 -m liquidctl status        
ERROR: LianLi-UNI SL: unexpected error: NotImplementedError()
````

Serphentas, 2023-11-25:
Thanks for your feedback. Forgot to mention only the `set fanX speed N` function is supported right now. The source repo I used did not showcase any kind of status probing, so that was left out.

Can you then test `liquidctl set fan1 speed 100` ? You may add `--sync-rgb` to sync RGB (or not, leave it out) and `--fan-mode {rpm,pwm}` where `pwm` will not use the provided speed percentage.

LuNeder, 2023-11-26:
Oh, oke!!

Now I get:
````
sudo python3 -m liquidctl set fan1 speed 100
ERROR: LianLi-UNI SL: unexpected error: ValueError('fan channel must be between 1 and 6')
````

I tried other stuff instead of "fan1" (such as numbers between 1 and 6), but the same error happens

Serphentas, 2023-11-29:
Yeah I've made a change to that part, I had misunderstood the expected value for channel. Would you mind pulling and trying again ?

LuNeder, 2023-12-01:
New error now:

````
sudo python3 -m liquidctl set fan1 speed 100
ERROR: LianLi-UNI SL: unexpected error: KeyError('fan_mode')
````

CUexter, 2024-06-29:
any further update ? for this issue ?

Luminatrix-eng, 2025-02-14:
Hi, my apologies for reviving a closed conversation. I was wondering if anyone here can help me with an issue. I am hoping to control my Lian Li uni fans through CoolerControl which depends on liquidctl. Currently, it looks like the liquidctl hardware support list doesn't show Lian Li as a supported device but I saw this thread with support for Lian-Li Uni. And typing 'liquidctl list' in the terminal doesn't show the Lian Li fans. 

How do I add the code mentioned above? And is there any further update for this issue? I'm a bit of a noob when it comes to this.

closed 2025-11-15 by jonasmalacofilho