liquidctl/liquidctl #631: Firmware 2.0.1 breaks Kraken 2023 support

jonasmalacofilho, 2023-10-12:
### Describe the bug

> The Kraken 2023 (non elite, non rgb) has had a firmware update (2.0.1 as reported by CAM) .   (I think the previous version was 1.2 if I remember correctly.)  CAM automatically upgraded itself without asking, so there was no choice.
> 
> Since then, setting an image using liquidctl on linux has given this error: 
> [ERROR] (cli) (log): NZXT Kraken 2023: unexpected error: AssertionError('missing messages (attempts=12, missing=1)')
> 
> All other functions of liquidctl seem to be unaffected.

_Originally posted by @sossw in https://github.com/liquidctl/liquidctl/issues/605#issuecomment-1745536722_

### Commands executed

N/A

### Output of all relevant commands with `--debug` flag

N/A

### Affected device

Kraken 2023 (non elite, non rgb)

### Operating system and version

N/A

### Installation method

N/A

### Version of liquidctl

one of the final states of PR #605

jensmeindertsma, 2023-10-13:
i have arch installed on a system with this cooler but haven't been able to install this PR😬 not familiar enough with AUR anyone got me some instructions?

aniongithub, 2023-10-14:
I have this cooler with the 2.0.1 firmware (running on Ubuntu 22.04) and have encountered this bug.

I can run/debug liquidctl, see my initial comments about this issue [here](https://github.com/liquidctl/liquidctl/pull/605#issuecomment-1739802897) and [here](https://github.com/brokenmass/AIOLCDUnchained/issues/4#issue-1920339497)

Happy to help however I can!

jklint, 2023-10-24:
[debug.log](https://github.com/liquidctl/liquidctl/files/13110863/debug.log)

Hitting this as well with 2.0.0 firmware on Manjaro. Attached debug log for reference.

Serphentas, 2023-11-08:
@jklint what command were you trying to run during the capture ? If possible, include the image you were trying to upload.

Any other dumps welcome as well !

sophipl, 2023-11-15:
Hi

I have firmware version 2.0.0 and have the same problem
I tried gifs, static images (png and jpeg, stored from gimp), multiple images, multiple gifs, nothing works

```
python3 -m liquidctl --match kraken initialize
NZXT Kraken 2023 (experimental)
├── Firmware version    2.0.0  
├── LCD Brightness         40  %
└── LCD Orientation         0  °
```

`python3  -m liquidctl -g --match kraken set lcd screen static /home/sophi/dragon.png`

[image]

```
[DEBUG] (cli) (_log_env_infos): script: /home/sophi/src/liquidctl/liquidctl/__main__.py
[DEBUG] (cli) (_log_env_infos): version: 1.14.0.dev6+gd71027d
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.5.0-4-amd64-x86_64-with-glibc2.37
[DEBUG] (cli) (_log_env_infos): python: 3.11.6 (main, Oct  8 2023, 05:06:43) [GCC 13.2.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): not installed, package metadata not available
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1e71:300e (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (open): selected interface: 0
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken 2023 (experimental)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b05:19af (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 2357:0107
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 0b05:19af
[DEBUG] (usb) (find_devices): USB device: 046d:c548
[DEBUG] (usb) (find_devices): USB device: 1e71:300e
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT Kraken 2023 (experimental)
[DEBUG] (usb) (write): writing report 0x30 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... 56 lines cut ...]
[DEBUG] (usb) (read): read 64 bytes: 75:02:fa:7e:03:02:d8:b4:e3:54:a2:45:ad:7f:01:22:00:46:05:1e:1e:01:00:00:00:3c:3c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 75:02:fa:7e:03:02:d8:b4:e3:54:a2:45:ad:7f:01:22:00:46:05:1e:1e:01:00:00:00:3c:3c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 75:02:fa:7e:03:02:d8:b4:e3:54:a2:45:ad:7f:01:22:00:64:05:1e:1e:01:00:00:00:3c:3c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[INFO] (cli) (log): detailed error: NZXT Kraken 2023 (experimental): unexpected error: AssertionError('missing messages (attempts=12, missing=1)')
Traceback (most recent call last):
  File "/home/sophi/src/liquidctl/liquidctl/cli.py", line 479, in main
    _device_set_screen(dev, args, **opts)
  File "/home/sophi/src/liquidctl/liquidctl/cli.py", line 287, in _device_set_screen
    dev.set_screen(args["<channel>"], args["<mode>"], args["<value>"], **opts)
  File "/home/sophi/src/liquidctl/liquidctl/driver/kraken3.py", line 789, in set_screen
    self._send_data(data, [0x02, 0x0, 0x0, 0x0] + list(len(data).to_bytes(4, "little")))
[... cut ...]
    return self._read_until_first_match({b"\x33\x02": parse_delete_result})
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sophi/src/liquidctl/liquidctl/driver/kraken3.py", line 732, in _read_until_first_match
    assert False, f"missing messages (attempts={_MAX_READ_ATTEMPTS}, missing={len(parsers)})"
AssertionError: missing messages (attempts=12, missing=1)
[ERROR] (cli) (log): NZXT Kraken 2023 (experimental): unexpected error: AssertionError('missing messages (attempts=12, missing=1)')
```

sophipl, 2023-11-15:
I used remote virtual windows machine (libvirt) with usb pass-through (and GPU pass-through because NZXT LCD needs GPU) and uploaded the same image with USB packet sniffing on Windows, it worked successfully.

Hope it will help to solve the problem
[nzxt.pcapng.gz](https://github.com/liquidctl/liquidctl/files/13370722/nzxt.pcapng.gz)

sophipl, 2023-11-18:
I managed to figure some parts about the data
Apparently they are sending 16 bit color data, that takes me back :P 
I don't know what was about the bucket thing, but I didn't notice that in the wireshark.

I'm including a rough patch that works for me and a image that I created for testing it
[image]
[liquidctl.patch.gz](https://github.com/liquidctl/liquidctl/files/13398569/liquidctl.patch.gz)

aniongithub, 2023-11-26:
@sophipl what branch is this patch for? I'm on d71027d and can't merge your patch.

sophipl, 2023-11-26:
Yes, it happens even when you use Windows software after setting in to Liquid (I can switch USB device to Windows)

sophipl, 2023-11-26:
My suspicion is that when we force "liquid" state, we are making firmware of the device to do the screen updates.
But the manufacturer software never uses this mode, screen update is software driven.
So when we force "legacy"/liquid mode, it fights with software display.
Therefore if you are intend to send the image, never set the liquid mode.
Perhaps there is a way to disable liquid mode, but obviously we don't have protocol description and since the original software doesn't disable liquid mode, I don't know how

jklint, 2023-11-26:
> I managed to figure some parts about the data Apparently they are sending 16 bit color data, that takes me back :P I don't know what was about the bucket thing, but I didn't notice that in the wireshark.
> 
> I'm including a rough patch that works for me and a image that I created for testing it ![grad](https://private-user-images.githubusercontent.com/127139172/283963137-163104f9-8bcf-45de-8351-cc57fcce03e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwMDM5MzEsIm5iZiI6MTcwMTAwMzYzMSwicGF0aCI6Ii8xMjcxMzkxNzIvMjgzOTYzMTM3LTE2MzEwNGY5LThiY2YtNDVkZS04MzUxLWNjNTdmY2NlMDNlMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTEyNlQxMzAwMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMjg1OTU0YzE2NDU1N2U3N2Y2ZWZiNTRlYzRkOWQ0YmUzNDE5MzJhZTZhYzJkZTExYmE5YTZkYWU3ZTI3ZGFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9._hB-1WSbL98LVAd_KRB8FugZW46o6zeuZSeg5sd2K6I) [liquidctl.patch.gz](https://github.com/liquidctl/liquidctl/files/13398569/liquidctl.patch.gz)

Tested this patch on my system and it appears to allow for me to set the image that was previously failing. I haven't set liquid mode since the last reboot, and image remains static with no issue.

killertofu86, 2023-11-29:
> Tested this patch on my system and it appears to allow for me to set the image that was previously failing. I haven't set liquid mode since the last reboot, and image remains static with no issue.

Can confirm. For me the patch allowed me to set a persistent static image.

Nosnahc, 2023-11-30:
I need it back to coolercontrol so here too. Thanks for your work !

maxreitner, 2023-11-30:
so theres no way to display a gif if you upgraded the device to 2.0.0 or higher atm?

sophipl, 2023-11-30:
I wouldn't know, when I put gif in the manufacturer software, it seems to stream the frames.
But I didn't analyze it too far.
Reverse engineering is not my forte :)

maxreitner, 2023-12-01:
so, all we need is a script that splits our gif to its frames in a compatible format and applies the images to the display using liquidctl and your patch

jklint, 2023-12-02:
> so, all we need is a script that splits our gif to its frames in a compatible format and applies the images to the display using liquidctl and your patch

I took a stab at it recently. Its been working on my system. Has the ability to stream a single default gif, with optional gifs being used at threshold temps. For the time being I just leave the script running in a terminal. I will probably add it as a service eventually.

https://github.com/jklint/liquidmotion

endepointe, 2024-03-29:
Question regarding the bug: is this bug only seen when sending gifs to the display? Does this bug not affect the operation of the cooler (kraken 360 standard) ?

aleksamagicka, 2024-03-29:
That's right. In fact, since #673, liquidctl will stop you if you try to set an image with an unsupported firmware.

maxawake, 2024-04-05:
On Firmware 2.0.0 and with the current main branch of liquidctl, i am able to display GIFs on my Kraken 2023 Elite 360. For this i have to just circumvent the firmware check... This is because 1) `self.fw` does not exist when i try to set the screen from command line and 2) I want to see if it works anyways, and it does!

Did something change on the main branch which would explain this? To be honest, i never really encountered any issues, before or after firmware upgrade. I have seen a similar error about the missing message but only if i did some unnecessary reads.

aleksamagicka, 2024-04-05:
> This is because 1) `self.fw` does not exist when i try to set the screen from command line

That seems to be my mistake in #673...

All affected reports in this issue and elsewhere seem related to non-Elite Krakens. There is a difference between them as explained [here](https://github.com/liquidctl/liquidctl/pull/605#issuecomment-1738871076).

So there are two things to fix: `self.fw` and to restrict that to non-Elite Krakens AFAICS.

maxawake, 2024-04-05:
@aleksamagicka Oh okay, that makes a lot of sense. If i change the firmware check from both Elite (`0x300C`) and non-Elite (`0x300E`) to only non-Elite, i can change the gif from the command line again (i.e. the `self.fw` problem is gone). The function then looks something like this

```python
def unsupported_fw_version():
    device_product_id = self.bulk_device.product_id
    if device_product_id == 0x300E and self.fw[0] == 2:
        _LOGGER.error(
            "Setting images for NZXT Kraken 240/360 (non-Elite) is not supported on firmware 2.X.Y, please see issue #631"
        )
        return True
    return False
```

If this problem only affects non-Elite Krakens, i would suggest to make this change on the main branch, because in cases like mine, its just annoying that i can't control my Kraken even though i could.

Its really a small change but i plan to finally do the PR relating to https://github.com/liquidctl/liquidctl/issues/606 , so in principle i could add the change regarding the firmware check. Hence, it would not be necessary to open a new PR only for this. @jonasmalacofilho What do you think about this?

jonasmalacofilho, 2024-04-05:
> Its really a small change but i plan to finally do the PR relating to #606 , so in principle i could add the change regarding the firmware check. Hence, it would not be necessary to open a new PR only for this. @jonasmalacofilho What do you think about this?

It's better to keep the two separate, as one is a fix for something on the main branch, and the other is support for new device variants (and it's not even of the same device, even though, I know, we're still talking about the same driver).

maxawake, 2024-04-05:
> It's better to keep the two separate

Fair enough! So, can you do the fix on the main branch or should i make a PR for this?

aleksamagicka, 2024-04-05:
I'm working on a PR, will open soon but I want to test first.

sophipl, 2024-04-05:
I'm confused, so there are no plan to handle new version of interface?

aleksamagicka, 2024-04-05:
There aren't any from my side, if that's what you're asking, as I only have the older Kraken Z53. If someone opens a PR, that would of course be appreciated.

jonasmalacofilho, 2024-04-05:
@sophipl, there are no plans _for or against_ handling the new firmware versions. Contributions are most welcome.

sophipl, 2024-04-05:
It reminds me of a scene in Start Trek Voyager, when captain says she feels she's getting drafted :D

jonasmalacofilho, 2025-01-04:
It seems I prematurely closed this: setting the screen to display GIFs still isn't supported.

vossiman, 2025-05-05:
short update.

➜ liquidctl --match kraken initialize
NZXT Kraken 2023 Elite (broken)
├── Firmware version    2.1.0  
├── LCD Brightness         80  %
└── LCD Orientation       270  °

still shows up as broken, although it seems most functionality is restored. 
GIF does not work, however setting an image does - altho it reverts back to default screen every X seconds.

BigosIgos, 2025-05-19:
can confirm gif still not working. its literally the only feature that i need. any way to contribute to a solution?

sophipl, 2025-05-20:
I'm not sure if this functionality still exists in new firmware.
From what I seen, windows software was streaming frames.
If gif functionality exists, I don't know what data to send to make it work.

damachine, 2026-01-21:
> Hi everyone,
> 
> One possible way to achieve animations on the Kraken LCD is by continuously sending static images that together form an animation. Using this approach, I created an animation inspired by the original NZXT CAM animation available on Windows.
> 
> The display shows two dynamic bars representing CPU and GPU temperatures. The script is written in Python and can run in the background starting at system boot.
> 
> Here is the project link if you’re interested in trying it out: [privato89/TempBarsGen](https://github.com/privato89/TempBarsGen?rgh-link-date=2025-12-24T15%3A09%3A53.000Z)
> 
> If you find the project useful or interesting, a star would be greatly appreciated.

Great idea I've also been looking into this problem for a while now and have taken the same approach to create an animation like the one for CAM.
https://github.com/damachine/coolerdash

PR #692 (2024-07-14): Kraken230 firmware 2 image support
Added ability to set images for Kraken with firmware version 2

Fixes: #631 
Closes: 
Related: 

---

Checklist:

- [x] Adhere to the [development process]
- [x] Conform to the [style guide]
- [x] Verify that the changes work as expected on real hardware
- [ ] Add automated tests cases
- [x] Verify that all (other) automated tests (still) pass
- [x] Update/add documentation
    - [x] README, with ["new/changed in" notes]
    - [x] applicable `docs/*guide.md` device guides, with ["new/changed in" notes]
    - [ ] `liquidctl.8` Linux/Unix/Mac OS man page
    -  [... cut]
commit 256311b9 (2024-07-14): kraken3: fix image support on Kraken 2023 with firmware 2.x (#692)

Fixes: #631