liquidctl/liquidctl #632: Unable to get --json function in liquidctl v1.5.1

CoolShades, 2023-10-23:
### Describe the bug

This is the latest liquidctl that can be pulled by installing `apt install liquidctl` on Ubuntu 22.04 (jammy)

I note that the --json flag and functionality was added, but this cannot be pulled on the actual program acquired from 
https://launchpad.net/ubuntu/+source/liquidctl/1.12.1-1

The functionality was added quite some time ago, so should have migrated its way into the latest release. Not sure why it hasnt.

### Commands executed

liquidctl status --match kraken --json

### Output of all relevant commands with `--debug` flag

sudo liquidctl --match kraken status --json --debug

Usage:
  liquidctl [options] list
  liquidctl [options] initialize [all]
  liquidctl [options] status
  liquidctl [options] set <channel> speed (<temperature> <percentage>) ...
  liquidctl [options] set <channel> speed <percentage>
  liquidctl [options] set <channel> color <mode> [<color>] ...
  liquidctl --help
  liquidctl --version

### Operating system and version

Ubuntu 22.04

### Installation method

Official Ubuntu apt

### Version of liquidctl

v1.5.1

jonasmalacofilho, 2023-10-23:
That's to be expected, as [`--json` was intruduced in liquidctl 1.7.0](https://github.com/liquidctl/liquidctl/blob/main/CHANGELOG.md?plain=1#L428).

It's just unfortunate that Ubuntu 22.04 still shipped (and then got stuck) with liquidctl 1.5.1.

closed 2023-10-23 by jonasmalacofilho

CoolShades, 2023-10-23:
Thanks for the rapid reply. Will it be possible for me to force install a later version? Or am I stuck with this version until it's rolled out?

jonasmalacofilho, 2023-10-23:
It's certainly possible and even easy to do that, modulo eventual issues with minimum Python version requirements (I haven't checked in a while, but I'm guessing Ubuntu 22.04 also ships with a relatively old Python version).

Take a look at our [Manual installation](https://github.com/liquidctl/liquidctl?tab=readme-ov-file#manual-installation) section for a starting point, although there are other routes would could take, depending on how you prefer to manage your system.