liquidctl/liquidctl #639: Corsair H150i Elite Capellix **XT**

gardotd426, 2023-11-21:
### Device type

AIO Liquid Cooler

### Product page

https://www.corsair.com/us/en/p/cpu-coolers/cw-9060070-ww/icue-h150i-elite-capellix-xt-liquid-cpu-cooler

### First-party software

Corsair iCue

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of embedded LEDs, lighting of accessories like RGB fans or LED strips

### Physical connection

Corsair Commander Core

### Connection protocol

Should be identical to the non-XT Capellix

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic, attempting to capture I²C/SMBus traffic

gardotd426, 2023-11-21:
Additional Context. I couldn't find support listed for the Capellix XT models, but non XT models are discontinued and don't support AM5. Literally everything seems identical to the non-XT models except the CPU block has more fancy LED lighting (though its still just RGB, no LCD, though you  can buy an aftermarket LCD to replace the cover with). 

I can't imagine it requiring anything other than adding the VID:PID info to the database. The API stuff should be identical to the Elite Capellix non-XT.

aleksamagicka, 2023-11-24:
> I can't imagine it requiring anything other than adding the VID:PID info to the database. The API stuff should be identical to the Elite Capellix non-XT.

I suppose that's worth a try. Since you haven't listed the IDs, can you please post the lsusb output, like [this](https://github.com/liquidctl/liquidctl/issues/511#issue-1385176066)?

gardotd426, 2024-01-18:
> I suppose that's worth a try. Since you haven't listed the IDs, can you please post the lsusb output, like [this](https://github.com/liquidctl/liquidctl/issues/511#issue-1385176066)?

I'm sorry I forgot to mention that I was planning on buying this, saw that it showed as supported, but then I see now that it's in the supported devices list but has a line crossed through it? 

I looked through the documentation, it says Core XT aren't meant to be used with AIOs, except that from what I can tell, the Elite Capellix AIOs COME with a Commander Pro XT specifically because they're ONLY for AIOs, you can't buy them standalone and a ton of their AIOs come with them. 

Could you maybe let me know what the issues are? I'm not going to buy it if i'm going to be relegated to stock pump speeds or if it doesn't work at all and my CPU overheats, but if you can let me know what works, what doesn't, I suppose I could set a pump curve in the BIOS and then use liquidctl to control RGB and the fan speed, assuming those work.

gardotd426, 2024-01-18:
Actually strike the Commander XT comment, but still, they come with the Commander Core, which should work fine. I just don't understand what's currently "broken" with those, when the entries for those devices don't mention any breakages.

jonasmalacofilho, 2024-01-18:
> I'm sorry I forgot to mention that I was planning on buying this, saw that it showed as supported, but then I see now that it's in the supported devices list but has a line crossed through it?

It showed as *experimentally* and *partially* supported, and at some point it did work fairly reliably, although with the limitations shown in the docs. However, since then Corsair changed the firmware in a significant way, and we haven't been able to update our driver to match it.

This is currently being tracked under several issues, either because they appear to be slightly different bugs, or because they affect different devices in the family: #520, #583, #598, #623. That said, all devices in the family appear to be affected in some way or another, as far as I know.

Unfortunately the people that were directly working on this driver don't have the time to do it anymore, and the core developers don't have access to one of these devices. So we don't even know if the issues above are still relevant, or if there are new issues.

Contributions here would be much appreciated. Those interested in working on this might benefit from also checking out closed issues and PRs that deal with this family of devices, as there are interesting tips on how the work was done some far (including a memory dump script) which might *not* be referenced in the still open issues.

> I looked through the documentation, it says Core XT aren't meant to be used with AIOs, except that from what I can tell, the Elite Capellix AIOs COME with a Commander Pro XT specifically because they're ONLY for AIOs, you can't buy them standalone and a ton of their AIOs come with them.

I haven't heard of a Commander *Pro* XT.  Did you by any chance mean *Core* XT?

gardotd426, 2024-02-22:
So, My Corsair H150i Elite I just got doesn't come with ANY corsair commander hub whatsoever. It's literally the same hookups as the NZXT Kraken x3 coolers, where it's just usb 2.0 coming from a usb type-c port on the AIO block, the only difference is that the fans are also controlled by this, whereas on NZXT you had to control the fans through the motherboard. 

I saw on the supported devices that the Elite (regular elite, not elite capellix), are crossed out and shown as broken in some major way. Could you elaborate? It SEEMS I'm able to control both the pumps and fans through liquidctl or CoolerControl (which uses liquidctl, so same thing). 

`liquidctl status`:
```
Corsair iCUE H150i Elite RGB
├── Liquid temperature    27.1  °C
├── Fan 1 speed           1846  rpm
├── Fan 1 duty             100  %
├── Fan 2 speed           1811  rpm
├── Fan 2 duty             100  %
├── Fan 3 speed           1856  rpm
├── Fan 3 duty             100  %
├── Pump speed            2945  rpm
└── Pump duty              100  %
```

What's the current issues? It wasn't crossed out when I bought it.

gardotd426, 2024-02-22:
Strike that Jonas, I was looking at the capellix listing, the iCUE elite capellix are what is shown to have major issues. The RGB Elite (what I ended up buying) is shown as fully supported. Sorry for bugging you I'm sure you got an email over this. Ignore it, my bad.