liquidctl/liquidctl #641: liquidctl not recognizing NZXT Kraken 360

moncefbenaicha, 2023-11-24:
### Describe the bug

I am using Ubuntu 22.04.3 LTS. I installed liquidctl in three different ways, the first time using apt install, then removed it and installed it using brew, and finally removed it and installed it again using python -m pip install. In all the cases the cooler is not detected. 

### Commands executed

```
$ liquidctl status
ERROR: no device matches available drivers and selection criteria
```

### Output of all relevant commands with `--debug` flag

```
$ liquidctl status --debug
[DEBUG] (cli) (_log_env_infos): script: /home/username/.local/share/pypoetry/venv/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.2.0-37-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_env_infos): python: 3.11.6 (main, Oct  2 2023, 13:45:54) [GCC 11.4.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.1.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1.post2
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-15
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-16
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4066 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4069 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:406a (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:4090 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4090 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4090 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4090 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4090 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 1e71:300e (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1e71:201d (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 174c:3074
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 8087:0033
[DEBUG] (usb) (find_devices): USB device: 1e71:201d
[DEBUG] (usb) (find_devices): USB device: 1e71:300e
[DEBUG] (usb) (find_devices): USB device: 046d:c52b
[DEBUG] (usb) (find_devices): USB device: 174c:2074
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002

```

### Affected device

NZXT kraken-360

### Operating system and version

Ubuntu 22.04.3 LTS

### Installation method

Python package

### Version of liquidctl

liquidctl v1.13.0

aleksamagicka, 2023-11-25:
Your device is getting detected as a USB device, which means it's present:

```
[DEBUG] (usb) (find_devices): HID device: 1e71:300e (usage_page=0xff00 usage=0x0001)
...
[DEBUG] (usb) (find_devices): USB device: 1e71:300e
```

Support for it in liquidctl has been merged just [last month](https://github.com/liquidctl/liquidctl/commit/846ac5f5aa9d5484e8ed1d549fd2478bbf51fdd8) and [v1.13.0](https://github.com/liquidctl/liquidctl/releases/tag/v1.13.0) is from July. Generally, please refer to the [compatibility table](https://github.com/liquidctl/liquidctl#supported-devices). You'll have to use the [Git version](https://github.com/liquidctl/liquidctl#installing-from-pypi-or-github) for the time being.

moncefbenaicha, 2023-11-25:
@aleksamagicka Thanks for the hint. I installed the last version from Git, now I am getting the following error:

```
$ liquidctl status
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
```

closed 2023-11-25 by moncefbenaicha

aleksamagicka, 2023-11-25:
> This problem was related to permission. Installing and running the package as root solved the issue. But not sure if that should be expected.

Permissions are covered in the [Allowing access to devices](https://github.com/liquidctl/liquidctl#allowing-access-to-the-devices) section of the README, after the Git install step. A package manager would set that up for you, but here you'd have to do it yourself. Additional instructions are available in the linked udev rules file, without it you'll indeed need root.