liquidctl/liquidctl #650: Error: no devices matches available drivers and selection criteria

a-d-wicklund, 2023-12-08:
I'm not able to find any of my fans with liquidctl, on Ubuntu 22.04.

`lsusb`
```
Bus 002 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 17ef:6099 Lenovo Lenovo Traditional USB Keyboard
Bus 001 Device 008: ID 8087:0033 Intel Corp. 
Bus 001 Device 006: ID 1e71:201d NZXT NZXT Motherboard
Bus 001 Device 004: ID 1e71:2011 NZXT NZXT RGB & Fan Controller
Bus 001 Device 007: ID 1e71:300c NZXT NZXT Kraken Elite
Bus 001 Device 005: ID 046d:c548 Logitech, Inc. USB Receiver
Bus 001 Device 003: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`lsusb -d 1e71: -v`
```
Bus 001 Device 006: ID 1e71:201d NZXT NZXT Motherboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x201d 
  bcdDevice            1.01
  iManufacturer           1 NZXT, Inc.
  iProduct                2 NZXT Motherboard
  iSerial                 3 0A02902DAB7C1CC162EB3572
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     349
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

Bus 001 Device 004: ID 1e71:2011 NZXT NZXT RGB & Fan Controller
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x2011 
  bcdDevice            1.07
  iManufacturer           1 NZXT, Inc.
  iProduct                2 NZXT RGB & Fan Controller
  iSerial                 3 0700C023AB4884A36224D707
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     432
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

Bus 001 Device 007: ID 1e71:300c NZXT NZXT Kraken Elite
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x300c 
  bcdDevice            2.00
  iManufacturer           1 NZXT, Inc.
  iProduct                2 NZXT Kraken Elite
  iSerial                 3 765893820E802B29
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 In Application
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     360
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
Device Status:     0x0001
  Self Powered
```

When I run liquidctl, I get an error:
```
$ sudo liquidctl status
Error: no devices matches available drivers and selection criteria
```
Is this a bug or some kind of system issue on my end?

aleksamagicka, 2023-12-08:
Minimum recommended liquidctl versions (MRLV) for your `NZXT RGB & Fan Controller` and `NZXT Kraken Elite` are v1.12.1 and git snapshots, respectively, as laid out in the [README](https://github.com/liquidctl/liquidctl#supported-devices). To gain support for both, you'll have to use the [Git version](https://github.com/liquidctl/liquidctl#installing-from-pypi-or-github) for the time being.

See #641 for a similar issue.

a-d-wicklund, 2023-12-08:
Ah of course, thanks for pointing this out.

I'm still running into problems with those two versions. With the git version, I just get:
```
$ liquidctl list
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
```
And with 1.12 I am getting 
 ```
$ liquidctl list
Device #0: NZXT RGB & Fan Controller (3+6 channels) (experimental)
$ liquidctl initialize
ERROR: NZXT RGB & Fan Controller (3+6 channels) (experimental): could not open, possibly due to insufficient permissions
```

Not sure if this sheds any light, but I'm not able to use the tool `pwmconfig` either... It seems to be unable to gain control of the fans. For example:
```
$ sudo pwmconfig
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

Found the following devices:
   hwmon0 is nvme
   hwmon1 is nvme
   hwmon2 is coretemp
   hwmon3 is nct6798
   hwmon4 is iwlwifi_1

Found the following PWM controls:
   hwmon3/pwm1           current value: 255
   hwmon3/pwm2           current value: 255
   hwmon3/pwm3           current value: 255
   hwmon3/pwm4           current value: 255
   hwmon3/pwm5           current value: 255
   hwmon3/pwm6           current value: 255
   hwmon3/pwm7           current value: 255

Giving the fans some time to reach full speed...
Found the following fan sensors:
   hwmon3/fan1_input     current speed: 0 ... skipping!
   hwmon3/fan2_input     current speed: 1134 RPM
   hwmon3/fan3_input     current speed: 0 ... skipping!
   hwmon3/fan4_input     current speed: 0 ... skipping!
   hwmon3/fan5_input     current speed: 0 ... skipping!
   hwmon3/fan6_input     current speed: 0 ... skipping!
   hwmon3/fan7_input     current speed: 0 ... skipping!

Warning!!! This program will stop your fans, one at a time,
for approximately 5 seconds each!!!
This may cause your processor temperature to rise!!!
If you do not want to do this hit control-C now!!!
Hit return to continue: 

Testing pwm control hwmon3/pwm1 ...
  hwmon3/fan2_input ... speed was 1134 now 1088
    no correlation

No correlations were detected.
There is either no fan connected to the output of hwmon3/pwm1,
or the connected fan has no rpm-signal connected to one of
the tested fan sensors. (Note: not all motherboards have
the pwm outputs connected to the fan connectors,
check out the hardware database on http://www.almico.com/forumindex.php)
```

aleksamagicka, 2023-12-08:
Did you set up the udev rules after installing a Git snapshot? To quote myself from the linked issue:

> Permissions are covered in the [Allowing access to devices](https://github.com/liquidctl/liquidctl#allowing-access-to-the-devices) section of the README, after the Git install step. A package manager would set that up for you, but here you'd have to do it yourself. Additional instructions are available in the linked udev rules file, without it you'll indeed need root.

---

As for pwmconfig:

```
Found the following devices:
   hwmon0 is nvme
   hwmon1 is nvme
   hwmon2 is coretemp
   hwmon3 is nct6798
   hwmon4 is iwlwifi_1
```

None of these are related to your devices. The Kraken Elite is not yet supported in the kernel driver (which isn't yet mainlined), and the other fan controller gained support in  the nzxt-smart2 kernel driver around 6.3, it seems. You're on 22.04, which is unlikely to have a recent kernel.

a-d-wicklund, 2023-12-11:
@aleksamagicka I followed what you suggested and it seems to have worked. I have control over the fans on both the Kraken device and the RGB device. Thank you!

closed 2023-12-11 by a-d-wicklund