liquidctl/liquidctl #657: Unable to upload any type of image/gif to Kraken Elite

MMaysin, 2023-12-28:
### Describe the bug

When I try to run the command to set the display to an image I get `AssertionError('Cannot find bulk out device')`. I don't really have much experience with python so it may be user error but I'm able to initialize and see the status of the device.

### Commands executed

I entered `liquidctl --match kraken set lcd screen gif gifs\dsc05067.jpg` and get `AssertionError('Cannot find bulk out device')`

Running `liquidctl --match kraken set lcd screen liquid` works and the orientation works but I cannot get gifs or static images to work.

### Output of all relevant commands with `--debug` flag

[DEBUG] (cli) (_log_env_infos): script: D:\Programs\LiquidCtl\liquidctl\liquidctl\__main__.py
[DEBUG] (cli) (_log_env_infos): version: 1.14.0.dev9+gaa16a87
[DEBUG] (cli) (_log_env_infos): platform: Windows-11-10.0.22631-SP0
[DEBUG] (cli) (_log_env_infos): python: 3.12.1 (tags/v3.12.1:2305ca5, Dec  7 2023, 22:03:25) [MSC v.1937 64 bit (AMD64)]
[DEBUG] (cli) (_log_env_infos): encoding: 1252 current, cp1252 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.1.0
[DEBUG] (cli) (_log_env_infos): with libusb-package: 1.0.26.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: 1.7
[DEBUG] (cli) (_log_env_infos): with smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0xff54 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 33e4:5807 (usage_page=0xffa0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b05:1a52 (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0x1337 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b05:19af (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): HID device: 33e4:5807 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 33e4:5807 (usage_page=0xffff usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1e71:300c (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken 2023 Elite (experimental)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0x1337 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1e71:2012 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1e71:2012 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 33e4:5807 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 33e4:5807 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0bda:1100 (usage_page=0xffda usage=0x00da)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 33e4:5807 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 33e4:5807 (usage_page=0xffff usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 31b2:0011 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 31e3:1512 (usage_page=0xff54 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 31e3:1312 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
[DEBUG] (usb) (find_devices): USB device: 1e71:2012
[DEBUG] (usb) (find_devices): USB device: 0489:e0e2
[DEBUG] (usb) (find_devices): USB device: 1022:15b8
[DEBUG] (usb) (find_devices): USB device: 0bda:1100
[DEBUG] (usb) (find_devices): USB device: 1022:15b6
[DEBUG] (usb) (find_devices): USB device: 0b05:19af
[DEBUG] (usb) (find_devices): USB device: 33e4:5807
[DEBUG] (usb) (find_devices): USB device: 31e3:1512
[DEBUG] (usb) (find_devices): USB device: 0b05:1a52
[DEBUG] (usb) (find_devices): USB device: 1022:15b7
[DEBUG] (usb) (find_devices): USB device: 1022:43f7
[DEBUG] (usb) (find_devices): USB device: 1e71:2012
[DEBUG] (usb) (find_devices): USB device: 1022:43f7
[DEBUG] (usb) (find_devices): USB device: 1e71:300c
[DEBUG] (usb) (find_devices): USB device: 31e3:1312
[DEBUG] (usb) (find_devices): USB device: 0bda:0411
[DEBUG] (usb) (find_devices): USB device: 0bda:5411
[DEBUG] (usb) (find_devices): USB device: 31b2:0011
[DEBUG] (usb) (find_devices): USB device: 046d:082d
[DEBUG] (cli) (main): device: NZXT Kraken 2023 Elite (experimental)
[DEBUG] (usb) (write): writing report 0x30 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 31:01:79:80:92:82:0e:90:20:09:00:00:00:00:05:00:80:00:00:10:80:02:80:02:64:01:03:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[INFO] (cli) (log): detailed error: NZXT Kraken 2023 Elite (experimental): unexpected error: AssertionError('Cannot find bulk out device')
Traceback (most recent call last):
  File "D:\Programs\LiquidCtl\liquidctl\liquidctl\cli.py", line 479, in main
    _device_set_screen(dev, args, **opts)
  File "D:\Programs\LiquidCtl\liquidctl\liquidctl\cli.py", line 287, in _device_set_screen
    dev.set_screen(args["<channel>"], args["<mode>"], args["<value>"], **opts)
  File "D:\Programs\LiquidCtl\liquidctl\liquidctl\driver\kraken3.py", line 796, in set_screen
    self._send_data(data, [0x01, 0x0, 0x0, 0x0] + list(len(data).to_bytes(4, "little")))
  File "D:\Programs\LiquidCtl\liquidctl\liquidctl\driver\kraken3.py", line 872, in _send_data
    assert self.bulk_device, "Cannot find bulk out device"
AssertionError: Cannot find bulk out device
[ERROR] (cli) (log): NZXT Kraken 2023 Elite (experimental): unexpected error: AssertionError('Cannot find bulk out device')

### Affected device

Kraken Elite 360

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Windows 11

### Installation method

Python

### Version of liquidctl

liquidctl v1.14.0.dev9+gaa16a87 (Windows-11-10.0.22631-SP0)

Mint-Martini, 2024-01-01:
Just wanted to add i am also experiencing this bug , the device is being seen but gifs are not loading in
[image]

jonasmalacofilho, 2024-02-23:
Since this and #631 are intrinsically related, let's only keep the former. Therefore, closing this as (now known to be a) duplicate of #631.

closed 2024-02-23 by jonasmalacofilho