liquidctl/liquidctl #660: Kraken 2023 280 not being seen

hollnagelc, 2024-01-09:
I'm using OpenSuse Tumbleweed. liquidctl 1.13.0

liquidctl status 
ERROR: no device matches available drivers and selection criteria

results from lsusb command has the kraken

Bus 001 Device 004: ID 1e71:300e NZXT NZXT Kraken Base

hollnagelc, 2024-01-09:
Probably something I'm doing wrong but i get this error. 

pip failed to build package:
    smbus

aleksamagicka, 2024-01-09:
That's a vague error log and I don't even know what platform you are on, but here's a shot in the dark - ensure that you're using Python 3 and not Python 2.

aleksamagicka, 2024-01-09:
Duh, sorry - you're on Tumbleweed. I'll spin up a VM to try.

hollnagelc, 2024-01-09:
> Duh, sorry - you're on Tumbleweed. I'll spin up a VM to try.

I think it's somehting on my end for sure. I just ran the command on my laptop with Nobara (fedora) and it install the git version no problem

hollnagelc, 2024-01-09:
That did the trick i can see the Kraken 2023 (experimental). Thanks alot