liquidctl/liquidctl #664: Corsair H110i

ciberkids, 2024-01-15:
### Device type

AIO LIquid cooler 

### Product page

https://www.corsair.com/us/en/p/cpu-coolers/cw-9060026-ww/hydro-series-h110i-280mm-extreme-performance-liquid-cpu-cooler-cw-9060026-ww

### First-party software

Icue

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, voltages, current or power limits

### Physical connection

USB header

### Connection protocol

USB

### Additional information about the device

```
$ lsusb
Bus 001 Device 004: ID 1b1c:0c04 Corsair Link Cooling Node
```

```
$ sudo OpenCorsairLink status
Dev=0, CorsairLink Device Found: H110i!
```

```
 $ sudo OpenCorsairLink --device 0
Dev=0, CorsairLink Device Found: H110i!

Vendor: Corsair
Product: H110i
Firmware: 2.0.0
Temperature 0: 26.98 C
Fan 0:	Default Mode (4PIN)
	Current/Max Speed 701/2128 RPM
Fan 1:	Default Mode (4PIN)
	Current/Max Speed 685/2148 RPM
Pump:	Mode 0x87
	Current/Max Speed 701/2128 RPM
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

ciberkids, 2024-01-15:
thanks, i didn't read that bug number and PR :) 

also how can i help?

should i start as described in the article and porting the device ex novo or i build on top that pr?

jonasmalacofilho, 2024-01-15:
You'll be almost certainly better building on top of that PR, since (IIRC the OpenCorsairLink code) those devices are all super similar.

Besides any specific changes needed for the non GT model, you'll likely need to fix the bug that PR is currently stuck on. Otherwise it should be fairly straightforward, I hope.

jonasmalacofilho, 2024-01-15:
Ping: @Serphentas

ciberkids, 2024-01-15:
i see, thanks for the fast answer, i will have a look this evening :)

Serphentas, 2024-04-30:
Sorry I didn't see the ping @jonasmalacofilho. I think OCL supports this device (non-GT) with the same driver, so perhaps we need fine tuning.

@ciberkids are you able to run OCL from the start (to avoid the issues you noted) in debug mode, or possibly capture from Windows using Corsair's tool ? Thanks.

ciberkids, 2024-05-07:
i have seen a couple of commits and i tried with the latest release on my machine, now it seems to work correctly

liquidctl-git 1.13.0.r70.01ef92f-1

```
$ sudo liquidctl status
Corsair H110i GT
├── Liquid temperature    26.7  °C
├── Fan 1 speed            648  rpm
├── Fan 2 speed            639  rpm
└── Pump speed            2356  rpm
```

jonasmalacofilho, 2024-05-11:
Oh, we forgot to query the devices for their ID, and we're binding with *any* CoolIT device from that era.

One way to fix this would be to add a custom `probe` function to the driver that queries the ID and matches the response to a set of know devices. However, that would result in traffic being sent to the device every time liquidctl lists (any) devices, which seems far from ideal.

I think a better way is for the driver to keep using the default `probe` logic and at that stage only recognize device as a generic CoolIT cooler. The on `connect` the driver can fetch the device ID and adjust the behavior based on it. We do something somewhat similar for ASUS GPUs.

---

Relevant OpenCorsairLink code for querying the device ID:

```c
int
corsairlink_coolit_device_id(
    struct corsair_device_info* dev, struct libusb_device_handle* handle, uint16_t* device_id )
{
    int rr;
    uint8_t response[64];
    uint8_t commands[64];
    memset( response, 0, sizeof( response ) );
    memset( commands, 0, sizeof( commands ) );

    uint8_t ii = 0;

    commands[++ii] = CommandId++; // Command ID
    commands[++ii] = ReadOneByte; // Command Opcode
    commands[++ii] = DeviceID; // Command data...
    commands[0] = ii; // Length

    rr = dev->lowlevel->write( handle, dev->write_endpoint, commands, 64 );
    rr = dev->lowlevel->read( handle, dev->read_endpoint, response, 64 );

    memcpy( device_id, response + 2, 1 );

    dump_packet( response, sizeof( response ) );

    return 0;
}
```

—https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/protocol/coolit/core.c#L32-L57

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.

closed 2025-11-15 by jonasmalacofilho