liquidctl/liquidctl #665: NZXT Kraken X no longer functioning: OSError('Could not write to device')

jpsalm, 2024-01-25:
### Describe the bug

I have a NZXT Kraken X62 which is no longer working properly with liquidctl.  I believe this started after updating to kernel 6.7, but I haven't reverted to 6.6 to verify yet.

Previously I was having no issues using liquidctl via coolercontrol but now it appears liquidctl is unable to write to the device.

### Commands executed

```
$  liquidctl status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature    28.0  °C
├── Fan speed              746  rpm
└── Pump speed            1698  rpm
```

```
$ liquidctl initialize all  
ERROR: NZXT Kraken X (X42, X52, X62 or X72): unexpected OS error: OSError('Could not write to device')
```

```
$  liquidctl set fan speed 50 
ERROR: NZXT Kraken X (X42, X52, X62 or X72): unexpected OS error: OSError('Could not write to device')
```

Running these as root doesn't seem to have any effect.  No logs in dmesg.

### Output of all relevant commands with `--debug` flag

```
$ liquidctl --debug initialize all
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.7.0-arch3-1-x86_64-with-glibc2.38
[DEBUG] (cli) (_log_env_infos): python: 3.11.6 (main, Nov 14 2023, 09:36:21) [GCC 13.2.1 20230801]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.2.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, Krak
enZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0424:274c (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b0e:0420 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b0e:0420 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b0e:0420 (usage_page=0xff30 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0b0e:0420 (usage_page=0x000b usage=0x0005)
[DEBUG] (usb) (find_devices): HID device: 0b0e:0420 (usage_page=0x000b usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0424:274b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d9:0356 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 04d9:0356 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d9:0356 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 04d9:0356 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d9:0356 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 04d9:0356 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d9:0356 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1038:1830 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1038:1830 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:1830 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1038:1830 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:1830 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:1830 (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1462:7c35 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1462:7c35 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1e71:170e (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (__init__): has kernel driver: nzxt-kraken2 (/sys/class/hidraw/hidraw1/device/hwmon/hwmon4)
[DEBUG] (usb) (probe): Kraken2 identified: NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 05e3:0626
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d50:606f
[DEBUG] (usb) (find_devices): USB device: 25a4:9311
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 045e:02ea
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1e71:170e
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1462:7c35
[DEBUG] (usb) (find_devices): USB device: 04e8:6860
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 0424:5734
DEBUG] (usb) (find_devices): USB device: 0424:5734
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1038:1830
[DEBUG] (usb) (find_devices): USB device: 04d9:0356
[DEBUG] (usb) (find_devices): USB device: 8087:0032
[DEBUG] (usb) (find_devices): USB device: 0424:274b
[DEBUG] (usb) (find_devices): USB device: 0b0e:0420
[DEBUG] (usb) (find_devices): USB device: 1d6c:0103
[DEBUG] (usb) (find_devices): USB device: 0424:2734
[DEBUG] (usb) (find_devices): USB device: 0424:274c
[DEBUG] (usb) (find_devices): USB device: 0424:2734
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] (usb) (read): read 64 bytes: 04:1d:06:02:ea:06:9e:00:00:00:ae:06:00:00:02:1e:00:00:00:00:32:00:00:00:1d:00:00:00:19:00:00:00:41:91:00:08:ea:02:00:00:9e:06:00:00:32:00:00:00:33:00:00:00:1d:00:00:00:64:12:00:20:00:00:00:00
[INFO] (kraken2) (set_speed_profile): setting fan PWM duty to 25% for liquid temperature >= 20°C
[DEBUG] (usb) (write): writing report 0x02 with 64 bytes: 4d:80:14:19:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[INFO] (cli) (log): detailed error: NZXT Kraken X (X42, X52, X62 or X72): unexpected OS error: OSError('Could not write to device')
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/liquidctl/cli.py", line 463, in main
    status = dev.initialize(**opts)
             ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/liquidctl/driver/kraken2.py", line 133, in initialize
    self.set_speed_profile('fan', _RESET_FAN_PROFILE)
  File "/usr/lib/python3.11/site-packages/liquidctl/driver/kraken2.py", line 251, in set_speed_profile
    self._write([0x2, 0x4d, cbase + i, temp, duty])
  File "/usr/lib/python3.11/site-packages/liquidctl/driver/kraken2.py", line 292, in _write
    self.device.write(data + padding)
  File "/usr/lib/python3.11/site-packages/liquidctl/driver/usb.py", line 480, in write
    raise OSError('Could not write to device')
OSError: Could not write to device
[ERROR] (cli) (log): NZXT Kraken X (X42, X52, X62 or X72): unexpected OS error: OSError('Could not write to device')
```

### Affected device

Kraken X62

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Arch Linux

### Installation method

Official Arch Linux Repository

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.7.0-arch3-1-x86_64-with-glibc2.38)

aleksamagicka, 2024-01-25:
The only recent change in the kernel driver itself (nzxt-kraken2) is [nzxt-kraken2: Fix error handling path in probe function](https://git.kernel.org/pub/scm/linux/kernel/git/groeck/linux-staging.git/commit/drivers/hwmon/nzxt-kraken2.c?h=hwmon-next&id=35fe2ad259a3bfca15ab78c8ffb5278cb6149c89), which fixed some error cases. It's in from 6.6.7 and 6.7 as far as I can see, possibly earlier 6.x versions as well.

Perhaps try without the kernel driver inserted?

```bash
sudo modprobe -r nzxt_kraken2
```

jpsalm, 2024-01-25:
Thanks, that did it for me, somehow the module wasn't loaded at all.

closed 2024-01-25 by jpsalm