liquidctl/liquidctl #667: 1.13.0: pytest fails

kloczek, 2024-01-29:
### Describe the bug

I'm packaging your module as an rpm package so I'm using the typical PEP517 based build, install and test cycle used on building packages from non-root account.
- `python3 -sBm build -w --no-isolation`
- because I'm calling `build` with `--no-isolation` I'm using during all processes only locally installed modules
- install .whl file in </install/prefix> using 'installer` module
- run pytest with $PYTHONPATH pointing to sitearch and sitelib inside </install/prefix>
- build is performed in env which is *`cut off from access to the public network`* (pytest is executed with `-m "not network"`)

<details>
<summary>List of installed modules in build env:</summary>

```console
Package            Version
------------------ -------
build              1.0.3
colorlog           6.8.2
cppclean           0.13
crcmod             1.7
distro             1.9.0
dnf                4.18.2
docopt             0.6.2
exceptiongroup     1.1.3
gpg                1.23.2
hidapi             0.14.0
importlib-metadata 7.0.1
iniconfig          2.0.0
installer          0.7.0
libdnf             0.72.0
olefile            0.47
packaging          23.2
pillow             10.2.0
pluggy             1.3.0
pyproject_hooks    1.0.0
pytest             8.0.0
python-dateutil    2.8.2
pyusb              1.2.1
setuptools         69.0.3
setuptools-scm     8.0.4
six                1.16.0
smbus              1.1
tomli              2.0.1
typing_extensions  4.9.0
wheel              0.42.0
zipp               3.17.0
```
</details>

### Commands executed

pytest

### Output of all relevant commands with `--debug` flag

N/A

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Linux x86/64

### Installation method

N/A

### Version of liquidctl

1.13.0

kloczek, 2024-01-29:
<details>
<summary>Here is pytest output:</summary>

```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/liquidctl-1.13.0-4.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/liquidctl-1.13.0-4.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra -m 'not network'
==================================================================================== test session starts ====================================================================================
platform linux -- Python 3.8.18, pytest-8.0.0, pluggy-1.3.0
rootdir: /home/tkloczko/rpmbuild/BUILD/liquidctl-1.13.0
configfile: pyproject.toml
collected 5 items / 86 errors

========================================================================================== ERRORS ===========================================================================================
__________________________________________________________________________ ERROR collecting liquidctl/__init__.py ___________________________________________________________________________
liquidctl/__init__.py:67: in <module>
    from liquidctl.driver import find_liquidctl_devices
liquidctl/driver/__init__.py:26: in <module>
    from liquidctl.driver import asetek
liquidctl/driver/asetek.py:23: in <module>
    from liquidctl.driver.usb import UsbDriver
liquidctl/driver/usb.py:67: in <module>
    import hidraw as hid
E   ImportError: dynamic module does not define module export function (PyInit_hidraw)
__________________________________________________________________________ ERROR collecting liquidctl/_version.py ___________________________________________________________________________
/usr/lib64/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
<frozen importlib._bootstrap>:1014: in _gcd_import
    ???
<frozen importlib._bootstrap>:991: in _find_and_load
    ???
<frozen importlib._bootstrap>:961: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:219: in _call_with_frames_removed
    ???
<frozen importlib._bootstrap>:1014: in _gcd_import
    ???
<frozen importlib._bootstrap>:991: in _find_and_load
    ???
<frozen importlib._bootstrap>:975: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:671: in _load_unlocked
    ???
<frozen importlib._bootstrap_external>:843: in exec_module
    ???
<frozen importlib._bootstrap>:219: in _call_with_frames_removed
    ???
liquidctl/__init__.py:67: in <module>
    from liquidctl.driver import find_liquidctl_devices
liquidctl/driver/__init__.py:26: in <module>
    from liquidctl.driver import asetek
liquidctl/driver/asetek.py:23: in <module>
    from liquidctl.driver.usb import UsbDriver
liquidctl/driver/usb.py:67: in <module>
    import hidraw as hid
E   ImportError: dynamic module does not define module export function (PyInit_hidraw)
_____________________________________________________________________________ ERROR collecting liquidctl/cli.py _____________________________________________________________________________
/usr/lib64/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
<frozen importlib._bootstrap>:1014: in _gcd_import
    ???
<frozen importlib._bootstrap>:991: in _find_and_load
    ???
<frozen importlib._bootstrap>:961: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:219: in _call_with_frames_removed
    ???
<frozen importlib._bootstrap>:1014: in _gcd_import
    ???
<frozen importlib._bootstrap>:991: in _find_and_load
    ???
[... 1848 lines cut ...]
ERROR tests/test_keyval.py
ERROR tests/test_kraken2.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_kraken2.py
ERROR tests/test_kraken3.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_kraken3.py
ERROR tests/test_nvidia.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_nvidia.py
ERROR tests/test_nzxt_epsu.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_nzxt_epsu.py
ERROR tests/test_nzxt_h1_v2.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_nzxt_h1_v2.py
ERROR tests/test_rgb_fusion2.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_rgb_fusion2.py
ERROR tests/test_smart_device.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_smart_device.py
ERROR tests/test_smart_device2.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_smart_device2.py
ERROR tests/test_smbus.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_smbus.py
ERROR tests/test_usb.py - ImportError: dynamic module does not define module export function (PyInit_hidraw)
ERROR tests/test_usb.py
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 86 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
==================================================================================== 86 errors in 4.91s =====================================================================================
```
</details>

jonasmalacofilho, 2024-01-29:
Perhaps #145 can be a good starting point to figuring this out.

Can you begin by running `ldd` on any `hid*.so` in the build environment?

jonasmalacofilho, 2024-01-29:
Also note that in the build environment

```
$ python -c "import usb; import hid; import hidraw"
```

should either pass or, at most, fail with `ModuleNotFoundError: No module named 'hidraw'`, depending on how cython-hidapi was built.

jonasmalacofilho, 2024-04-05:
Closing due to lack of response and because so far it seems that the issue is a broken cython-hidapi (`hid` & `hidraw` Python modules) installation.