liquidctl/liquidctl #668: Opening any GUI software (Firefox, VSCode, etc...) makes the fans spin at max speed for a couple of seconds

st1vms, 2024-02-04:
### Describe the bug

Recently I noticed a strange behaviour on my Ubuntu LTS machine with my H100i Elite RGB fan speed, whenever I open a GUI program such as the browser (Firefox) or VSCode for example, my fans start spinning at the maximum speed for a couple of seconds, then they return back to their original state set by liquidctl...This happens every time the GUI software gets opened ( even after closing it and re-opening it).

I don't have any other software controlling the fan speeds. I do have three chassis fans set to Silent directly from BIOS, two are in DC mode ( the ones that came along the case ) and two other PWM fans i gathered from an old liquid corsair kit. I usually never had this problem before doing so. 
Until one day when booting, I noticed the chassis fans were spinning at maximum speed constantly, so I set the Q-fan profile to Silent in my BIOS for these chassis fans directly.

Don't know if this bug is directly caused by liquidctl tho...

 

### Commands executed

### liquidctl status

```
$ liquidctl status

Corsair iCUE H100i Elite RGB (experimental)
├── Liquid temperature    29.1  °C
├── Fan 1 speed            745  rpm
├── Fan 1 duty              38  %
├── Fan 2 speed            741  rpm
├── Fan 2 duty              38  %
├── Pump speed            2013  rpm
└── Pump duty               55  %

WARNING: some attributes cannot be read from corsair-psu kernel driver
Corsair HX1000i (2022) (experimental)
├── VRM temperature            42.2  °C
├── Case temperature           40.5  °C
├── Fan speed                     0  rpm
├── Input voltage            230.00  V
├── +12V output voltage       11.94  V
├── +12V output current        4.00  A
├── +12V output power         46.00  W
├── +5V output voltage         5.03  V
├── +5V output current         4.31  A
├── +5V output power          21.50  W
├── +3.3V output voltage       3.31  V
├── +3.3V output current       2.81  A
├── +3.3V output power         9.00  W
├── Total power output        98.00  W
├── Estimated input power    936.00  W
└── Estimated efficiency         10  %

ASUS Aura LED Controller
├── ARGB channels: 3      
└──  RGB channels: 1 
```

### udev rules
```
$ cat /etc/udev/rules.d/71-liquidctl.rules

# ASUS Aura LED Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="19af", TAG+="uaccess"

# ASUS Aura LED Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="1939", TAG+="uaccess"

# ASUS Aura LED Controller
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="18f3", TAG+="uaccess"

# Corsair iCUE H100i Elite RGB
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c35", TAG+="uaccess"

# Corsair HX1000i (2022)
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="1c1e", TAG+="uaccess"
```
[... 176 lines cut ...]
├── Device Config: 7     0x00, 0x00, 0x00, 0x00, 0x00, 0x00  
├── Device Config: 8     0x00, 0x00, 0x00, 0x00, 0x00, 0x00  
├── Device Config: 9     0x00, 0x00, 0x00, 0x00, 0x00, 0x00  
└── Device Config: 10    0x00, 0x00, 0x00, 0x00, 0x00, 0x00 
```

### Affected device

Corsair H100i Elite RGB

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Ubuntu 22.04.3 LTS

### Installation method

pip install liquidctl ( installed both in user mode and in sudo mode )

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.5.0-15-generic-x86_64-with-glibc2.35)

aleksamagicka, 2024-02-04:
I don't think I follow completely... which fans are controlled by liquidctl, which ones are controlled by the BIOS, how are they all connected to the PC, which ones are you having a problem with?

st1vms, 2024-02-04:
I'm sorry apparently it's not actually liquidctl causing this issue, as liquidctl is responsible for controlling the H100i fans only... and by watching fans speed I canno't actually see a noticeable difference in the H100i speeds, so it's probably the chassis fans.

closing this...

closed 2024-02-04 by st1vms

aleksamagicka, 2024-02-04:
> so it's probably the chassis fans.

That's what I thought, but first wanted to ask; this sounds like the BIOS reacting to CPU temp jumping. Try setting the hysteresis/delay in BIOS, it should have the option. My ASUS mobo does.