liquidctl/liquidctl #671: unable to set speeds after power cycle with git version | octo

nyllewin, 2024-02-16:
### Describe the bug

Hey, I switched to the git version due to recent pwm issue and fix (which is working just fine btw thanks! :D) and just noticed now that I'm unable to set speeds for fans after a power off and power on.
If I were to just upgrade from the pacman (is this how to call it? sorry I'm new lol) version it would work fine, even after a reboot.
However after a Off/On cycle it would not. The Octo would just stay at fallback speeds (which I set sometime ago in windows) and basically not react to anything I tell it until I go through and downgrade to pacman version and then upgrade to git version via aur again.

Also it's requiring sudo now, is that normal?

Thank you!

### Commands executed

sudo liquidctl set fan4 speed 0
sudo liquidctl set fan4 speed 100

### Output of all relevant commands with `--debug` flag

uname -r
6.7.4-arch1-1
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.14.0.dev37+g14dd33a
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.7.4-arch1-1-x86_64-with-glibc2.39
[DEBUG] (cli) (_log_env_infos): python: 3.11.7 (main, Jan 29 2024, 16:03:57) [GCC 13.2.1 20230801]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.2.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0c70:f011 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (__init__): has kernel driver: aquacomputer_d5next (/sys/class/hidraw/hidraw0/device/hwmon/hwmon3)
[DEBUG] (usb) (probe): Aquacomputer identified: Aquacomputer Octo
[DEBUG] (usb) (find_devices): HID device: 0424:274b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:0043 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0424:274b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa14 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa14 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 25a7:fa14 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 2109:8110
[DEBUG] (usb) (find_devices): USB device: 0424:5734
[DEBUG] (usb) (find_devices): USB device: 0424:5734
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 2109:2811
[DEBUG] (usb) (find_devices): USB device: 0424:274b
[DEBUG] (usb) (find_devices): USB device: 1532:0043
[DEBUG] (usb) (find_devices): USB device: 0424:2734
[DEBUG] (usb) (find_devices): USB device: 0424:274b
[DEBUG] (usb) (find_devices): USB device: 045e:0719
[DEBUG] (usb) (find_devices): USB device: 0424:2734
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1397:0510
[DEBUG] (usb) (find_devices): USB device: 8087:0029
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 05e3:0749
[DEBUG] (usb) (find_devices): USB device: 05e3:0626
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 0c70:f011
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Aquacomputer Octo
[INFO] (aquacomputer) (set_fixed_speed): bound to aquacomputer_d5next kernel driver, writing fixed speed to hwmon

### Affected device

aquacomputer octo

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

arch linux

### Installation method

aur

### Version of liquidctl

liquidctl v1.14.0.dev37+g14dd33a (Linux-6.7.4-arch1-1-x86_64-with-glibc2.39)

aleksamagicka, 2024-02-17:
Also, is anything else configuring the device? Are you maybe booting into windows with aquasuite after poweroff or similar?

nyllewin, 2024-02-17:
I got in to this and this looks familiar to the liquidctl output.

```
[nylle@nyArch hwmon]$ cd hwmon3
[nylle@nyArch hwmon3]$ ls
curr1_input  curr8_input  fan6_label  in4_label     power3_label  pwm4          temp14_input  temp1_offset  temp6_input
curr1_label  curr8_label  fan7_input  in5_input     power4_input  pwm5          temp14_label  temp20_input  temp6_label
curr2_input  device       fan7_label  in5_label     power4_label  pwm6          temp15_input  temp20_label  temp7_input
curr2_label  fan1_input   fan8_input  in6_input     power5_input  pwm7          temp15_label  temp2_input   temp7_label
curr3_input  fan1_label   fan8_label  in6_label     power5_label  pwm8          temp16_input  temp2_label   temp8_input
curr3_label  fan2_input   in0_input   in7_input     power6_input  subsystem     temp16_label  temp2_offset  temp8_label
curr4_input  fan2_label   in0_label   in7_label     power6_label  temp10_input  temp17_input  temp3_input   temp9_input
curr4_label  fan3_input   in1_input   name          power7_input  temp10_label  temp17_label  temp3_label   temp9_label
curr5_input  fan3_label   in1_label   power         power7_label  temp11_input  temp18_input  temp3_offset  uevent
curr5_label  fan4_input   in2_input   power1_input  power8_input  temp11_label  temp18_label  temp4_input
curr6_input  fan4_label   in2_label   power1_label  power8_label  temp12_input  temp19_input  temp4_label
curr6_label  fan5_input   in3_input   power2_input  pwm1          temp12_label  temp19_label  temp4_offset
curr7_input  fan5_label   in3_label   power2_label  pwm2          temp13_input  temp1_input   temp5_input
curr7_label  fan6_input   in4_input   power3_input  pwm3          temp13_label  temp1_label   temp5_label
[nylle@nyArch hwmon3]$ cat name
octo

[nylle@nyArch hwmon3]$ sudo echo 255 > pwm5
bash: pwm5: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm4
bash: pwm4: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm3
bash: pwm3: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm2
bash: pwm2: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm1
bash: pwm1: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm6
bash: pwm6: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm7
bash: pwm7: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm0
bash: pwm0: Permission denied
[nylle@nyArch hwmon3]$ sudo echo 255 > pwm8
bash: pwm8: Permission denied

this then finally worked:

nylle@nyArch hwmon3]$ sudo su
[sudo] password for root: 
[root@nyArch hwmon3]# echo 255 > pwm4
[root@nyArch hwmon3]# exit
exit
[nylle@nyArch hwmon3]$ 

```

the last step worked. I have no Idea why it isn't working via sudo.
It swapped back to 0rpm after a second or two - what I set before via direct-access.

> should be quadro
It's octo, same same :D

Rebooted now, same behavior.
off-on cycle.

none of these work until after I talked to that specific fan via direct-access flag then and only then the pwm is accepted.
- there is no error output, no reaction on the device as fan speeds stay the same.

```
[nylle@nyArch hwmon3]$ sudo su
[sudo] password for root: 
[root@nyArch hwmon3]# echo 0 > pwm4
[root@nyArch hwmon3]# echo 0 > pwm2
[root@nyArch hwmon3]# echo 0 > pwm3
[root@nyArch hwmon3]# echo 255 > pwm3
[root@nyArch hwmon3]# echo 255 > pwm2
[root@nyArch hwmon3]# echo 255 > pwm4
[root@nyArch hwmon3]# exit
exit
[nylle@nyArch hwmon3]$ sudo liquidctl set fan3 speed 100 --direct-access
WARNING: directly writing fixed speed despite aquacomputer_d5next kernel driver having support
[nylle@nyArch hwmon3]$ sudo su
[root@nyArch hwmon3]# echo 255 > pwm4
[root@nyArch hwmon3]# echo 0 > pwm3
[root@nyArch hwmon3]# exit
exit
[nylle@nyArch hwmon3]$ sudo liquidctl set fan4 speed 100 --direct-access
WARNING: directly writing fixed speed despite aquacomputer_d5next kernel driver having support
[nylle@nyArch hwmon3]$ sudo su
[root@nyArch hwmon3]# echo 0 > pwm4
[root@nyArch hwmon3]# 
```

I tried installing the rules for not needing sudo but after installing them and reloading I still needed to authorize, not sure if the way I did it was wrong but the reloads didn't trigger any errors when executed with sudo. I went in to the directory for user rules, sudo nano filename 71xxxx then copied via ctrl c from github raw the text and pasted, reloaded. So after that didn't work i renamed them to 71xx.rules.backup in both directories where they were present. the distro one and the user one.
not sure idk if related mentioned for good measure.

> Also, is anything else configuring the device? Are you maybe booting into windows with aquasuite after poweroff or similar?
Controlling not so much. There is coolercontrol where I am also not able to set speeds with on either device, the hwmon one and the liquidctl one so I wouldn't call that " controlling " .. its reading the sensors and so on but that's about it. I can uninstall.

I don't have a windows install anymore so that's out.

I wish I could have brought better news but yea. let me know, I'm down for anything basically :D

aleksamagicka, 2024-02-17:
Yeah, you'll need `sudo su` for hwmon, sorry for not mentioning.

> It swapped back to 0rpm after a second or two - what I set before via direct-access.

That seems weird, looks like something else is controlling it?

Can you please try the github version of the driver ([instructions](https://github.com/aleksamagicka/aquacomputer_d5next-hwmon/tree/main?tab=readme-ov-file#compiling-and-installation))? Then go to hwmon again, and do:

```bash
cat pwm1_enable
```

And see what [the value is](https://github.com/aleksamagicka/aquacomputer_d5next-hwmon/blob/main/docs/aquacomputer_d5next.rst#usage-notes). Should be `1` for a direct PWM value, but you say:

> none of these work until after I talked to that specific fan via direct-access flag then and only then the pwm is accepted.

So maybe it's something else before a "direct access" setting. If it is, set it to `1` and then try setting `pwm1` as before.

I am not familiar with coolercontrol as I don't use it, but let's try to rule it out, so kill the process or something.

And yes, Octo or Quadro should make no difference.

nyllewin, 2024-02-17:
I stopped the coolercontrol daemon service did this after:

> Can you please try the github version of the driver ([instructions](https://github.com/aleksamagicka/aquacomputer_d5next-hwmon/tree/main?tab=readme-ov-file#compiling-and-installation))?
```

make[2]: Leaving directory '/usr/lib/modules/6.7.4-arch1-1/build'
make[1]: Leaving directory '/home/nylle/Documents/aquacomputer_d5_next_hwmon/aquacomputer_d5next-hwmon'
sudo rmmod aquacomputer_d5next || true
[sudo] password for root: 
sudo insmod aquacomputer_d5next.ko
[nylle@nyArch aquacomputer_d5next-hwmon]$ 
```

this succeeded I think? I did this, off / on cycle and did this again. I'm sorry I'm really new lmao

checked for coolercontrol service or if its working doing anything
```
ps aux | grep coolercontrol
nylle       3180  0.0  0.0   6560  2304 pts/3    S+   12:45   0:00 grep --color=auto coolercontrol
```
and we got nothing, good!

Now I did this:
cat pwm1_enable
or rather tried..
```
[nylle@nyArch hwmon]$ cd hwmon3/
[nylle@nyArch hwmon3]$ cat name
octo
[nylle@nyArch hwmon3]$ cat pwm1_enable
cat: pwm1_enable: No such file or directory
[nylle@nyArch hwmon3]$ 

```
So I went back and did make dev in the d5 next driver again and then they showed up. Also assuming that's normal? because i reboot and then it loads the old kernel version and so on.. surely.

anyways.

```
 cat pwm1_enable 
3
 cat pwm2_enable 
2
 cat pwm3_enable 
2
 cat pwm4_enable 
2
 cat pwm5_enable 
1
 cat pwm6_enable 
2
 cat pwm7_enable 
1
 cat pwm8_enable 
1

```
this is from fresh boot, device running at fallback speeds as far as I can tell because this stupid 80mm fan is driving me nuts with its noise and some others also just run rather high rpm.

Ok
```
[nylle@nyArch hwmon3]$ sudo su
[sudo] password for root: 
[root@nyArch hwmon3]# echo 1 > pwm4_enable 
```
this kicked the fan from 25% (fallback speed) to 100%

`echo 0 > pwm4`
worked flawlessly and set it to 0.
now
```
cat pwm4_enable 
1
```
is actually 1.

aleksamagicka, 2024-02-17:
> this succeeded I think? I did this, off / on cycle and did this again. I'm sorry I'm really new lmao

Yeah, on boot it loads the built-in one. `make dev` compiles and installs the github one (or, more specifically, whatever is in that directory).

> sorry for not using pwm1 as you told me but its the pump and I'd like to keep it running so I'm using another channel :D

That was just an example, use whichever channel you want. I just got tired of typing "pwmX, where X is ...".

pwm_enable sheds some light and proves what I thought:

```
 cat pwm4_enable 
2
```

So it's using PID mode, and the driver in the kernel is (right now) only setting the PWM value and _not_ setting the internal PWM controller value to the equivalent of `1`. In other words, when you just set pwm4 to whatever, it does not update the device to actually use that mode of control.

That's a fix for the kernel, then, as I suspected. Thanks for going through this with me, much appreciated.

nyllewin, 2024-02-17:
Thank you! Can I somehow just .. run the github version of the kernel driver for the time being until things are sorted or would it be more wise to just to default to direct access for now?
It would honestly just be fine to default to direct access I think, I dont use much.. maybe I can make something up with delta from the sensors and set speeds automatically accordingly with direct-access flag once every 30 seconds or so.
Could just run that until kernel driver update I guess.

aleksamagicka, 2024-02-17:
Yes, you can run the github version just fine. Or just use direct access, it doesn't matter to the device. The outcome is the same.

> scrap that I just realized that its the same problem because we did this with the git version, right. ok Nevermind!

Nope. I told you to try the github version so we get access to pwm_enable to read it out; if you use it, liquidctl (and everything else) will work just fine because it knows how to set _both_ pwmX_enable and pwmX. (And the driver kind of forces you to do the same, when done manually.)

So long story short, either use the github driver until a future kernel update arrives or use direct access.

jonasmalacofilho, 2024-02-17:
@aleksamagicka, should we maybe revert[^1] #670? It turns out that `pwmX_enable` is actually required, even if only as a proxy for the kernel driver being functional enough for that operation be done through hwmon instead of direct access.

[^1]: Maybe in addition to the `git revert` we should also leave a comment in the code explaining in more detail why the check is necessary, when in theory it generally shouldn't be.

aleksamagicka, 2024-02-17:
I've just sent a fix for the in-kernel driver to LKML, so that should end up in stable trees. That fixes it up so that just setting pwmX does the work of setting both pwmX_enable and pwmX values, since the in-kernel one only has pwmX. The github one has both, which makes it obvious that you have to set them both to see an effect.

Then it boils down to: do we require the kernel to be up to date, since we're accessing its subsystem? If we revert, there will still be plenty of versions that only have pwmX still, even after one day I send in pwmX_enable.

Or, let's put it this way: say that only pwmX existed, and that it was broken in-tree. We'd be forced to say that you have to be be up to date, right?

As for adding a comment, I'm all for it. Drivers develop over time and differences should be noted.

aleksamagicka, 2024-02-17:
Looks like I commented a minute too soon. Guenter rejected the patch, so that leaves us with indeed reverting #670. I'll see if I can create pwm_enable patches for the in-tree driver tomorrow.