liquidctl/liquidctl #674: Arctic Liquid Freezer III/II

gardotd426, 2024-02-22:
### Device type

AIO Liquid Cooler

### Product page

https://www.arctic.de/us/Liquid-Freezer-III-Series

### First-party software

There isn't one, they recommend bios control for pump and fans. 

### Physical connection

No USB, seems like a single 4-pin header

### Connection protocol

other?

### Additional information about the device

I couldn't find any issues asking about the liquid freezer, and as you may know, Gamers Nexus recently named the new Liquid Freezer III as the best AIO they've ever tested. So, since I'm looking to upgrade from a 5900X to a 7950X/X3D, and their temps run way higher, I'm thinking of moving from my iCUE R

gardotd426, 2024-02-22:
From the Amazon page, here's the connector layout it seems:

[image]

aleksamagicka, 2024-02-22:
> No USB, seems like a single 4-pin header

That's a standard PWM connector, usually controlled by a Super I/O chip on the mobo. On Linux, drivers for those are in hwmon (and there's tens of models, variations, manufacturer specific variations, magic constants and offsets in there) and on Windows there's support for them in [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/SuperIOHardware.cs).

I don't think it makes sense to even try and support those chips in liquidctl for many reasons. Perhaps a GUI for hwmon would be helpful instead.

jonasmalacofilho, 2024-02-23:
> I don't think it makes sense to even try and support those chips in liquidctl for many reasons.

While this cooler seems nice, I agree that we don't want to go into (directly) supporting Super I/O chips in liquidctl.

Thus, closing this as out of scope.

closed 2024-02-23 by jonasmalacofilho

gardotd426, 2024-02-25:
>I don't think it makes sense to even try and support those chips in liquidctl for many reasons. Perhaps a GUI for hwmon would be helpful instead.

So you're saying that I could control the fans/pump with CoolerControl's hwmon support? If that's the case then yeah, you're right I see no need to add support to liquidctl.

marcandrebenoit, 2024-03-28:
I have an Arctic 240 Liquid Freezer 3 on my computer running an AMD 7700x on a x670e mobo. I'm running on 22.04.4.LTS.
I have the split connector installed and I use [Freon ](https://extensions.gnome.org/extension/841/freon/) for all other device monitoring.
Where should I start to make this work on liquidctl ?
I understand the issue is presently closed, but i'd like to re-open it and be given instructions to help identify/add this hardware as supported.

aleksamagicka, 2024-03-28:
This AIO is controlled through the motherboard Super I/O chips, as it has PWM connectors. It's out of scope for liquidctl to support those chips. On Linux, that support is provided by the kernel itself.