liquidctl/liquidctl #675: Input power curve for HX1500i is absurd

yshui, 2024-02-23:
### Describe the bug

I know the comment says the curve is not accurate for 2~10%. But it's much worse than that. For 230V input voltage, if I plugin in output power of 750W, I got input power of 39866.7W (!). An efficiency of 1.88% !

### Commands executed

`liquidctl status`

### Output of all relevant commands with `--debug` flag

N/A

### Affected device

Corsair HX1500i

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

NixOS

### Installation method

nixpkgs

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.7.5-lqx1-x86_64-with-glibc2.38)

yshui, 2024-02-23:
Plot of the power curve for 230V

[image]

jonasmalacofilho, 2024-02-23:
Yeap, that's wrong.

Can you give #676 a try? I recomputed the curves and I think they're sensible now.

For reference, here're the new input power and efficiency curves for the HX1500i at 230 V:

[image]

yshui, 2024-02-23:
Looks much more reasonable now. Thanks!

commit 5f01650f (2024-02-23): corsair_hid_psu: fix input power curves for HX1000i (2022) and HX1500i

Previously the curves were incorrectly computed in normalized mode,
while liquidctl expects them to yield absolute input power values.

Fixed by regenerating the curves with [`interpolate`] and suitable
`--rated` values.

[`interpolate`]: https://github.com/liquidctl/collected-device-data/blob/master/psu-efficiency/interpolate

Fixes: 3e3f36bd1e4f017d5e36f79a7dcdd8eb8fec46de
Fixes: #675
Related: https://github.com/liquidctl/collected-device-data/commit/4da9df3cfb7dc09831a1d3a6e3d86fb3b16ae92b
PR #676 (2024-02-23): corsair_hid_psu: fix input power curves for HX1000i (2022) and HX1500i
Previously the curves were incorrectly computed in normalized mode, while liquidctl expects them to yield absolute input power values.

Fixed by regenerating the curves with [`interpolate`](https://github.com/liquidctl/collected-device-data/blob/master/psu-efficiency/interpolate) and suitable `--rated` values.

Fixes: 3e3f36bd1e4f017d5e36f79a7dcdd8eb8fec46de
Fixes: #675
Related: https://github.com/liquidctl/collected-device-data/commit/4da9df3cfb7dc09831a1d3a6e3d86fb3b16ae92b

---

Checklist:

- [x] Adhere to the [developme [... cut]
commit f20c4f0d (2024-02-23): corsair_hid_psu: fix input power curves for HX1000i (2022) and HX1500i

Previously the curves were incorrectly computed in normalized mode,
while liquidctl expects them to yield absolute input power values.

Fixed by regenerating the curves with [`interpolate`] and suitable
`--rated` values.

[`interpolate`]: https://github.com/liquidctl/collected-device-data/blob/master/psu-efficiency/interpolate

Fixes: 3e3f36bd1e4f017d5e36f79a7dcdd8eb8fec46de
Fixes: #675
Related: https://github.com/liquidctl/collected-device-data/commit/4da9df3cfb7dc09831a1d3a6e3d86fb3b16ae92b
closed 2024-02-23 by jonasmalacofilho