liquidctl/liquidctl #677: ASUS ROG RYUJIN III 360 ARGB

MiahNelah, 2024-02-23:
### Device type

AIO liquid cooler

### Product page

https://rog.asus.com/cooling/cpu-liquid-coolers/rog-ryujin/rog-ryujin-iii-360-argb/

### First-party software

Armory Crate

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of embedded LEDs

### Physical connection

USB header

### Connection protocol

USB

### Additional information about the device

This AIO is seen as ASUS ROG Ryujin II by system.

```
~ lsusb -tv
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 013: Dev 009, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 001: Dev 011, If 0, Class=Vendor Specific Class, Driver=[none], 480M
            ID 0b05:1988 ASUSTek Computer, Inc. 
        |__ Port 001: Dev 011, If 1, Class=Human Interface Device, Driver=usbhid, 480M
            ID 0b05:1988 ASUSTek Computer, Inc. 
```

```
~ lsusb -d 0b05:1988 -v
Bus 001 Device 011: ID 0b05:1988 ASUSTek Computer, Inc. ROG RYUJIN II
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1988 ROG RYUJIN II
  bcdDevice            2.00
  iManufacturer           1 AsusTek Computer Inc.
  iProduct                2 ROG RYUJIN II
  iSerial                 3 377238833430
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      51
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic

aleksamagicka, 2024-02-23:
[Indeed](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/asus_ryujin.py#L44) the same ID as Ryujin II by ID. Support for Ryujin II is in Git, can you [try it](https://github.com/liquidctl/liquidctl?tab=readme-ov-file#installing-from-pypi-or-github) and see what happens? AFAICS from the docs, Ryujin 3 does not have a separate controller box, so things may be different.

MiahNelah, 2024-02-28:
Since Asus Ryujin II 360 support have been merged, I pulled last commit. Here is some test result.

Device is found as Asus Ryujin II 360 as expected.
```
$ ./liquidctl list --verbose
Device #0: ASUS Ryujin II 360
├── Vendor ID: 0x0b05
├── Product ID: 0x1988
├── Release number: 0x0200
├── Serial number: 377238833430
├── Bus: hid
├── Address: /dev/hidraw3
└── Driver: AsusRyujin
```

I can get temperature, settings and speed, which is already awesome !
```
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      22.1  °C
├── Pump duty                100  %
├── Pump speed              2700  rpm
├── Pump fan duty             30  %
├── Pump fan speed          1890  rpm
├── External fan duty         63  %
├── External fan 1 speed    1350  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm
```

I tested to change pump speed, and it seems to work: 
```
$ ./liquidctl --serial 377238833430 set pump speed 99
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      22.2  °C
├── Pump duty                 99  %
├── Pump speed              2730  rpm
├── Pump fan duty             30  %
├── Pump fan speed          1920  rpm
├── External fan duty         63  %
├── External fan 1 speed    1380  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm
```

Speed change is not only visual, it looks it's correctly applied : fan speed change accordingly. Here is a test with pump speed at 75%, fan are slower than with 100% speed.
```
$ ./liquidctl --serial 377238833430 set pump speed 75
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      22.1  °C
├── Pump duty                 75  %
├── Pump speed              2220  rpm
├── Pump fan duty             30  %
├── Pump fan speed          1920  rpm
├── External fan duty         63  %
├── External fan 1 speed    1380  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm
```

So, except that device's label is wrong, everything looks good !

aleksamagicka, 2024-02-28:
Thanks! What about pump fan and external fans, are they controllable? AFAICS the device does not have connectors for the external fans?

MiahNelah, 2024-02-28:
I used Asus Ryujin II 360 docs to get channels name. Everything looks controllable like on Asus Ryujin II 360.

Pump fan speed test :
```
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      24.2  °C
├── Pump duty                100  %
├── Pump speed              2700  rpm
├── Pump fan duty             30  %
├── Pump fan speed          2010  rpm
├── External fan duty         63  %
├── External fan 1 speed    1380  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm

$ ./liquidctl --serial 377238833430 set pump-fan speed 40
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      24.1  °C
├── Pump duty                100  %
├── Pump speed              2700  rpm
├── Pump fan duty             40  %
├── Pump fan speed          2340  rpm
├── External fan duty         63  %
├── External fan 1 speed    1380  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm
```

External fan speed test :
```
 ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      23.5  °C
├── Pump duty                100  %
├── Pump speed              2730  rpm
├── Pump fan duty             30  %
├── Pump fan speed          1920  rpm
├── External fan duty         63  %
├── External fan 1 speed    1260  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm

$ ./liquidctl --serial 377238833430 set external-fans speed 73
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      23.5  °C
├── Pump duty                100  %
├── Pump speed              2700  rpm
├── Pump fan duty             30  %
├── Pump fan speed          1920  rpm
├── External fan duty         73  %
├── External fan 1 speed    1380  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm
```

"Every fans" speed test :
```
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      22.9  °C
├── Pump duty                100  %
├── Pump speed              2700  rpm
├── Pump fan duty             30  %
├── Pump fan speed          1920  rpm
├── External fan duty         63  %
├── External fan 1 speed    1500  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm

$ ./liquidctl --serial 377238833430 set fans speed 50
$ ./liquidctl --serial 377238833430 status
ASUS Ryujin II 360
├── Liquid temperature      22.8  °C
├── Pump duty                100  %
├── Pump speed              2700  rpm
├── Pump fan duty             50  %
├── Pump fan speed          1950  rpm
├── External fan duty         50  %
├── External fan 1 speed    1380  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm
```

I don't really know why some fans are not displayed. I'm pretty new to Linux and all my fans are not detected at all. I still digging this issue.

aleksamagicka, 2024-02-28:
These seem stuck:

```
├── External fan 1 speed    1380  rpm
├── External fan 2 speed       0  rpm
├── External fan 3 speed       0  rpm
└── External fan 4 speed       0  rpm
```

How are external fans connected to the AIO? I can't really discern from the quick guide on ASUS' website.

> I don't really know why some fans are not displayed. I'm pretty new to Linux and all my fans are not detected at all. I still digging this issue.

Do you mean those above or those generally on your system? If the latter, try running `sensors` if you have `lm-sensors` installed.

MiahNelah, 2024-02-28:
> These seem stuck at 1380 (and zeroes) more often than not:
> 
> ```
> ├── External fan 1 speed    1380  rpm
> ├── External fan 2 speed       0  rpm
> ├── External fan 3 speed       0  rpm
> └── External fan 4 speed       0  rpm
> ```
External fan 1 isn't stuck to 1380, it change a little during my tests: on my first test, external fan 1 was as 1350. Other external fans do stuck to 0 however.

> How are external fans connected to the AIO? I can't really discern from the quick guide on ASUS' website.

From manual, pump is plug to CPU_FAN or AIO_PUMP or W_PUMP pin. External fans are plug to regular FAN 4-pins. I may take a closer look this weekend to I plugged all of this, I may have fucked something.
Can all external fans be controlled like one ?
 
> Do you mean those above or those generally on your system? If the latter, try running `sensors` if you have `lm-sensors` installed.
`sensors` do see my PWM fans system-wide. I tried `sensors-detect` and `pwmconfig` but without success. However, I do see fans in BIOS, and can control them in PWM mode.
```
$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +22.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +22.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +17.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +18.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +19.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 24:       +19.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +18.0°C  (high = +80.0°C, crit = +100.0°C)
Core 32:       +23.0°C  (high = +80.0°C, crit = +100.0°C)
Core 33:       +23.0°C  (high = +80.0°C, crit = +100.0°C)
Core 34:       +23.0°C  (high = +80.0°C, crit = +100.0°C)
Core 35:       +22.0°C  (high = +80.0°C, crit = +100.0°C)
Core 36:       +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 37:       +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 38:       +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 39:       +21.0°C  (high = +80.0°C, crit = +100.0°C)

nvme-pci-0200
Adapter: PCI adapter
Composite:    +25.9°C  (low  = -20.1°C, high = +83.8°C)
                       (crit = +88.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +27.9°C  (low  = -60.1°C, high = +89.8°C)
                       (crit = +94.8°C)
```

aleksamagicka, 2024-02-28:
> External fan 1 isn't stuck to 1380, it change a little during my tests: on my first test, external fan 1 was as 1350

I said that because I noticed these:

```
├── External fan duty         73  %
├── External fan 1 speed    1380  rpm
```
and 

```
├── External fan duty         50  %
├── External fan 1 speed    1380  rpm
```

Perhaps it just needed time to update. If those work, that's great.

-----

> From manual, pump is plug to CPU_FAN or AIO_PUMP or W_PUMP pin. External fans are plug to regular FAN 4-pins. I may take a closer look this weekend to I plugged all of this, I may have fucked something.

Pump is plugged in _probably_ to silence the BIOS warnings for having no fans? I don't see another reason right now, if it's controlled through USB... Or maybe it's just for power.

> Can all external fans be controlled like one ?

Depends on how you connect them, you can use PWM splitter cables and chain them to one PWM output from anywhere.

Either way, AFAIU your AIO works well with liquidctl, and the only thing is that there are three superflous readings for external fans 2, 3 and 4?

MiahNelah, 2024-02-28:
> I said that because I noticed these:
>
> ```
> ├── External fan duty         73  %
> ├── External fan 1 speed    1380  rpm
> ```
>
> and
>
> ```
> ├── External fan duty         50  %
> ├── External fan 1 speed [...]

Yes, I run status command less than a second after changing speed, fan need alittle more time to reduce speed.

> Either way, AFAIU your AIO works well with liquidctl, and the only thing is that there are three superflous readings for external fans 2, 3 and 4?

That's it, have temperature and global pump+main fan speed is already great! I'll dig more on my side about fans and update this thread if I discover something.

MiahNelah, 2024-02-28:
I'm the biggest idiot! Of course my device is seen as Asus Ryujin II since IT IS an Asus Ryujin II and not a Ryuijin III !
I'm so sorry for this lost time for nothing, I should have check this before everything.

MiahNelah, 2024-02-28:
There is, but all 3 fans are combined into one PWM, that's why all three fans are controllable and used as a whole.

aleksamagicka, 2024-02-28:
Ah, then that's indeed a RYUJIN II.

@jonasmalacofilho, can this be closed as-is, or should I change the title and/or add a label? (I don't see a proper label in the list, though...)

davija, 2024-03-01:
I recently purchased an ASUS ROG RYUJIN III 360 (not the ARGB version) and can confirm that it is not currently detected.

This one connects via USB and has a 4-pin header for MB CPU-Fan / AIO header.

`lsusb -d 0b05:1aa2 -v`
```

Bus 001 Device 010: ID 0b05:1aa2 ASUSTek Computer, Inc. ROG RYUJIN III
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1aa2 
  bcdDevice            2.00
  iManufacturer           1 AsusTek Computer Inc.
  iProduct                2 ROG RYUJIN III
  iSerial                 3 305A354F3332
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      51
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

```

aleksamagicka, 2024-03-01:
@davija, [this](https://github.com/aleksamagicka/liquidctl-aqc/commit/6faf1f12e70dcf7aa8d1bc1168bc299cc9fc491b) is all you need to do to get liquidctl to see it as a RYUJIN II. Try it out and report back, perhaps it's similar. (If you need help applying the commit/change, I'll give instructions.)

Mardoxx, 2024-07-01:
```
```python
import math, usb

# ID 0b05:1ada ASUSTek Computer, Inc. ROG RYUJIN III WHITE EDITION
VENDOR_ID = 0x0B05
PRODUCT_ID = 0x1ADA

dev = usb.core.find(idVendor=VENDOR_ID, idProduct=PRODUCT_ID)

# Ensure the device is found
if dev is None:
    raise ValueError('Device not found')

# https://elixir.bootlin.com/linux/latest/source/drivers/usb/core/devio.c#L1558
# the usbfs driver in Linux has a limitation that if any interface of the device is claimed by a kernel
# driver no interface can be claimed by a libusb process. So you have to iterate through all of the
# interfaces and detach the kernel driver on all of them
for cfg in dev:
    for intf in cfg:
        if dev.is_kernel_driver_active(intf.bInterfaceNumber):
            print(f"Detaching kernel driver from interface {intf.bInterfaceNumber}")
            try:
                dev.detach_kernel_driver(intf.bInterfaceNumber)
            except usb.core.USBError as e:
                print(f"Could not detach kernel driver from interface {intf.bInterfaceNumber}: {str(e)}")

def write_frame_buffer(dev, bgr_buff):
    endpoint = 0x01

    BUFF_LEN = 320 * 240 * 3

    buff = (bgr_buff[:BUFF_LEN] + [0x00] * BUFF_LEN)[:BUFF_LEN]

    try:
        dev.write(endpoint, buff)
    except usb.core.USBError as e:
        raise ValueError(f"Could not write to endpoint {endpoint}: {str(e)}")

def flush_frame_buffer(dev):
    endpoint = 0x02

    CMD_PAGE = 0xec
    CMD_LEN = 64

    # Additional command data to be sent
    cmds = [
        [0x7f, 0x03, 0x00, 0x84, 0x03], # flush
        #[0xd0], # ??
        #[0x9a], # ??
        #[0x1a, 0x01, 0x17, 0x11] # ??
    ]

    for cmd in cmds:
        cmd_data = [CMD_PAGE] + (cmd[:CMD_LEN] + [0x00] * CMD_LEN)[:CMD_LEN]
        try:
            dev.write(endpoint, cmd_data)
        except usb.core.USBError as e:
            raise ValueError(f"Could not write to endpoint {endpoint}: {str(e)}")

def frame_generator():
    width, height = 320, 240
    t = 0

    while True:
        frame = [[[0, 0, 0] for _ in range(width)] for _ in range(height)]

        for y in range(height):
            for x in range(width):
                value = (
                    math.sin(x * 0.01 + t) +
                    math.sin(y * 0.012 + t) +
                    math.sin((x + y) * 0.015 + t) +
                    math.cos((x - y) * 0.018 + t) +
                    math.sin(math.sqrt(x**2 + y**2) * 0.008 + t)
                )
                normalized_value = (value + 5) * math.pi * 10
                
                r = int((1 + math.sin(normalized_value * 0.1 + t)) * 127.5)
                g = int((1 + math.sin(normalized_value * 0.1 + t + 2)) * 127.5)
                b = int((1 + math.sin(normalized_value * 0.1 + t + 4)) * 127.5)
                
                frame[y][x] = [b, g, r]

        t += 0.05

        # Flatten the frame
        flat_frame = [color for row in frame for pixel in row for color in pixel]
        yield flat_frame

def main_loop(dev):
    frame_gen = frame_generator()
    
    while True:
        write_frame_buffer(dev, next(frame_gen))
        flush_frame_buffer(dev)

if __name__ == "__main__":
    main_loop(dev)
```

This is what I captured from AIDA. It sends the whole frame, runs the flush(?) command and these other two. Not sure. Someone with some good reskills can have a look. This is the best I could do right now.

There's some comment which must switch it into this mode, but I haven't discovered it yet.

Mardoxx, 2024-07-01:
It needs to be in mode `0x20` which you do by sending `[0xec, 0x51, 0x20]`

Mardoxx, 2024-07-06:
`[0x1a, 0x01, 0xXX, 0xYY]`
`XX` is pump duty from `0x00` to `0x64`
`YY` is fan duty from `0x00` to `0x64`

Mardoxx, 2024-07-06:
```python
def ensure_length(data, length):
    if isinstance(data, str):
        exploded_data = [*data]
    else:
        exploded_data = data

    padded = exploded_data[:length] + [0x00] * max(0, length - len(exploded_data))
    
    return [ord(x) if isinstance(x, str) else x for x in padded[:length]]

def test_cmds(dev):
    CMD_PAGE = 0xec
    CMD_LEN = 64

    endpoint = 0x02

    ###############
    # SET MODE TO 0x21

    HW_INFO_MODE = 0x21

    CMD_SET_MODE = [0x51, HW_INFO_MODE]

    ###############

    BG_STYLE = 0x02 # 0x00 galactic, 0x01 cyberpunk, 0x02 custom

    NUM_INFOS = 0x02 # 0x00 1, 0x01 2, 0x02 3

    bg_col = [0xff, 0x00, 0xff]
    txt_col0 = [0x00, 0xff, 0xff]
    txt_col1 = [0x00, 0xff, 0x00]
    txt_col2 = [0xff, 0xff, 0x00]
    txt_col3 = [0x00, 0x00, 0x00]

    CMD_SETUP_COLOURS = [
        0x52, BG_STYLE, NUM_INFOS, 0x03, 0x00,
        *bg_col, 0xFF,
        *txt_col0, 0xFF,
        *txt_col1, 0xFF,
        *txt_col2, 0xFF,
        *txt_col3, 0xFF
    ]

    TXT_LEN = 17
    VAL_LEN = 11

    txt = "12345678901234567"
    val = "ABCDEFGHIJK"

    UNIT = "oC"

    unit = []
    match UNIT:
        case None:
            ...
        case "oC":
            unit = [0xe2, 0x84, 0x83]
        case "V":
            unit = [0xe2, 0x86, 0x8a]
        case "FPS":
            unit = [0xe2, 0x86, 0x8b]
        case "RPM":
            unit = [0xe2, 0x86, 0x8c]
        case "L/M":
            unit = [0xe2, 0x86, 0x8d]
        case "W":
            unit = [0xe2, 0x86, 0x8f]            
        case "Hz":
            unit = [0xe3, 0x8e, 0x90]
        case "KHz":
            unit = [0xe3, 0x8e, 0x91]
        case "MHz":
            unit = [0xe3, 0x8e, 0x92]
        case "GHz":
            unit = [0xe3, 0x8e, 0x93]

    val_len = VAL_LEN - len(unit)

    ###############

    CMD_SET_TEXT = [
        [
            0x53, LINE_NO,
            *ensure_length(txt, TXT_LEN),
            0x00,
            *ensure_length(val, val_len),
            *unit
        ] for LINE_NO in range(NUM_INFOS + 1)
    ]

    cmds = [
        CMD_SET_MODE,
        CMD_SETUP_COLOURS,
        *CMD_SET_TEXT
    ]
    
    for cmd in cmds:
        cmd_data = [CMD_PAGE] + ensure_length(cmd, CMD_LEN)
        try:
            dev.write(endpoint, cmd_data)
        except usb.core.USBError as e:
            raise ValueError(f"Could not write to endpoint {endpoint}: {str(e)}")
```

Not sure why 5 colours are passed here?

Mardoxx, 2024-07-07:
Image data is uploaded like this:

```
['0x71', '0x01', '0x01'] # unk
['0x72', '0x01', '0x02', '0x09'] # unk

['0x73', '0x01'] # Begin xfer mode 
['0x7f', '0x02', *LITTLE_ENDIAN_FILE_SIZE_BITES] ## Notify of size e.g. '0xf3', '0x7c', '0xa0' for 10517747 bytes
# Bulk xfer chunked and padded 4096
['0x73', '0xff'] # Notify end xfer mode
```

It's mega slow, ~4kB/s. There'll be a way to upload to, delete, and display gif slots, I'll work that out in a bit.

None of the slideshow features are done on the device, they are commands sent from armoury crate.

The clock mode `0x08` seems to use a real clock, perhaps bios.

Mardoxx, 2024-07-08:
`0xDC` gets device info
`0x5C` sets device info

```
ORIENTATION = 0x00 # 0x00 landscape, 0x01 portrait
BRIGHTNESS = 0x00 # 0x00-0x64
[0x5c, 0x01, 0x00, 0x64, ORIENTATION, 0x00, BRIGHTNESS, 0x14, 0x00, 0x04]
```

Mardoxx, 2024-07-08:
It would be good if anyone who has NOT updated the firmware on the device could capture the packets sent to the device, 1. prior, 2. during transfer, 3. after transfer, 4. after entering armoury crate, and clicking settings

Mardoxx, 2024-07-08:
Clock mode gets really weird... clearly whomever implemented this got hired from a leet-code interview.... roll_eyes_emoji
```

def convert(in_num: int) -> int:
    """
    returns corresponding hexadecimal value, represented as an integer
    assert convert(15) == 0x15
    """
    return (in_num // 10 << 4) + in_num % 1

HR_FMT = 0x01 # 0x00 (actually SHOW AM/PM)
AM_PM = 0x00 # 0x01 PM
CLOCK_BYTES = [0x08, 0x00, HR_FMT]

HH = convert(12)
MM = convert(13)
SS = convert(14)
TIME_BYTES = [0x10, 0x11, 0x12, AM_PM] # These are 

[0x11, 0xc4, 0x07] + CLOCK_BYTES + HH + MM + SS + [0x01],
[0x5d, 0x00, 0x01] + CLOCK_BYTES + [0x05]
[0x51] + CLOCK_BYTES
```

LBlaboon, 2024-12-10:
> It would be good if anyone who has NOT updated the firmware on the device could capture the packets sent to the device, 1. prior, 2. during transfer, 3. after transfer and update, 4. after entering armoury crate, and clicking settings

New ROG Ryujin III owner here. I'm happy to help get this moved along if there's still interest. I haven't done anything with the official software yet so my device should still be in a pristine state as far as firmware goes (assuming they didn't ship it with an already-updated firmware).

I have a fair bit of experience with capturing network traffic on Linux but basically no experience with USB capture on Windows. Is there a recommended tool?

Mardoxx, 2024-12-10:
I debugged armoury crate before. You can stick a breakpoint just before it
executes the installer- it is all encrypted so you can find the windows
decrypt handler and bp there. Or I think there is a wininternals tool you
could maybe use to copy the files. I did the former.

Usb capture on windows you can use winpcap and usbcap? Or whatever the
plugin is called.

Let me write up how to do it. Perhaps we can get on discord or something.

LBlaboon, 2024-12-10:
Yea, found the in-repo guide like two minutes after making my previous comment :sweat_smile: 

I'll start messing around in the coming days.

Mardoxx, 2024-12-11:
Nice! Wireshark sorry. I can’t remember exactly what filters to use, there
is likely a good way to find the device channel but I did it through trial
and error. Then you can filter out the chatty packets. I can not remember
how I exported the entire packet since it truncates it. The chunk size is
specified though.

I would hold off on the firmware update. Once you update it you can’t go
back.

 I have not managed to find a way to re-request the binaries from Asus.
Every stage of armoury crate updates is encrypted. You can break it by
breakpointing on windows crypto apis to extract the keys, I didn’t manage
to find out how the updates are requested. Perhaps patching the aio dll to
return an earlier version number, not sure.

LBlaboon, 2024-12-13:
Small update: I've managed to MitM my connections to Asus's update servers so I can dump the packets and see what's going on. Upon first inspection all the endpoints appear to be authenticated, so fetching updates will be a bit more complicated than just "fetch this URL to retrieve an update/manifest". Gonna dig deeper over the weekend and try to forge my own update requests.

Mardoxx, 2024-12-13:
The auth to their update server contains a lot of fingerprinting - I
couldn’t work it out and gave up.

The payloads are encrypted with hs256(?) the shared key is in one of the
binaries, it’s all obfuscated pointlessly since you can attach a debugger
and breakpoint at windows crypto api to extract them.

I am a novice at all this, but conceivably you could modify the executable
to prevent it from running the installer and deleting it after encryption.

LBlaboon, 2024-12-13:
From what I'm seeing the server auth is based on JWT tokens which can be obtained by POSTing to https://rog-live-service.asus.com/developer/api-token-auth/ with some kind of initial token (not sure where that comes from). The resulting token is then passed in an authorization header for requests to the other endpoints. Still need to capture more traffic though as I've only got dumps of a few endpoints that don't paint the full picture.

LBlaboon, 2024-12-14:
Upon further digging, the token you need to initially pass to the endpoint mentioned above is embedded within `ArmouryCrate.dll` in the app files with the identifier `FormalTokenP`. It is unique per installation and presumably gets generated at install time. However it's getting generated, it's not using the rog-live-service API but something different, so that will need investigation, or we could just steal one from an existing installation. The JWT tokens returned by the API are fairly short-lived but the app's formal token seems to be valid Forever™

Using my installation's token, I managed to forge some requests to the API to retrieve a JSON manifest containing the latest firmware version along with a download URL from their CDN and checksums, and parameters for launching the updater exe. The CDN is unauthenticated and I was able to download and extract the latest FW package which contains the updater exe and firmware blob. So next steps will be rev-eng'ing the executable/USB sniffing to see how the blob is being sent to the device.

This is the URL I got for the latest firmware btw: https://dlcdn-rogboxbu2.asus.com/pub/ASUS/APService/Gaming/SYS/ROGS/15817-O4D97J-b016900f8204aa08869b7e797719ed50.zip

Mardoxx, 2024-12-14:
That’s fantastic. The firmware update procedure is in AIOsdk.dll(?) I think
I documented it. It puts the device in fw update mode (similar way you put
it in framebuffer mode) sends the data chunked, then puts it back into
normal mode?

Mardoxx, 2024-12-14:
Commands are received like this:
```py
CMD_REQ_KEY = IN_BUFF[1]
CMD_RESP_KEY = OUT_BUFF[1]

correct_response == (CMD_REQ_KEY == CMD_RESP_KEY || CMD_REQ_KEY == (CMD_RESP_KEY | 0x80))
status = OUT_BUFF[2]

success = correct_response && status == 0
```

The pump is an Asetek Gen8, as are almost all AiOs on the market.

Looking at the strings inside that binary

And the LCDs are 
TM035KDH03-ILI9322
TM035KDGP01-ST7272A
And it uses an ILI9322 SoC for the LCD driver

I don't fancy taking mine to bits to verify this.

Mardoxx, 2024-12-14:
The Ryujin III has a 640x480 LCD but that firmware only has mention of 320x240 and 320x480 parts. Are you sure it is the III not the 2? The installer only has reference to Ryujin 2 also.

LBlaboon, 2024-12-14:
Interesting....the API response very clearly labels it for the Ryujin III. Here's the full response I got from the API: https://gist.github.com/LBlaboon/dc453967a271c9b8ab0daf0b49de94d5

Mardoxx, 2024-12-14:
What is in the payload to request that? If it takes a device ID, I believe these are different for the editions. I have a white one if we want to try requesting the firmware for that?

LBlaboon, 2024-12-14:
There's a few API requests needed to get to that. First request returns a list of all downloads ("profiles") for the requested device, for that I gave it "0B051AA2" which I found just from traffic sniffing (not sure what the significance of that string is, model number maybe?). Then from the resulting list you request the details for each individual package. In my case there were 7 packages returned in the initial list, but only one of them was for firmware (the results for which are in my prior gist). The rest all appear to be various components of Armoury Crate itself.

LBlaboon, 2024-12-14:
I did find it suspicious that the modification date for that firmware is 2023 despite the Ryujin III being from this year.

Mardoxx, 2024-12-14:
`0B05 1AA2` this is the vendor and product id, I can't remember where I saw the list of them - I think in the javascript of the UI.

```py
# ID 0b05:1ada ASUSTek Computer, Inc. ROG RYUJIN III WHITE EDITION
VENDOR_ID = 0x0B05
PRODUCT_ID = 0x1ADA
``

Try with those?

LBlaboon, 2024-12-14:
Ah duh, I should have realized that. For some reason my tired self was counting the string as only 7 characters instead of 8 :sweat_smile: 

Here's the relevant result from feeding it your ID: https://gist.github.com/LBlaboon/04ce6f3127b66a8aba9848ef902c7ba2
It looks....similarly old.

I also found out that you can get a different list of results that contains _only_ the firmware package if you add one to the product ID (i.e. 1AA3 instead of 1AA2, or 1ADB instead of 1ADA). Still the same firmware though, but at least eliminates having to poll all the other items.

LBlaboon, 2024-12-14:
Here's the full un-curated list of results for my device: https://gist.github.com/LBlaboon/9fe64c585f08a2675a67ddd338bea153

Maybe one of the other results is also a firmware package in disguise?

LBlaboon, 2024-12-14:
One of those packages simply contains a JSON manifest referencing other profiles (different than the ones returned by the API, but the same types). Here's the API results for that list: https://gist.github.com/LBlaboon/a073bc11db76a8b3b6fa2023b71ce910

No idea what that's all about.

Mardoxx, 2024-12-14:
Nice one, thanks. Yeah they have the same digests. I guess the update procedure/command schema didn't change since the II, so no point re-building it. Interesitng how OP's III was reporting as a II.

The microcontroller it uses is a STM32.. something. The firmware isn't compressed and it seems ida/ghidra eat it up quite easily (cortex processor). There seems to be some slideshow modes that I haven't seen, I don't know where the built-in graphics are, I think they are mjpegs rather than gifs.

Renari, 2025-02-03:
> Nice one, thanks. Yeah they have the same digests. I guess the update procedure/command schema didn't change since the II, so no point re-building it. Interesitng how OP's III was reporting as a II.

OP actually had a 2, the posted that here: https://github.com/liquidctl/liquidctl/issues/677#issuecomment-1969191598

Renari, 2025-02-25:
```
❯ lsusb -d 0b05:1ade -v

Bus 003 Device 009: ID 0b05:1ade ASUSTek Computer, Inc. ROG RYUJIN III EVA EDITION
Couldn't open device, some information will be missing
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 [unknown]
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1ade ROG RYUJIN III EVA EDITION
  bcdDevice            2.00
  iManufacturer           1 AsusTek Computer Inc.
  iProduct                2 ROG RYUJIN III EVA EDITION
  iSerial                 3 365832723432
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      51
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```
Unlike OP it does show up as `ROG RYUJIN III` for me.

Jozefik, 2025-03-06:
Hi guys i'm not sure if this is correct place to ask but i will try it. I have Asus ROG RYUJIN III ARGB stuck on firwmare 0107. Armory crate prompts for 0108 but it does not finish correctly. It will spin the cooler says it is updated but then when i check i get again propmted to update to 0108. Can someone help me how to upgrade it ? I have sniffed the package it downloaded in ProgramFiles\ASUS\RSLDownload\ROG RYUJIN III but when i run the .exe with parameter for .bin file it does the same again. Can i somehow troubleshoot it ? Thanks.

Renari, 2025-03-06:
> Hi guys i'm not sure if this is correct place to ask but i will try it. I have Asus ROG RYUJIN III ARGB stuck on firwmare 0107. Armory crate prompts for 0108 but it does not finish correctly. It [...]

I don't know why you would think this is the correct place, go contact ASUS support.

jonasmalacofilho, 2025-03-07:
@Renari, that was uncalled for. Please don't do it again.

@Jozefik this isn't the correct place to discuss it. You can try our Discord, or maybe OpenRGB's. But bear in mind that what you're asking about is only tangentially related to the stuff we're doing, so please try to find the lest intrusive place to ask.

---

I'll hide both comments, as well as this one, as off-topic.

BlakeBarrett, 2025-05-23:
I have an ASUS ROG Ryujin 3 240 (USB IDs: 0b05:1aa2) on a fresh Kubuntu 24.04 install. I’ve tried both the latest apt and pip/venv versions of liquidctl, but the device is not detected. OpenRGB also does not see it.

Here’s the relevant dmesg output:
```
usb 1-8: New USB device found, idVendor=0b05, idProduct=1aa2, bcdDevice= 2.00
usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-8: Product: ROG RYUJIN III
usb 1-8: Manufacturer: AsusTek Computer Inc.
```

I’m happy to help test any scripts, branches, or patches to get this device working. Please let me know if you need additional logs, USB dumps, or other info.

gsornsen, 2025-10-31:
@mzonski - your PR is cleaner, maybe I'll keep mine open and stack the _MATCHES additional and piggy-back off of your tests after yours is merged.

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.