liquidctl/liquidctl #679: Unable to control the Pump on my H150i cappllix LCD

TheRandomizer23, 2024-02-27:
### Describe the bug

Have a script setup like this:

cd /D "D:\Utilities\LiquidCTL\venv\Scripts"
call activate.bat
liquidctl initialize --match st
liquidctl --match st set pump speed 100

Keep getting the error:
ERROR: Corsair Commander ST (experimental): unexpected error: AssertionError('response does not match command')

Was working for some time, then I messed around with the fan curves in icue then exited out of icue as I just want it to follow hardware control and now this isnt working, not sure what to do.

### Commands executed

cd /D "D:\Utilities\LiquidCTL\venv\Scripts"
call activate.bat
liquidctl initialize --match st
liquidctl --match st set pump speed 100

### Output of all relevant commands with `--debug` flag

ERROR: Corsair Commander ST (experimental): unexpected error: AssertionError('response does not match command')

### Affected device

H150i Elite Capellix XT LCD, Commander CORE

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Win 10

### Installation method

pip install liquidctl

### Version of liquidctl

liquidctl v1.13.0 (Windows-10-10.0.19045-SP0)

Serphentas, 2024-04-30:
Are you able to run this command in debug mode and paste the output here ? Can you run the same commands from Linux (VM is OK) to see if behaviour is any different ? Thanks.

jonasmalacofilho, 2024-12-29:
From the limited information available, this appears to be related to #754 (and the issues it replaced). Closing as duplicate.

closed 2024-12-29 by jonasmalacofilho