liquidctl/liquidctl #680: Add support for MSI MPG Coreliquid K240 **V2**

gamer0mega, 2024-03-09:
### Device type

AIO liquid cooler

### Product page

https://www.msi.com/Liquid-Cooling/MPG-CORELIQUID-K240-V2

### First-party software

MSI Center/Coreliquid

### Physical connection

USB 2 header on the motherboard

### Connection protocol

USB

### Additional information about the device

```
> sudo lsusb -v
...
Bus 001 Device 012: ID 0db0:b130 Micro Star International MPG CORELIQUID
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0db0 Micro Star International
  idProduct          0xb130 
  bcdDevice            1.00
  iManufacturer           1 MSI
  iProduct                2 MPG CORELIQUID
  iSerial                 3 A02021031800
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     154
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
...
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic

sarzeaud, 2024-11-15:
It might be worth testing the insertion of this single line to the msi.py driver, in _MATCHES list (around line 320):

> (0x0DB0, 0xb130, "MSI MPG Coreliquid K240", {"fan_count": 5}),

sarzeaud, 2024-11-30:
Built my own driver for this, available here: https://github.com/sarzeaud/coreliquid_driver
Must be easy to adapt to your exact needs.

gamer0mega, 2024-12-04:
> Built my own driver for this, available here: https://github.com/sarzeaud/coreliquid_driver Must be easy to adapt to your exact needs.

Thanks, I'll try it out later

sarzeaud, 2024-12-04:
Before testing, do not forget to update the product ID on line 156 of my_msi_driver.c. You should put 0xb130 (MPG K240 version) instead of 0x6a05 (MEG S360 version). Some other adjustments may be needed if the message protocol is not the same. See here for MEG S360 protocol: https://github.com/liquidctl/liquidctl/blob/main/docs/developer/protocol/coreliquid.md

closed 2025-11-15 by jonasmalacofilho