liquidctl/liquidctl #681: 1.13.0: replace use `crcmod` by something else

kloczek, 2024-03-09:
### Describe the bug

Issues:
- `crcmod` is not maintained since 2010
- looks like only liquidctl uses `crcmod`
- there are some replacements in standard python library like `binascii.crc32` or `zlib.crc32`

### Commands executed

N/A

### Output of all relevant commands with `--debug` flag

N/A

### Affected device

N/A

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Linux x86/64

### Installation method

N/A

### Version of liquidctl

1.13.0

aleksamagicka, 2024-03-09:
For my reference: it's used for `crc-8` and `crc-16-usb` throughout the project.

> looks like only liquidctl uses crcmod

Anything to back this up?

kloczek, 2024-03-10:
> For my reference: it's used for crc-8 and crc-16-usb throughout the project.

So why not use one of the python standard libraries crc?
It does not look like use CRC in liquidctl requires fastest-possible-crc 🤔 

> > looks like only liquidctl uses crcmod
> 
> Anything to back this up?

Check any OS distribution.
I was wrong. It i used in two projects
```console
[tkloczko@domek fedora]$ rpm -qp --qf "%{NAME}: [%{REQUIRENAME} ]\n" */*rpm | grep 'python3dist(crcmod)'
liquidctl: (python3dist(tomli) if python3-devel < 3.11) pyproject-rpm-macros python3-devel python3-devel python3dist(colorlog) python3dist(crcmod) python3dist(docopt) python3dist(hidapi) python3dist(packaging) python3dist(pillow) python3dist(pip) python3dist(pytest) python3dist(pyusb) python3dist(setuptools) python3dist(setuptools-scm) python3dist(setuptools-scm) python3dist(setuptools-scm[toml]) python3dist(smbus) python3dist(tox) python3dist(tox-current-env) python3dist(wheel) rpmlib(CompressedFileNames) rpmlib(DynamicBuildRequires) rpmlib(FileDigests) rpmlib(RichDependencies) systemd-rpm-macros
python-gcsfs: ((python3dist(aiohttp) < 4~a0 or python3dist(aiohttp) > 4~a0) with (python3dist(aiohttp) < 4~a1 or python3dist(aiohttp) > 4~a1)) latexmk make pyproject-rpm-macros python3-devel python3-devel python3-sphinx-latex python3dist(crcmod) python3dist(decorator) python3dist(fsspec) python3dist(fusepy) python3dist(google-auth) python3dist(google-auth-oauthlib) python3dist(google-cloud-storage) python3dist(packaging) python3dist(pip) python3dist(pytest) python3dist(requests) python3dist(setuptools) python3dist(sphinx) python3dist(sphinx-rtd-theme) python3dist(vcrpy) python3dist(wheel) rpmlib(CompressedFileNames) rpmlib(DynamicBuildRequires) rpmlib(FileDigests) rpmlib(RichDependencies) rpmlib(TildeInVersions)
```

jonasmalacofilho, 2024-04-23:
Like Aleksa mentioned, we need CRC-8 (CRC-8/SMBUS) and CRC-16/USB, so there are no implementations from the standard library we can use.

While [`crcmod`](https://pypi.org/project/crcmod/) hasn't seen changes since 2010, it still works pretty well, and is the commonly used Python package for generic CRC computation by a very large margin ([5 M monthly downloads](https://pypistats.org/packages/crcmod) from PyPI).

There's also a new player, [`crc`](https://pypi.org/project/crc/), which seems to offer a more modern and purely-Python implementation ([85 K monthly downloads](https://pypistats.org/packages/crcmod)). I see that it generates and caches lookup tables, so it should be efficient enough for most use cases, and certainly for ours. But, in addition to being less commonly used, it's not yet packaged for Arch Linux (I haven't checked other distros).

For completeness, in the past we used our own CRC-8 implementation (that I had copied and then adapted from somewhere), but with the need for CRC-16/USB, I don't think this approach makes sense.

So, practically, I think we can choose between `crcmod` and `crc`. And _for now_ `crcmod` is simpler to include as dependency because it's already packaged everywhere. But _once_ that advantage no longer exists I think we should indeed move to the more modern and maintained alternative.

jonasmalacofilho, 2024-04-23:
Packing statuses according to Repology:

- `crcmod`: https://repology.org/project/python:crcmod/versions
- `crc`: https://repology.org/project/python:crc/versions