liquidctl/liquidctl #682: ModuleNotFoundError: No module named 'smbus'

liluzibird, 2024-03-16:
So for context, I ran liquidctl fine for like a year or two and no issues popped up. Recently, I accidentally deleted the /usr/lib/python.3.11/site-packages/ directory and a lot of stuff broke including protonvpn (fixed with reinstall and update), I tried to reinstall and update liquidctl, but it still gives me the same error. I have no clue why. I have python-smbus2, lm_sensors, and lib32-lm_sensors installed, so I should have all the required dependencies. I am using Arch Linux.

```
systemctl status liquidcfg.service
× liquidcfg.service - AIO startup service
     Loaded: loaded (/etc/systemd/system/liquidcfg.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Sat 2024-03-16 11:01:39 PDT; 7s ago
    Process: 849107 ExecStart=liquidctl initialize all (code=exited, status=1/FAILURE)
   Main PID: 849107 (code=exited, status=1/FAILURE)
        CPU: 44ms

Mar 16 11:01:39 arch liquidctl[849107]:   File "/usr/lib/python3.11/site-packages/liquidctl/driver/__init__.py",>
Mar 16 11:01:39 arch liquidctl[849107]:     from liquidctl.driver import ddr4
Mar 16 11:01:39 arch liquidctl[849107]:   File "/usr/lib/python3.11/site-packages/liquidctl/driver/ddr4.py", lin>
Mar 16 11:01:39 arch liquidctl[849107]:     from liquidctl.driver.smbus import SmbusDriver
Mar 16 11:01:39 arch liquidctl[849107]:   File "/usr/lib/python3.11/site-packages/liquidctl/driver/smbus.py", li>
Mar 16 11:01:39 arch liquidctl[849107]:     from smbus import SMBus
Mar 16 11:01:39 arch liquidctl[849107]: ModuleNotFoundError: No module named 'smbus'
Mar 16 11:01:39 arch systemd[1]: liquidcfg.service: Main process exited, code=exited, status=1/FAILURE
Mar 16 11:01:39 arch systemd[1]: liquidcfg.service: Failed with result 'exit-code'.
Mar 16 11:01:39 arch systemd[1]: Failed to start AIO startup service.
```

jonasmalacofilho, 2024-03-16:
The `smbus` Python package is part of the `i2c-tools` Arch package.

liluzibird, 2024-03-16:
Yeah. Then why isn't it working. Makes no sense to me

jonasmalacofilho, 2024-03-16:
You should try re-installing `i2c-tools`, since it appears to still be in a corrupted state after your problem with `/usr/.../site-packages/`.

liluzibird, 2024-03-16:
That fixed it. Thanks.

`sudo pacman -Syy i2c-tools`

closed 2024-03-16 by liluzibird