liquidctl/liquidctl #686: Self-contained executable for downloading?

dustContributor, 2024-03-27:
Hi! It's really convenient how  [amdgpu_top](https://github.com/Umio-Yasuno/amdgpu_top) ships an AppImage which is easily managed on its own, you can just add it to the path and that's it, update it whenever you feel like it too.

Would it be possible to ship this in a similar way? It'd be quite accessible and easy for people to try the tool out.

That being said I believe python has also its own "native" ways of shipping self-contained executables. Just having a self contained executable in any way would be useful, AppImage or not.

Thank you for your time!

jonasmalacofilho, 2024-04-05:
In the past there was a Windows executable, but at some point most AVs decided that any Python program that's packaged as single binary should trigger some very silly heuristics.

See: #370

jonasmalacofilho, 2024-04-05:
Oh, wait, you mentioned AppImage... so you're thinking about self-contained executables for Linux, not Windows.

I think that's overkill for a Python project with very conservative dependencies, and that's already packages almost every(Linux)where.

But I'm also not opposed, if someone does the work. It should be something we can set up to build in CI.

jonasmalacofilho, 2024-04-07:
On ~~second~~ third thought, I don't think we should get involved in distributing Python.

AppImage is overkill here, and the risks for the user (stemming from the, realistically, outdated/unmaintained bundled Python runtime) outweigh the benefits.

closed 2024-04-07 by jonasmalacofilho