liquidctl/liquidctl #691: Cant set Pump&fan speed on Kraken Elite 2023

Leonm99, 2024-04-05:
Hello, im new to Liquidctl,

ive installed it with 
`# the latest snapshot of the official source code repository (requires git)
python -m pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl
`

Liquidctl status gives me:
`NZXT Kraken 2023 Elite (broken)
├── Liquid temperature    32.2  °C

├── Pump speed            1357  rpm

├── Pump duty               30  %

├── Fan speed              714  rpm

└── Fan duty                30  %
`

but when i try and set pump speed with for example:  " sudo liquidctl --match Kraken set pump speed 40 " i get:

`ERROR: no device matches available drivers and selection criteria`

ive read that setting an image on the LCD is currently not working but as far as i could tell setting pump speed should work right?

im really sorry if im overlooking something obvious im fairly new to linux in general, but i would greatly appreciate any help or nods in the right direction.

ill gladly provide any missing info if there is any.

jonasmalacofilho, 2024-04-05:
Can you report the output from the four commands bellow:

```
which liquidctl
liquidctl --version
sudo which liquidctl
sudo liquidctl --version
```

Leonm99, 2024-04-05:
oh seems like i messed somethin up..

which liquidctl > ~/.local/bin/liquidctl
liquidctl --version > liquidctl v1.14.0.dev48+g726aab8 (Linux-6.8.2-201.fsync.fc39.x86_64-x86_64-with-glibc2.38)
sudo which liquidctl > /usr/bin/liquidctl
sudo liquidctl --version > liquidctl v1.13.0 (Linux-6.8.2-201.fsync.fc39.x86_64-x86_64-with-glibc2.38)

also i should mention im using nobara linux wich is based on fedora iirc

i greatly appreciate you taking out time for me!

Leonm99, 2024-04-05:
i apologize for creating this issue, but im really happy that you gave me the right place to look.

i sudo removed liquidctl

then i needed to install python3-devel

after that i could do  " sudo python -m pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl "

now i can set pump speed without any issues.

i believe i tried this earlier but got a compilation error because i didnt have python3-devel.

thanks again, i wish you a great day/night :)

closed 2024-04-05 by Leonm99

jonasmalacofilho, 2024-04-05:
Happy to hear, and don't worry, that's a fairly common problem.