liquidctl/liquidctl #697: AIO RGB turns off and on once per CLI command to change speeds

CPPlus, 2024-04-21:
### Describe the bug

I have the iCUE H170i ELITE CAPELLIX XT
Every time I do a command like `liquidctl set fans speed 70` the RGB of the pump and the 3 fans "flickers" - it goes off for about half a second and then comes back on.
I know this model is not supported, but the fan speed settings work - my question is: should I worry about the RGB on/off flicker or is it purely a visual thing.
I am asking because I wrote a C++ program that constantly changes the fan speeds based on the CPU package temperature and the problematic command is executed quite often.

### Commands executed

`liquidctl set fans speed 70`

### Output of all relevant commands with `--debug` flag

`[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.8.7-arch1-1-x86_64-with-glibc2.39
[DEBUG] (cli) (_log_env_infos): python: 3.11.8 (main, Feb 12 2024, 14:50:05) [GCC 13.2.1 20230801]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.3.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-19
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-17
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-15
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-18
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-16
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0b0e:24c8 (usage_page=0x000b usage=0x0005)
[DEBUG] (usb) (find_devices): HID device: 0b0e:24c8 (usage_page=0x000b usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0b0e:24c8 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b0e:24c8 (usage_page=0xff30 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 0b0e:24c8 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 0b0e:24c8 (usage_page=0xff99 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1532:00a7 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3233:6301 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3233:6301 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3233:6301 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3233:6301 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3233:6301 (usage_page=0x000c usage=0x0001)
[... 84 lines cut ...]
[DEBUG] (usb) (write): writing report 0x00 with 96 bytes: 08:05:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 00:05:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[DEBUG] (usb) (write): writing report 0x00 with 96 bytes: 08:01:03:00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00`

### Affected device

iCUE H170i ELITE CAPELLIX XT

### Does your version of liquidctl support the device in question?

I've checked and it does not

### Operating system and version

Arch Linux 6.8.7-arch1-1

### Installation method

yay -S liquidctl (1.13.0-1)

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.8.7-arch1-1-x86_64-with-glibc2.39)

jonasmalacofilho, 2024-04-21:
Related: #448

I'm not sure if the response there is sill applicable.

> There is currently no path to fixing this as liquidctl only supports hardware mode.

And we might need to document this behavior in the appropriate guide.

CPPlus, 2024-04-24:
- How hard would it be to fix the issue?
- I will read the developer documentation and investigate how `liquidctl` works in the upcoming days, but can someone give a TLDR summary on roughly what kind of effort would be required?
- I haven't written drivers before, but I do have some experience with low level protocols, so it would be useful if I can gauge how much time that would take if I were to take on this issue...

Outside of the "RGB flicker" the other issue is that the command takes half a second to complete and during that time (probably because of the controller "wake up"?) parallel requests also fail - can someone also shed light on whether these two issues are related and actually a single issue? (I am assuming after reading the other issue that the software-hardware mode switch + the "wake up" results in both the delay without parallel requests AND the RGB flicker)

Note: sorry for closing the issue, happened by accident

jonasmalacofilho, 2024-04-24:
The limitation seems to be of how the driver implements its hardware modes (=modes in which no program is required to remain running after the initial setup/configuration). In software mode (=when such program is required, and effects are implemented by it in software) this doesn't seem to be an issue:

> I also implemented the OpenRGB controller. And no it will not. But the OpenRGB controller has to always be running.
> —Parker

The liquidctl CLI can only support hardware mode since it's supposed to be one-shot. Adding software mode APIs to our driver would be a welcome improvement, but they wouldn't be used by our main/official CLI.

> How hard would it be to fix the issue?

It seems to be a liquidctl vs device protocol design disconnect that, as far as we know (about the protocol), cannot be bridged.

> will read the developer documentation and investigate how `liquidctl` works in the upcoming days, but can someone give a TLDR summary on roughly what kind of effort would be required?

Discovering new features of the protocol used by these devices, or some creative way to bypass the issue within the design limitations I mentioned above.

The actual implementation shouldn't be very hard. Essentially it's all about reading and writing from/to a HID device. This happens in our drivers, in this case, [`commander_core.py`](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/commander_core.py).

> Outside of the "RGB flicker" the other issue is that the command takes half a second to complete and during that time (probably because of the controller "wake up"?) parallel requests also fail - [...]

I'm not 100% sure (this is not a driver I maintain directly, since I don't own one of these devices), but I think your assumption makes sense.

vimlucid, 2024-10-31:
@ruineka The flicking is the least of the issues with this workaround - while one command is handled (e.g. set fan speeds) other commands fail.

If you want different fan curves per fan tied to different temperatures (e.g. one set of fans are controlled by GPU temperature and another by CPU temperature) then the fan curves will be fighting for the "one command at a time" resource so if your temperatures go up and down very quickly you will always have a delayed response on one of the fan curves...

This needs to be fixed properly, but I definitely won't have the time to get into it. For now I am simply ignoring the flickering and I have a single fan curve for all of my fans.

closed 2024-12-16 by jonasmalacofilho