liquidctl/liquidctl #700: Liquidctl broken on python 3.12.3

robzombie91, 2024-04-28:
### Describe the bug

Liquidctl cant find the module named Liquidctl

`Traceback (most recent call last):
  File "/home/rob/.local/share/pipx/venvs/liquidctl/bin/./liquidctl", line 5, in <module>
    from liquidctl.cli import main
ModuleNotFoundError: No module named 'liquidctl'
`

### Commands executed

`cd /home/rob/.local/share/pipx/venvs/liquidctl/bin/ && sudo ./liquidctl initialize all && sudo ./liquidctl --match kraken set pump speed 25 40 35 60 45 80 55 100 && sudo ./liquidctl --match kraken set fan speed 20 30 30 50 34 80 40 90 50 100 && sudo ./liquidctl status`

Expected outcome: liquidctl enables my kraken 240, sets the pump and fan speed in a curve and then shows the output of status

Observed outcome: liquidctl crashes with a module error most likely due to python being updated on arch linux

### Output of all relevant commands with `--debug` flag

`[rob@robpc bin]$ ./liquidctl --debug
Traceback (most recent call last):
  File "/home/rob/.local/share/pipx/venvs/liquidctl/bin/./liquidctl", line 5, in <module>
    from liquidctl.cli import main
ModuleNotFoundError: No module named 'liquidctl'
`
Each command shows the same error no matter what is used

### Affected device

NZXT Kraken 240 2023

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Arch Linux 6.8.7

### Installation method

Used # the latest snapshot of the official source code repository (requires git) python -m pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl to install to a venv

### Version of liquidctl

liquidctl github snapsnot 85b32a7

jonasmalacofilho, 2024-04-28:
Cannot reproduce this on a just updated Arch with Python 3.12.3.

The only thing that comes to mind is that your venv got broken once Python was updated.

robzombie91, 2024-04-29:
fixed with pipx uninstall since pip doesnt work in arch under a venv