liquidctl/liquidctl #703: Kraken X60 error on initialize

TehNomad, 2024-05-12:
### Describe the bug

TypeError with Kraken X60

### Commands executed

``` liquidctl --legacy-690lc initialize --debug ```

### Output of all relevant commands with `--debug` flag

``` 
[DEBUG] (cli) (_log_env_infos): script: C:\Users\Timothy Wiryaman\liquidctl\Scripts\liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.14.0.dev76+g6426a06
[DEBUG] (cli) (_log_env_infos): platform: Windows-10-10.0.19045-SP0
[DEBUG] (cli) (_log_env_infos): python: 3.12.3 (tags/v3.12.3:f6650f9, Apr  9 2024, 14:05:25) [MSC v.1938 64 bit (AMD64)]
[DEBUG] (cli) (_log_env_infos): encoding: 1252 current, cp1252 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.3.0
[DEBUG] (cli) (_log_env_infos): with smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_env_infos): with winusbcdc: 1.7
[DEBUG] (cli) (_log_env_infos): with libusb-package: 1.0.26.1
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, Coolit, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0xff05 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0xff03 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0xff02 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0xff04 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff05 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff03 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff02 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3554:f526 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff04 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0xffb5 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b05:18a3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
[DEBUG] (usb) (find_devices): USB device: 2433:b200
[DEBUG] (usb) (probe): Legacy690Lc identified: Asetek 690LC (assuming NZXT Kraken X)
[DEBUG] (cli) (main): device: Asetek 690LC (assuming NZXT Kraken X)
[DEBUG] (usb) (open): selected interface: 0
[DEBUG] (asetek) (_configure_flow_control): set clear to send = True
[DEBUG] (usb) (ctrl_transfer): sending control transfer with (64, 2, 2), {}
[INFO] (cli) (log): detailed error: Asetek 690LC (assuming NZXT Kraken X): unexpected error: TypeError("'NoneType' object is not iterable")
Traceback (most recent call last):
  File "C:\Users\Timothy Wiryaman\liquidctl\Lib\site-packages\liquidctl\cli.py", line 456, in main
    with dev.connect(**opts):
         ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Timothy Wiryaman\liquidctl\Lib\site-packages\liquidctl\driver\asetek.py", line 354, in connect
    loc = f'bus{self.bus}_port{"_".join(map(str, self.port))}'
                                        ^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not iterable
[ERROR] (cli) (log): Asetek 690LC (assuming NZXT Kraken X): unexpected error: TypeError("'NoneType' object is not iterable") 
```

### Affected device

Kraken X60

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Windows 10

### Installation method

pip with latest git version

### Version of liquidctl

liquidctl v1.14.0.dev76+g6426a06

jonasmalacofilho, 2024-05-12:
This seems to come from somewhere deep down our USB stack. Can you set some debugging environment variables (on that terminal only) and run the command again?

```
set PYUSB_DEBUG=debug
set LIBUSB_DEBUG=4
liquidctl --legacy-690lc initialize --debug
```

TehNomad, 2024-05-13:
This is the output after setting those environment variables:

```
[DEBUG] (cli) (_log_env_infos): script: C:\Users\---\liquidctl\Scripts\liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.14.0.dev76+g6426a06
[DEBUG] (cli) (_log_env_infos): platform: Windows-10-10.0.19045-SP0
[DEBUG] (cli) (_log_env_infos): python: 3.12.3 (tags/v3.12.3:f6650f9, Apr  9 2024, 14:05:25) [MSC v.1938 64 bit (AMD64)]
[DEBUG] (cli) (_log_env_infos): encoding: 1252 current, cp1252 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.3.0
[DEBUG] (cli) (_log_env_infos): with smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_env_infos): with winusbcdc: 1.7
[DEBUG] (cli) (_log_env_infos): with libusb-package: 1.0.26.1
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, Coolit, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff05 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff03 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff02 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0xff04 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 3554:f524 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0xffb5 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 05ac:024f (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b05:18a3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (enumerate): using libusb_package.find
2024-05-13 06:40:10,345 ERROR:usb.libloader:'Libusb 1' could not be found
[ERROR] (libloader) (load_locate_library): 'Libusb 1' could not be found
2024-05-13 06:40:10,345 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
[ERROR] (libusb1) (get_backend): Error loading libusb 1.0 backend
2024-05-13 06:40:10,352 ERROR:usb.libloader:'Libusb 1' could not be found
[ERROR] (libloader) (load_locate_library): 'Libusb 1' could not be found
2024-05-13 06:40:10,353 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
[ERROR] (libusb1) (get_backend): Error loading libusb 1.0 backend
2024-05-13 06:40:10,355 ERROR:usb.libloader:'OpenUSB library' could not be found
[ERROR] (libloader) (load_locate_library): 'OpenUSB library' could not be found
2024-05-13 06:40:10,355 ERROR:usb.backend.openusb:Error loading OpenUSB backend
[ERROR] (openusb) (get_backend): Error loading OpenUSB backend
2024-05-13 06:40:10,363 INFO:usb.core:find(): using backend "usb.backend.libusb0"
[INFO] (core) (find): find(): using backend "usb.backend.libusb0"
2024-05-13 06:40:10,364 DEBUG:usb.backend.libusb0:_LibUSB.enumerate_devices()
[DEBUG] (_debug) (_trace_function_call): _LibUSB.enumerate_devices()
2024-05-13 06:40:10,364 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x000002089DE920D0>)
[DEBUG] (_debug) (_trace_function_call): _LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x000002089DE920D0>)
[DEBUG] (usb) (find_devices): USB device: 2433:b200
[DEBUG] (usb) (probe): Legacy690Lc identified: Asetek 690LC (assuming NZXT Kraken X)
[DEBUG] (cli) (main): device: Asetek 690LC (assuming NZXT Kraken X)
2024-05-13 06:40:10,366 DEBUG:usb.backend.libusb0:_LibUSB.open_device(<usb.backend.libusb0._usb_device object at 0x000002089DE920D0>)
[DEBUG] (_debug) (_trace_function_call): _LibUSB.open_device(<usb.backend.libusb0._usb_device object at 0x000002089DE920D0>)
2024-05-13 06:40:10,366 DEBUG:usb.backend.libusb0:_LibUSB.get_configuration(2235989602944)
[DEBUG] (_debug) (_trace_function_call): _LibUSB.get_configuration(2235989602944)
2024-05-13 06:40:10,366 DEBUG:usb.backend.libusb0:_LibUSB.ctrl_transfer(2235989602944, 128, 8, 0, 0, array('B', [0]), 100)
[DEBUG] (_debug) (_trace_function_call): _LibUSB.ctrl_transfer(2235989602944, 128, 8, 0, 0, array('B', [0]), 100)
2024-05-13 06:40:10,371 DEBUG:usb.backend.libusb0:_LibUSB.get_configuration_descriptor(<usb.backend.libusb0._usb_device object at 0x000002089DE920D0>, 0)
[... cut ...]
  File "C:\Users\---\liquidctl\Lib\site-packages\liquidctl\driver\asetek.py", line 354, in connect
    loc = f'bus{self.bus}_port{"_".join(map(str, self.port))}'
                                        ^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not iterable
[ERROR] (cli) (log): Asetek 690LC (assuming NZXT Kraken X): unexpected error: TypeError("'NoneType' object is not iterable")
2024-05-13 06:40:10,391 DEBUG:usb.backend.libusb0:_LibUSB.close_device(2235989602944)
[DEBUG] (_debug) (_trace_function_call): _LibUSB.close_device(2235989602944)
```

jonasmalacofilho, 2024-05-13:
Oh, I see what's wrong, libusb0 doesn't provide the device's port number[s].

(More precisely, PyUSB assumes that they don't, because they didn't and probably still don't... although there are so many variants of libusb0 that anything is possible, really).

Can you run:

```
python -c 'exec("import usb.core\nfor d in usb.core.find(idVendor=0x2433, find_all=True): print(hex(d.idVendor), hex(d.idProduct), d.product, d.bus, d.address, d.port_number, d.port_numbers)")'
```

and post the output?

jonasmalacofilho, 2024-05-13:
Alternatively, just use libusb1. Apart from obscure quirks, there aren't many reasons to use libusb0.

TehNomad, 2024-05-14:
> Alternatively, just use libusb1. Apart from obscure quirks, there aren't many reasons to use libusb0.
> 
> EDIT: to be clear, I still appreciate the report, and your output from the command in my previous comment may help us make this driver compatible with lisusb0 too, which would be nice. It's just that you (personally) might want to consider dropping libusb0, especially if you only "picked" it because of some outdated guide you found online.

Ah okay, it works now with libusb1. I thought that libusb1 would be included with liquidctl, but I was wrong.

jonasmalacofilho, 2024-05-14:
We try to have it installed automatically, but the method we use has limitations.

commit b5204bbe (2024-05-14): asetek: replace port numbers with bus address on storage key

Port numbers are only available on libusb-1.0.

Fixes: #703
PR #704 (2024-05-15): Don't depend on port numbers
Port numbers are only available on libusb-1.0.

Fixes: #703
Closes: 
Related: 

---

Checklist:

- [x] Adhere to the [development process]
- [x] Conform to the [style guide]
- [ ] Verify that the changes work as expected on real hardware
- [x] ~~Add~~ Adjust automated tests cases
- [x] Verify that all (other) automated tests (still) pass
- [ ] Update/add documentation
    - [ ] README, with ["new/changed in" notes]
    - [ ] applicable `docs/*guide.md` device guides, with ["new/changed in" notes]
    - [ ] `liquidctl.8` Linux/Unix/Mac OS man page
    - [ ] protocol [... cut]
commit 1ddcc80b (2024-05-15): asetek: replace port numbers with bus address on storage key

Port numbers are only available on libusb-1.0.

Fixes: #703
closed 2024-05-15 by jonasmalacofilho