liquidctl/liquidctl #705: Corsair Commander ST:  fan curve still not working

ps-kl, 2024-05-15:
### Describe the bug

The last change made Commander Core XT  Fan Curves work but the Commander ST is still not working for me.
ERROR: Corsair Commander ST: unexpected error: IndexError('index out of range')

### Commands executed

root@cc182207b6e4:/# liquidctl --version
liquidctl v1.14.0.dev76+g6426a06 (Linux-6.1.79-Unraid-x86_64-with-glibc2.36)
root@cc182207b6e4:/# liquidctl list
Device #0: Corsair Commander Core XT
Device https://github.com/liquidctl/liquidctl/issues/1: Corsair Commander ST
Device https://github.com/liquidctl/liquidctl/issues/2: ASUS Aura LED Controller
root@cc182207b6e4:/# liquidctl --match ST set fans speed 20 10 30 20 40 30 50 50 60 90
ERROR: Corsair Commander ST: unexpected error: IndexError('index out of range')
root@cc182207b6e4:/# liquidctl --match ST set fans speed 30 10 35 30 40 40 45 50 55 60 60 70 65 100
ERROR: Corsair Commander ST: unexpected error: IndexError('index out of range')
root@cc182207b6e4:/# liquidctl --match XT set fans speed 30 10 35 30 40 40 45 50 55 60 60 70 65 100

### Output of all relevant commands with `--debug` flag

root@cc182207b6e4:/# liquidctl list --debug
[DEBUG] (cli) (_log_env_infos): script: /opt/venv/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.14.0.dev76+g6426a06
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.1.79-Unraid-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_env_infos): python: 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, utf-8 preferred, utf8_mode 1
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.3.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1.post2
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, Coolit, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core XT
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c39 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c32 (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander ST
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c32 (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (probe): AuraLed identified: ASUS Aura LED Controller
[DEBUG] (usb) (find_devices): HID device: 0b05:1a53 (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[... 184 lines cut ...]
  File "/opt/venv/lib/python3.11/site-packages/liquidctl/driver/commander_core.py", line 172, in set_speed_profile
    count = res[i+1]
            ~~~^^^^^
IndexError: index out of range
[ERROR] (cli) (log): Corsair Commander ST: unexpected error: IndexError('index out of range')

### Affected device

Corsair Commander ST

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Unraid Docker

### Installation method

Docker 

### Version of liquidctl

1.14.0.dev76+g6426a06

jonasmalacofilho, 2024-05-15:
@agireud , do you have any idea of what might be the issue here?

agireud, 2024-05-16:
@ps-kl could you please post the currently installed firmware versions of both devices?

ps-kl, 2024-05-16:
oh ok good point. I don't even though about this. sorry ^^

Corsair Commander ST
├── Firmware version            2.0.17 

Corsair Commander Core XT
├── Firmware version        1.3.54 

A friend of mine already has   Commander ST with 2.0.19 ...... It's a long time ago I had a windows with Icue up and running. I will update everything and test again. Will report back.

ps-kl, 2024-05-22:
the XT Version is working flawless. Just to clarify.

letter-v, 2025-04-10:
I have the same issue with ST.