liquidctl/liquidctl #706: ERROR: NZXT Kraken Z (Z53, Z63 or Z73) (experimental): insufficient permissions

mansman12, 2024-05-21:
### Describe the bug

I have attempted to put it in a script to run at startup, however the startup script fails because of insufficient permissions. The script runs fine manually if using sudo, but does not even with adding liquidctl to rules.d

I have followed the adding `71-liquidctl.rules` to *both* `/usr/lib/udev/rules.d/` and `/etc/udev/rules.d` (even though just `/etc/udev/rules.d` should be required), then run:

```
udevadm control --reload
udevadm trigger
```

And rebooted the system.

The startup script still fails because of insufficient permissions.

At a loss on how to proceed further, can just workaround it by remembering to run the script manually at startup.

### Commands executed

```
#!/usr/bin/env bash

liquidctl initialize all
liquidctl --match kraken set pump speed  20 35  40 60  50 70  65 85  80 95  85 95  90 100
liquidctl --match kraken set fan speed  20 40  65 65  80 75  85 90  90 95  95 100
liquidctl --match kraken set lcd screen brightness 50
liquidctl --match kraken set lcd screen orientation 90
```

### Output of all relevant commands with `--debug` flag

(This is just for the `liquidctl --match kraken set pump speed  20 35  40 60  50 70  65 85  80 95  85 95  90 100` command)

```
024-05-21 11:24:28,628 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 5ad3123e51e0 at 0x78c261f2eae0>)
DEBUG: _LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 5ad3123e51e0 at 0x78c261f2eae0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000007] [00001578] libusb: debug [libusb_init_context] libusb v1.0.27.11882
[ 0.000016] [00001578] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000022] [00001578] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000024] [00001578] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000029] [00001578] libusb: debug [get_kernel_version] reported kernel version is 6.9.1-arch1-1
[ 0.000044] [00001578] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000047] [00001578] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000056] [00001578] libusb: debug [op_init] sysfs is available
[ 0.000175] [0000157b] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.012994] [00001578] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.012999] [00001578] libusb: debug [linux_get_device_address] scan usb1
[ 0.013024] [00001578] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.013027] [00001578] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.013030] [00001578] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.013108] [00001578] libusb: debug [linux_get_device_address] getting address for device: 1-5 detached: 0
[ 0.013111] [00001578] libusb: debug [linux_get_device_address] scan 1-5
[ 0.013132] [00001578] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.013135] [00001578] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.013136] [00001578] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.013159] [00001578] libusb: debug [linux_get_parent_info] dev 0x5ad3123ea570 (1-5) has parent 0x5ad3123e9e90 (usb1) port 5
[ 0.013221] [00001578] libusb: debug [linux_get_device_address] getting address for device: 1-5.1 detached: 0
[ 0.013224] [00001578] libusb: debug [linux_get_device_address] scan 1-5.1
[ 0.013247] [00001578] libusb: debug [linux_get_device_address] bus=1 dev=4
[ 0.013250] [00001578] libusb: debug [linux_enumerate_device] busnum 1 devaddr 4 session_id 260
[ 0.013252] [00001578] libusb: debug [linux_enumerate_device] allocating new device for 1/4 (session 260)
[ 0.013276] [00001578] libusb: debug [linux_get_parent_info] dev 0x5ad3123ea980 (1-5.1) has parent 0x5ad3123ea570 (1-5) port 1
[ 0.013337] [00001578] libusb: debug [linux_get_device_address] getting address for device: 1-5.3 detached: 0
[ 0.013340] [00001578] libusb: debug [linux_get_device_address] scan 1-5.3
[ 0.013363] [00001578] libusb: debug [linux_get_device_address] bus=1 dev=5
[ 0.013365] [00001578] libusb: debug [linux_enumerate_device] busnum 1 devaddr 5 session_id 261
[ 0.013367] [00001578] libusb: debug [linux_enumerate_device] allocating new device for 1/5 (session 261)
[ 0.013390] [00001578] libusb: debug [linux_get_parent_info] dev 0x5ad3123eab00 (1-5.3) has parent 0x5ad3123ea570 (1-5) port 3
[... 397 lines cut ...]
[ 0.023594] [00001578] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.023605] [0000157b] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.023656] [00001578] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.023660] [00001578] libusb: debug [usbi_remove_event_source] remove fd 3
```

### Affected device

NZXT Kraken Z73

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Arch Linux

### Installation method

Official Arch Linux repository

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.9.1-arch1-1-x86_64-with-glibc2.39)

jonasmalacofilho, 2024-05-22:
> I have attempted to put it in a script to run at startup,

This probably hints at the problem. Our udev rules use the `TAG+=uaccess` mechanism, which only works for seats (basically: physical sessions).[^1][^2]

I'm guessing your startup script runs without a seat (or as root).

[^1]: https://www.freedesktop.org/wiki/Software/systemd/multiseat/
[^2]: https://github.com/systemd/systemd/issues/4288

jonasmalacofilho, 2024-05-22:
You should be able to work around this limitation by changing the rules to set a suitable group and mode.

https://github.com/liquidctl/liquidctl/blob/d343c0da647599cd543c6ce906277cb36e8eeac5/extra/linux/71-liquidctl.rules#L28-L35

Note: I don't think `plugdev` is available by default on Arch Linux. See: <https://wiki.archlinux.org/title/Users_and_groups#Pre-systemd_groups>.

mansman12, 2024-05-22:
> > I have attempted to put it in a script to run at startup,
> 
> This probably hints at the problem. Our udev rules use the `TAG+=uaccess` mechanism, which only works for seats (basically: physical sessions).[1](#user-content-fn-1-a53fd3172aa39c650df0342c77992b85)[2](#user-content-fn-2-a53fd3172aa39c650df0342c77992b85)
> 
> I'm guessing your startup script runs without a seat (or as root).
> ## Footnotes
> 
>     1. https://www.freedesktop.org/wiki/Software/systemd/multiseat/ [↩](#user-content-fnref-1-a53fd3172aa39c650df0342c77992b85)
> 
>     2. https://github.com/systemd/systemd/issues/4288 [↩](#user-content-fnref-2-a53fd3172aa39c650df0342c77992b85)

Not sure, even just running the script manually has the same issue so it doesn't seem to matter whether it's a login script or not.

I should probably clarify that the script runs at user log in, not at system boot.

mansman12, 2024-05-22:
> You should be able to work around this limitation by changing the rules to set a suitable group and mode. [...]

Arch linux manual states that appropriate udev rules can be used:

> liquidctl is a utility for overseeing and controlling some hardware monitoring devices not yet supported at the kernel level. Because liquidctl directly accesses the hardware devices, root privileges are generally required, though this can be avoided with appropriate udev rules.

https://man.archlinux.org/man/liquidctl.8.en

So I'm at a bit of a loss at the appropriate group to be used if not udev.

mansman12, 2024-05-22:
Ended up working around the issue by using systemd.

closed 2024-05-22 by mansman12

jonasmalacofilho, 2024-05-22:
What was the issue, exactly?

mansman12, 2024-05-23:
> What was the issue, exactly?

Couldn't run the command `liquidctl --match kraken set pump speed  20 35  40 60  50 70  65 85  80 95  85 95  90 100` without sudo.

For putting it in a script that executes on log in, it requires sudo at the front so the script doesn't work.

To get around that limitation I just used set is as a systemd service.