liquidctl/liquidctl #712: Linux Mint and liquidctl installation with NZXT Kraken 2023 don't work

Trigger58, 2024-06-04:
Hello,
i tried since 3 days to get liquidctl to work but all i did was absolute useless.

I use Linux Mint Virginia (clean install WITHOUT any other tools) and did all the steps from the github page include to copy the 71.liquidctl.rules file to `/usr/lib/udev/rules.d/`

Nothing helps. I always need to use the sudo comand to get liquidctl to work but if i try to list my cooling system nothiong happens. if i use "initialize all" comand then i get an error:

Error: no devices matches available drivers and selection criteria

Please help. Is there anywhere a FULL step by step installation guide how to install this to Linux Mint?

jonasmalacofilho, 2024-07-19:
One thing you can try is to look how (if) the official distro packages change the udev rules in some way to make them work on those systems.

Trigger58, 2024-07-20:
Sorry, but i never got it to work. Tried all methods written on github and different other sites but nothing helped. Without that i can't use Linux so i buried it.

TsunamiMommy, 2024-08-04:
I have a Kraken Elite 240 working with liquidctl and the kraken 2023 driver with mint 21.3. I compiled from git and it works fine.

closed 2024-08-04 by jonasmalacofilho

Trigger58, 2024-08-23:
Can someone please tell me how to get this Kranken AIO to work with liquidctl?

Is there somewhere a package to install and if yes, where?
I don't know how to compile something and all i tried didn't work. I can't get this AIO to work with liquidctl!!!

Trigger58, 2024-08-23:
I got it now to work. i think.

ThimbleFire, 2024-08-24:
> I got it now to work. i think.

That's great. Please explain how for future readers, myself included when I one day go back to Mint.

mayank-17, 2024-08-26:
@Trigger58 It would be really great, if you can list down the steps how did you make it work.

TsunamiMommy, 2024-08-26:
If you have python3-pip installed you can simply do `sudo python -m pip install git+https://github.com/liquidctl/liquidctl#egg=liquidctl` and that should install the latest git which should give you control over your kraken cooler. Also, if you want an alternate you should be able to do it by installing the latest xanmod kernel. It contains the `nzxt_kraken3` kernel module which will expose the fan,pump, and liquid temp controls via hwmod directly. If you have an application like coolercontrol installed it should be able to control the device directly with no need for liquidctl. The main pro of the liquidctl git driver is that it contains basic lcd controls. Note that it can cause issues if you try to add an image to the lcd but you should be able to control orientation and brightness.