liquidctl/liquidctl #713: Add Firmware and Bootloader Information to Corsair Commander Pro Kernel Driver Implementation

RayJW, 2024-06-10:
Currently, there is a warning when liquidctl is leveraging the corsair-cpro driver since the driver didn't support firmware and bootloader version info, when the feature was implemented. This seems to have changed with [this](https://github.com/MisterZ42/corsair-cpro/commit/366545440d4f76533382502761da5a7d4612cd06) recent commit. 

I think it might be worth leveraging this functionality to simplify the documentation and getting rid of the ominous warning `WARNING: some attributes cannot be read from corsair-cpro kernel driver`. Not sure if it would help simplify the code, so I won't be making that claim. But I think generally it is desirable to have the same functionality across all implementations :)

jonasmalacofilho, 2024-12-16:
Could you maybe make a PR?

RayJW, 2024-12-17:
> Could you maybe make a PR?

I attempted once, but realized it'll take me more time to read into all related things than what it's worth. I suspect for someone who actually knows how everything works, this will be a few minutes tops. However, I'll gladly help with testing any PRs though.

jonasmalacofilho, 2024-12-17:
> I suspect for someone who actually knows how everything works, this will be a few minutes tops.

By thinking that it will take "a few minutes tops" to someone else you set a goal that you can't match. But the thing is that it's very unlikely anyone could match either.

For instance, I wrote most of infrastructure code in this repository and many of its drivers, and have worked on the Commander Pro driver directly. I would still need to read some of the code to remember exactly what needs to be done, check exactly which kernel attributes are available, when is debugfs available or not, how to find the right path in debugfs for a particular hwmon sysfs path, think how I'll handle older versions of the driver... And then make the necesssary modifications, write the tests, double check things will actually work in practice (I also don't have the device, so I can't just try the code), check that I haven't broken anything else or other OSs...

It wouldn't take me "a few minutes tops" either.

My unsolicited advice to ask for help or tips getting started, instead of throwing the feature that you want to someone else to implement. If something like this would take me `x` minutes, with some assistance from me you could probably in `y ∈ [x, 2·x]` minutes, which isn't bad at all.

> I attempted once, but realized it'll take me more time to read into all related things than what it's worth.

The issue in putting it that way is that it's even less worth it to other people, especially those that don't have this particular itch or maybe not even this particular device.

RayJW, 2024-12-17:
> For instance, I wrote most of infrastructure code in this repository and many of its drivers, and have worked on the Commander Pro driver directly. I would still need to read some of the code to [...]

See, that's the thing. Most of the things you mentioned are already things I've never worked with. I distinctly remember it took me like 2 hours to somewhat understand how debugfs works because the resources on it are rather sparse. Then I remember trying to read attributes from the debugfs file that the driver creates which just crashed my system with no error and the only help I found was some Stack Overflow question which said “that's not how you're supposed to do that” and nothing else. Also, I imagined there was precedence on how to handle old driver versions or at least someone who overlooks this project would know where to check how other drivers are handled. Now I need someone explaining all this to me as well.

> It wouldn't take me "a few minutes tops" either.

I'm sorry if that seemed rude. I didn't mean to finish the whole thing, which is why I offered help to test. However, I still think writing the code to read the info and funnelling that wherever it's supposed to go would still be much less than the four hours. That's how much I've spent working on this to not even get anything working.

> The issue in putting it that way is that it's even less worth it to other people, especially those that don't have this particular itch or maybe not even this particular device.

I agree. The thing is, I don't have any itch. It changes nothing for me whether this is implemented or not. I just helped with a liquidctl issue once involving this driver, and then saw by chance that this could be something making life easier for maintainers.

Just so it's clear. I'm not trying to be unthankful here, and I definitely have no expectations to the maintainers whatsoever. I've contributed quite some things in open source and I know I'm not entitled to anything, which also isn't the intention here. It's just that some issues inevitably take much more than 2x the time of someone more familiar with the matter. I mean, I helped with fwupd debugging but opening a PR would've taken me at least weeks just trying to understand all the firmware related things. Meanwhile, the maintainers found the right like 20 lines of code to add with my help and their vast knowledge.