liquidctl/liquidctl #714: liquidctl permission error

wroseland, 2024-06-12:
### Describe the bug

I had liquidctl running for several months and then needed to re-install my Ubuntu 22.04 OS. Now when I run "liquidctl initialize --pump-mode balanced" or "sudo liquidctl initialize --pump-mode balanced", I get "PermissionError: [Errno 13] Permission denied: '/sys/bus/i2c/devices'". Previously I did not need to use "sudo" and I get the same error anyway. I am not sure this has anything to do with liquidctl, but I am lost.

### Commands executed

sudo liquidctl initialize --pump-mode balanced

### Output of all relevant commands with `--debug` flag

liquidctl status --debug
[DEBUG] (cli) (_log_env_infos): script: /snap/liquidctl/36/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0.dev9+gc20fbb0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.5.0-35-generic-x86_64-with-glibc2.35
[DEBUG] (cli) (_log_env_infos): python: 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.13.1
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1.post2
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
Traceback (most recent call last):
  File "/snap/liquidctl/36/bin/liquidctl", line 8, in <module>
    sys.exit(main())
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/cli.py", line 420, in main
    selected = list(find_liquidctl_devices(**opts))
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/driver/__init__.py", line 61, in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/driver/smbus.py", line 60, in find_devices
    for i2c_dev in devices.iterdir():
  File "/usr/lib/python3.10/pathlib.py", line 1017, in iterdir
    for name in self._accessor.listdir(self):
PermissionError: [Errno 13] Permission denied: '/sys/bus/i2c/devices'

### Affected device

Corsair H115i

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Ubuntu 22.04

### Installation method

sudo snap install liquidctl

### Version of liquidctl

liquidctl v1.13.0.dev9+gc20fbb0 (Linux-6.5.0-35-generic-x86_64-with-glibc2.35)

wroseland, 2024-06-12:
I checked and I can as my user navigate that path to the destination using cd commands.

jonasmalacofilho, 2024-06-12:
This is a problem with the snap. Please check <https://github.com/degville/liquidctl-snap/issues/2> for updates.

Duplicate-of: #617

closed 2024-06-12 by jonasmalacofilho

jonasmalacofilho, 2024-06-12:
If you don't actually have any liquidctl-compatible i2c devices, then it *might* be enough to do:

```
 sudo snap connect liquidctl:raw-usb
 sudo snap connect liquidctl:hardware-observe
```
—https://snapcraft.io/liquidctl
—https://github.com/liquidctl/liquidctl/issues/617#issuecomment-1759083974

wroseland, 2024-06-12:
I did:
sudo snap connect liquidctl:raw-usb
and
sudo snap connect liquidctl:hardware-observe
then
sudo liquidctl initialize -v --pump-mode balanced
INFO: detailed error: Corsair Hydro H115i Platinum: could not open, possibly due to insufficient permissions: OSError('open failed')
Traceback (most recent call last):
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/cli.py", line 461, in main
    with dev.connect(**opts):
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/driver/hydro_platinum.py", line 184, in connect
    ret = super().connect(**kwargs)
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/driver/usb.py", line 130, in connect
    self.device.open()
  File "/snap/liquidctl/36/lib/python3.10/site-packages/liquidctl/driver/usb.py", line 416, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hidraw.pyx", line 154, in hidraw.device.open_path
OSError: open failed
ERROR: Corsair Hydro H115i Platinum: could not open, possibly due to insufficient permissions

It seems to know I have an H115i,
I have not rebooted yet.

jonasmalacofilho, 2024-07-14:
I don't use snaps, so unfortunately you'll need to check with the (third-party) snap maintainer for more information.