liquidctl/liquidctl #717: Request: NZXT HUE+ RGB Controller

eduncan911, 2024-06-22:
### Device type

RGB LED Controller

### Product page

https://www.bhphotovideo.com/c/product/1266788-REG/nzxt_ac_hueps_m1_hue_advanced_pc_lighting.html

### First-party software

NZXT CAM

### What configuration functionality does the device support?

lighting of accessories like RGB fans or LED strips

### Physical connection

USB Header

### Connection protocol

USB

### Additional information about the device

```
$ lsusb -d 04d8:00df
Bus 001 Device 002: ID 04d8:00df Microchip Technology, Inc. MCP2200 USB Serial Port Emulator

$ liquidctl list
Device #0: Corsair CMD16GX4M4B3600C18 DIMM1
Device #1: Corsair CMD16GX4M4B3600C18 DIMM2
Device #2: Corsair CMD16GX4M4B3600C18 DIMM3
Device #3: Corsair CMD16GX4M4B3600C18 DIMM4

(bonus points on how to RGB the DIMMs)
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, implementing the changes

eduncan911, 2024-06-22:
This seems to be a device that was briefly sold and quickly replaced within a few months with the HUE 2 device.  So it's kind of rare.  But, I have one.  

Technically, the device's model number on the device is: `AC-HUEPS-M1`

Odd how it shows up as a simple serial emulator.

jonasmalacofilho, 2024-07-14:
While it's not something I'd be willing to work on myself, if you can send a PR (with some tests using a mock device), I'll gladly review it and (if everything is ok) merge it.

You can take a look at #24 for a previous technical discussion on how this could be implemented. That's a very old issue, and liquidctl has changed a bit since then, but I think most of it will still be useful.

Just a word of caution: the last person to try this bricked their device. Admittedly (from what I recall) they tried to send Smart Device commands to their HUE+ but, still, these HUE+ are a bit sensitive...

eduncan911, 2024-07-15:
Thanks!  I will close this for now as a path forward with #24 if one is so inclined.

closed 2024-07-15 by eduncan911