liquidctl/liquidctl #723: Can't control 1/6 Corsair fans

pflavio, 2024-07-14:
### Describe the bug

Hello guys

I have a [H115i ELITE CAPELLIX Liquid CPU Cooler](https://www.corsair.com/us/en/p/cpu-coolers/cw-9060047-ww/icue-h115i-elite-capellix-liquid-cpu-cooler-cw-9060047-ww), a set of [3 LL120 RGB LED PWM fans ](LL120 RGB 120mm Dual Light Loop RGB LED PWM Fan — 3 Fan Pack with Lighting Node PRO)(came with a Lighting Node PRO that I'm not using) and another, unknown 120 mm Corsair RGB fan (which only has 8 instead of 16 LEDS based on ```liquidctl initialize all```).

The total amount of fans is therefore 6 and they're all shown in liquidctl, but I cannot control the 6th (the one where I don't know the exact model). It it always running at around 1500 rpm (max level) and it's super annoying.

I am using CoolerControl to set the fan curves, but I'm willing to adjust it manually (every time if necessary) to get rid of the noise. I tried ``` liquidctl --match corsair set fan6 speed 30``` but got ```ERROR: Corsair Commander Core (experimental): unspecified liquidctl error
```and ```ERROR: Corsair Commander Core (experimental): unexpected error: AssertionError('response does not match command')``` (not sure why two different errors when running the same command).

Also, I'm using OpenRGB to control the lights. Everything works, except for the 6 fans and the pump (all connected to the Commander Core that came with the pump). Is there any way to set the lights using liquitctl? One color all the time would be more than sufficient. At the moment it just keeps blinking yellow after booting up which is slightly annoying. I tried. I tried setting RGB colors as well as ```liquidctl --match corsair set led color off``` but it returned ```ERROR: Corsair Commander Core (experimental): operation not supported by the driver```. I saw that there is this [driver](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/commander_core.py) but I'm not sure whether it was installed automatically and if not, how I'd go about installing it manually.

Your help is truly appreciated. Thank you!

### Commands executed

liquidctl --match corsair set fan6 speed 30
liquidctl --match corsair set led color off

### Output of all relevant commands with `--debug` flag

[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.9.9-arch1-1-x86_64-with-glibc2.39
[DEBUG] (cli) (_log_env_infos): python: 3.12.4 (main, Jun  7 2024, 06:33:07) [GCC 14.1.1 20240522]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.4.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff43 usage=0x0202)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x000a)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x000c)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0x0001 usage=0x003a)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0xff00 usage=0x000c)
[DEBUG] (usb) (find_devices): HID device: 1038:12b3 (usage_page=0x0001 usage=0x003a)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): HID device: 1532:024e (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1532:024e (usage_page=0x0001 usage=0x0006)
[... 164 lines cut ...]
    dev.set_color(args['<channel>'].lower(), args['<mode>'].lower(), color, **opts)
  File "/usr/lib/python3.12/site-packages/liquidctl/driver/commander_core.py", line 137, in set_color
    raise NotSupportedByDriver
liquidctl.error.NotSupportedByDriver: operation not supported by the driver
[ERROR] (cli) (log): Corsair Commander Core (experimental): operation not supported by the driver

### Affected device

Corsair Commander Core and iCUE H115i ELITE CAPELLIX Liquid CPU Cooler

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Arch Linux

### Installation method

official repo

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.9.9-arch1-1-x86_64-with-glibc2.39)

jonasmalacofilho, 2024-07-14:
Hi!

> I have a [H115i ELITE CAPELLIX Liquid CPU Cooler](https://www.corsair.com/us/en/p/cpu-coolers/cw-9060047-ww/icue-h115i-elite-capellix-liquid-cpu-cooler-cw-9060047-ww), a set of [3 LL120 RGB LED PWM [...]

Can you paste that output from `liquidctl intiialize all`?

> I am using CoolerControl to set the fan curves, but I'm willing to adjust it manually (every time if necessary) to get rid of the noise. I tried ` liquidctl --match corsair set fan6 speed 30` but [...]

There are several bugs/issues with the experimental driver for the Commander Core family. Some caused by firmware changes, some unknown... For a _non-exhaustive_ list of open issues see [this comment](https://github.com/liquidctl/liquidctl/issues/639#issuecomment-1898820041).

> I tried. I tried setting RGB colors as well as `liquidctl --match corsair set led color off` but it returned `ERROR: Corsair Commander Core (experimental): operation not supported by the driver`.

The feature really isn't yet supported by the driver.

pflavio, 2024-07-15:
Good morning :) 

> Can you paste that output from `liquidctl intiialize all`?

Here you go: 
```Corsair Commander Core (experimental)
├── Firmware version            2.11.221
├── AIO LED count                     29
├── RGB port 1 LED count              16
├── RGB port 2 LED count              16
├── RGB port 3 LED count              16
├── RGB port 4 LED count               8
├── RGB port 5 LED count               8
├── RGB port 6 LED count               8
├── AIO port connected               Yes
├── Fan port 1 connected             Yes
├── Fan port 2 connected             Yes
├── Fan port 3 connected             Yes
├── Fan port 4 connected             Yes
├── Fan port 5 connected             Yes
├── Fan port 6 connected             Yes
├── Water temperature sensor         Yes
└── Temperature sensor 1              No

ASUS Aura LED Controller
└── Firmware version    AULA3-6K75-0222
```

> There are several bugs/issues with the experimental driver for the Commander Core family. Some caused by firmware changes, some unknown... For a _non-exhaustive_ list of open issues see [this comment](https://github.com/liquidctl/liquidctl/issues/639#issuecomment-1898820041).

Yeah I read that, but before I changed that one fan it worked so I'm wondering what caused it to stop. There probably aren't any fans that allow RGB control but not RPM I guess?

> The feature really isn't yet supported by the driver.

Do I have to install the [driver](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/commander_core.py) manually?

jonasmalacofilho, 2024-07-15:
> Yeah I read that, but before I changed that one fan it worked so I'm wondering what caused it to stop.

In some cases the firmware state gets corrupted. We don't know the cause for the other issues.

> There probably aren't any fans that allow RGB control but not RPM I guess?

That's not necessarily true, specially if you consider faulty PWM chips or cables.

> Do I have to install the [driver](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/commander_core.py) manually?

No, it's already in use.

pflavio, 2024-07-15:
OK thank you. Is there anything I could try to get it working atm or should i just wait for a future version?

-------- Original Message --------

jonasmalacofilho, 2024-07-15:
Closing as a duplicate (of any of the open issues mentioned in the linked comment, above). Unfortunately I can't offer more help right now.

Contributions in fixing or at least debugging these issues would be very much appreciated.

jonasmalacofilho, 2024-07-15:
> OK thank you. Is there anything I could try to get it working atm or should i just wait for a future version?

My genuine advice is for you to try to learn how to capture and reverse engineer the protocol. It's not as hard as it may seem, and you're in the best position to find and fix the problem (you have one of these devices, and you're affected by one of these issues).

closed 2024-07-15 by jonasmalacofilho