liquidctl/liquidctl #724: Using my H150i results in Error: no devices matches available drivers and selection criteria

jamestrotter, 2024-07-16:
### Describe the bug

Using my H150i results in Error: no devices matches available drivers and selection criteria

I have added the udev rules, disabled openrgb, tried `sudo rmmod usbhid && sudo modprobe usbhid `

### Commands executed

```
jamestrotter@jamespc:~$ sudo liquidctl status
Error: no devices matches available drivers and selection criteria
```

### Output of all relevant commands with `--debug` flag

```
jamestrotter@jamespc:~$ sudo liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-10
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-11
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0250
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c1a
[DEBUG] liquidctl.driver.usb: found HID device 1462:7b85
[DEBUG] liquidctl.driver.usb: found HID device 1462:7b85
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 8087:0025
[DEBUG] liquidctl.driver.usb: found USB device 1462:7b85
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0626
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c1a
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c12
[DEBUG] liquidctl.driver.usb: found USB device 045e:02e6
[DEBUG] liquidctl.driver.usb: found USB device 05ac:0250
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:1b70
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0610
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
Error: no devices matches available drivers and selection criteria
```

### Affected device

Corsair Hydro H150i Pro

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

PopOS linux

### Installation method

manual install from git

### Version of liquidctl

liquidctl v1.14.0.dev81+g256311b (Linux-6.9.3-76060903-generic-x86_64-with-glibc2.35)

jonasmalacofilho, 2024-07-16:
```
[DEBUG] liquidctl.cli: running liquidctl v1.5.1
```

You have more than one version of liquidctl installed, and the one accessible by root/with `sudo` (due to, presumably, differences in the `PATH` variable) is very old and, as expected, doesn't support that device.

jamestrotter, 2024-07-17:
Sure was, I installed from `apt` first, thought I'd remove from `apt`, installed from source. Clearly missed removing like I thought I had.

Now if I try and `sudo liquidctl` (believe I need this for reead/write access to smbus? Correct me if wrong) it tells me the module is missing. This is almost certainly more a issue in my linux knowledge than it is a problem with liquidctl, but how do I make it accessible from root with `sudo` when installed from source? I can see that I have the executable in `~/.local/bin` but I assume this is not in the `PATH` when `sudo`'ing.

I tried symlinking from `~/.local/bin/liquidctl` > `/usr/local/bin/liquidctl` but that didn't help and I subsequently undid it.

jonasmalacofilho, 2024-07-18:
While you can get it to work with symlinks and adjustments to where Python looks for modules, it's easier to have `sudo` preserve the `PATH` from where it's called from. There are several ways to do it, from global configuration to invocating it with `--preserve-env`...

However, you shouldn't need `sudo`: the udev rules, if properly set up (though sometimes a reboot is also necessary), should allow root-less liquidctl to work even with SMBus.

---

> I could re-run the install steps with sudo, but don't really want to do that unless explicitly told to

There's an argument to be made that the liquidctl installation should be writable only by root: regardless if with `sudo` or  udev rules,, it's a process that has access to sensitive parts of your system. Ideally it shouldn't be easy to replace it some rogue script that does all sorts of other things... (and this is course worse when running with `sudo`, since then the rogue script wouldn't be limited by only what's allows by the udev rules).

jamestrotter, 2024-07-18:
Gotcha - I do have the udev rules installed but was not seeing my ram temps, so assumed it was a privilege issue, but perhaps its just that my ram doesn't support it instead. Thanks for your help.