liquidctl/liquidctl #726: Corsair H150i Not Detected But Supported

brando56894, 2024-08-11:
I just purchased a new H150i due to the pump failing on my older model. I'm using Debian Linux (Proxmox actually) and it appears that **liquidctl** won't manage my AIO, even though its detected by Linux and apparently supported in **liquidctl**.

````
root@pve:/# lsusb -v -s 001:002

Bus 001 Device 002: ID 1b1c:0c37 Corsair H150iELITE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c37 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H150iELITE
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
Device Status:     0x0001
  Self Powered
````

[Line 137 in the driver includes those Bus IDs](https://github.com/liquidctl/liquidctl/blob/7637f173be167daeb1a2a21fd4fb0a9dd217d158/liquidctl/driver/hydro_platinum.py#L136) so I'm not sure what the issue is.

        (0x1b1c, 0x0c37, 'Corsair iCUE H150i Elite RGB',
            {'fan_count': 3, 'fan_leds': 0}),

```
 root@pve:/# liquidctl list

 root@pve:/# liquidctl initialize
ERROR: no device matches available drivers and selection criteria
```

aleksamagicka, 2024-08-11:
What version of liquidctl do you have installed?

brando56894, 2024-08-11:
The actual command shows v0

> What version of liquidctl do you have installed?

liquidctl v0.0.0-unknown (Linux-6.8.12-1-pve-x86_64-with-glibc2.36)

But according to the repo it's **v1.12.1-1**

According to the changelog it looks like support was added in 1.13? 
`
Corsair Hydro Elite RGB: add support for H100i and H150i Elite RGB (https://github.com/liquidctl/liquidctl/issues/556, PR https://github.com/liquidctl/liquidctl/pull/557, PR https://github.com/liquidctl/liquidctl/pull/559)`

I'm used to running Arch and being on the bleeding edge, so I forget how a lot of distros repos are pretty old.

aleksamagicka, 2024-08-11:
Yes, the one you have installed is indeed too old for your device. You need to be on 1.13.0 for it, at the minimum.

You can follow the [README instructions](https://github.com/liquidctl/liquidctl?tab=readme-ov-file#installing-from-pypi-or-github) to install a newer version directly.

brando56894, 2024-08-11:
Thanks for the quick replies! :) This is the reason why I've stuck with Arch most of the time haha

closed 2024-08-11 by brando56894