liquidctl/liquidctl #729: "Corsair Commander Core" returns unspecified liquidctl error when setting speed

P2949, 2024-09-03:
### Describe the bug

liquidctl returns unspecified error when trying to set any speed in any of the fans/pump. not sure if it's something with nixos being weird, but from what i saw nobody had the same issue.

### Commands executed

sudo liquidctl initialize all --debug

sudo liquidctl --match Commander set pump speed 100 --debug

### Output of all relevant commands with `--debug` flag

sudo liquidctl initialize all --debug

[DEBUG] (cli) (_log_env_infos): script: /nix/store/fxjsn0chxim0qvzs3crbavfhsw4ncyx4-python3.12-liquidctl-1.13.0/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.10.6-xanmod1-x86_64-with-glibc2.39
[DEBUG] (cli) (_log_env_infos): python: 3.12.4 (main, Jun  6 2024, 18:26:44) [GCC 13.3.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0.post2
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.4.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: 7-0037 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: 7-0053 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: 7-0051 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: 7-0036 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 0b05:18f3 (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (probe): AuraLed identified: ASUS Aura LED Controller
[DEBUG] (usb) (find_devices): HID device: c0f4:02f5 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: c0f4:02f5 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: c0f4:02f5 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c077 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c077 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c1c (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x0064)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x0065)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x0066)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x0067)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x00c8)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x00c9)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x00ca)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0a52 (usage_page=0xffc5 usage=0x00cb)
[... 199 lines cut ...]
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/fxjsn0chxim0qvzs3crbavfhsw4ncyx4-python3.12-liquidctl-1.13.0/lib/python3.12/site-packages/liquidctl/driver/commander_core.py", line 207, in _read_data
    raise ExpectationNotMet('device returned incorrect data type')
liquidctl.error.ExpectationNotMet: unspecified liquidctl error
[ERROR] (cli) (log): Corsair Commander Core (experimental): unspecified liquidctl error

### Affected device

Corsair Commander Core

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

linux (nixos)

### Installation method

Nix package manager

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.10.6-xanmod1-x86_64-with-glibc2.39)

8BITADDICTION, 2024-10-16:
Can confirm I am getting the same issue.

jonasmalacofilho, 2024-12-16:
Merged with other similar issues into #753.

closed 2024-12-16 by jonasmalacofilho