liquidctl/liquidctl #730: Corsair HX1000i set speed fan fail

Blindiak, 2024-09-04:
### Describe the bug

impossible set speed fan

### Commands executed

sudo liquidctl set Corsair speed 0 --debug

### Output of all relevant commands with `--debug` flag

[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.10.7-arch1-1-x86_64-with-glibc2.40
[DEBUG] (cli) (_log_env_infos): python: 3.12.5 (main, Aug  9 2024, 08:20:41) [GCC 14.2.1 20240805]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 10.4.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0xff43 usage=0x0602)
[DEBUG] (usb) (find_devices): HID device: 046d:c33c (usage_page=0xff43 usage=0x0604)
[DEBUG] (usb) (find_devices): HID device: 1462:7d89 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 1462:7d89 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:1c1e (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (__init__): has kernel driver: corsair-psu (/sys/class/hidraw/hidraw3/device/hwmon/hwmon7)
[DEBUG] (usb) (probe): CorsairHidPsu identified: Corsair HX1000i (2022) (experimental)
[DEBUG] (usb) (find_devices): HID device: 0db0:6c09 (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c08d (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c08d (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c08d (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c08d (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c08d (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c08d (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c08d (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 05e3:0625
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0a52
[DEBUG] (usb) (find_devices): USB device: 046d:c08d
[DEBUG] (usb) (find_devices): USB device: 2109:2812
[DEBUG] (usb) (find_devices): USB device: 1b1c:0d0f
[DEBUG] (usb) (find_devices): USB device: 8087:0033
[DEBUG] (usb) (find_devices): USB device: 0db0:6c09
[DEBUG] (usb) (find_devices): USB device: 1b1c:1c1e
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1462:7d89
[DEBUG] (usb) (find_devices): USB device: 046d:c33c
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: Corsair HX1000i (2022) (experimental)
[INFO] (corsair_hid_psu) (set_fixed_speed): ensuring fan control is in software mode
[DEBUG] (usb) (write): writing report 0x00 with 64 bytes: 02:f0:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 02:f0:01:e2:52:53:41:49:52:20:48:58:31:30:30:30:69:20:50:53:55:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[INFO] (corsair_hid_psu) (set_fixed_speed): setting fan PWM duty to 0%
[DEBUG] (usb) (write): writing report 0x00 with 64 bytes: 02:3b:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 64 bytes: 02:fe:01:e2:52:53:41:49:52:20:48:58:31:30:30:30:69:20:50:53:55:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[INFO] (cli) (log): detailed error: Corsair HX1000i (2022) (experimental): unexpected error: AssertionError('invalid response (possible conflict with another program)')
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/liquidctl/cli.py", line 475, in main
    _device_set_speed(dev, args, **opts)
  File "/usr/lib/python3.12/site-packages/liquidctl/cli.py", line 294, in _device_set_speed
    dev.set_fixed_speed(args['<channel>'].lower(), int(args['<percentage>'][0]), **opts)
  File "/usr/lib/python3.12/site-packages/liquidctl/driver/corsair_hid_psu.py", line 253, in set_fixed_speed
    self._exec(WriteBit.WRITE, CMD.FAN_COMMAND_1, [duty])
  File "/usr/lib/python3.12/site-packages/liquidctl/driver/corsair_hid_psu.py", line 288, in _exec
    assert ret[0:2] == out[0:2], f'invalid response (possible conflict with another program)'
AssertionError: invalid response (possible conflict with another program)
[ERROR] (cli) (log): Corsair HX1000i (2022) (experimental): unexpected error: AssertionError('invalid response (possible conflict with another program)')

### Affected device

Corsair HX1000i

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Arch Linux

### Installation method

yay

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.10.7-arch1-1-x86_64-with-glibc2.40)

jonasmalacofilho, 2024-09-04:
As the error message indicates, the most likely cause is a "possible conflict with another program".

Blindiak, 2024-09-04:
but i have no another program for fan control :/

Blindiak, 2024-09-04:
update : i try fort speed 13 it's work ! below 12 crash ... any idea for bypasse this ?

jonasmalacofilho, 2024-09-04:
> but i have no another program for fan control :/

Conflicts can happen with any programs that communicate with the PSU, even if just for power/temperature monitoring.

> update : i try fort speed 13 it's work ! below 12 crash ... any idea for bypasse this ?

If it's not a fluke that 13% worked (a conflict wouldn't necessary cause issues every time), then you can't bypass a restriction like that.

But restricting fan duty >= 13% feels somewhat arbitrary and improbable, plus other models in this family do support 0% fan duty. Not that you necessary should set it that low...

jonasmalacofilho, 2024-09-04:
From the docs, a 30% minimum fan duty value was always intended; 896bad9 finally implements it.

This means that if the issue wasn't caused by a conflict, and any value >= 13% is accepted by the PSU, then it's no longer possible to trigger this issue.

closed 2024-09-04 by jonasmalacofilho

Blindiak, 2024-09-04:
normaly this alimentation have zero rpm mode but never at 0 and the noze is insuportable :/

Blindiak, 2024-09-04:
> From the docs, a 30% minimum fan duty value was always intended; [896bad9](https://github.com/liquidctl/liquidctl/commit/896bad90e3db51094b0027a4988a7fe3edb1e065) finally implements it.
>
> This means [...]

fuuuu why 30% is to hight :/

jonasmalacofilho, 2024-09-05:
If your PSU has automatic zero-rpm mode, but isn't actually being able to use, you should definitively not disable automatic fan control and try to achieve the same manually!

Moreover, if your PSU's fan is running super loudly, that's also an indication that it's not only running at all, but running _fast!_ Again, you should definitely not force it to run slow!

I suggest you take a look a `liquidctl status` and try to see why your PSU is doing that. Maybe it's not outputting that much power, but it's still running super hot, which could indicate a problem with the fan, a blocked intake, etc. Or maybe it is outputting a lot a power, in which case I really don't know why you thought it would be a good idea to slow the fan down.

Just don't blow up your PSU, please?

commit 896bad90 (2024-09-04): corsair_hid_psu: enforce a 30% minimum user-set fan duty

From the docs, this limit should already be in place, but it was never
actually implemented correctly.

Fixes: #57
Closes: #730