liquidctl/liquidctl #731: Crash after  kernel 6.11 update.

edpil02, 2024-09-16:
### Describe the bug

My distro just upgraded  to kernel-6.11 and  since  liquidctl crashes  . I have a nzxt cooler device.

### Commands executed

$ liquidctl list
OSError: [Errno 6] No such device or address

Downgrade to kernel6.10 help the issue.

### Output of all relevant commands with `--debug` flag

liquidctl -g list

Traceback (most recent call last):
  File "/usr/bin/liquidctl", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/usr/lib/python3.13/site-packages/liquidctl/cli.py", line 420, in main
    selected = list(find_liquidctl_devices(**opts))
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/__init__.py", line 61, in find_liquidctl_devices
    for dev in bus_cls().find_devices(**kwargs):
               ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/smbus.py", line 71, in find_devices
    yield from i2c_bus.find_devices(drivers, **kwargs)
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/smbus.py", line 97, in find_devices
    yield from drv.probe(self, **kwargs)
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/ddr4.py", line 152, in probe
    eeprom = smbus.load_eeprom(spd_addr)
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/smbus.py", line 174, in load_eeprom
    eeprom = self._i2c_dev.joinpath(dev, 'eeprom').read_bytes()
  File "/usr/lib64/python3.13/pathlib/_abc.py", line 626, in read_bytes
    return f.read()
           ~~~~~~^^
OSError: [Errno 6] No such device or address
edmond@tuxmachine:~$ 

### Affected device

Kraken  nzxt cooler

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Fedora Linux

### Installation method

fedora install 

### Version of liquidctl

liquidctl-1.13.0-6.fc41 rawhide fedora

edpil02, 2024-09-16:
The kernel  also  have a driver (nzxt-kraken3)  and the monitoring have no issue :

$ Sensors 
z53-hid-3-4
Adapter: HID adapter
Pump speed:   1948 RPM
Fan speed:    1169 RPM
Coolant temp:  +49.5°C

jonasmalacofilho, 2024-09-16:
Can you paste the complete output with `--debug`?

edpil02, 2024-09-17:
I wonder if it is a problem in the kernel. Here is a debug of kernel 6.11 and 6.10 for comparison.

[6.11.txt](https://github.com/user-attachments/files/17023390/6.11.txt)
[6.10.txt](https://github.com/user-attachments/files/17023391/6.10.txt)

edpil02, 2024-09-17:
I made  a test  
change  line       eeprom = self._i2c_dev.joinpath(dev, 'eeprom').read_bytes()
to                         eeprom = self._i2c_dev.joinpath(dev, 'name').read_bytes()   makes   my device is detected again.

So it's    'eeprom'  that crashes the program ; perhaps some  changes  in the kernel.
https://lore.kernel.org/lkml/yxymqbpeuhq52w3ie5nl5pkoojjuybaeodqtrd6uhe2rkmfhfb@3dqr6vefsdxh/

FeRChImoNdE, 2024-09-19:
Failing also with Commander Core :(

jonasmalacofilho, 2024-09-20:
The problem doesn't appear to be related to any particular device supported by liquidctl. Likewise, changing the argument  from `'eeprom'` to some other string, while it's a workaround of sorts, does _not_ fix the issue: you're effectively just disabling DDR detection in liquidctl.

Instead, the underlying issue appears to be that in Linux 6.11 there's now some SMBus device that exposes an `eeprom` file in sysfs, but attempting to read it fails with `ENXIO`. Without knowing which device/kernel driver is involved, I'm not yet sure if it's an oversight on the part of the kernel driver author, or if liquidctl should indeed guard against these Schrödinger files.

It would help if you could track down which I2C device and kernel driver is causing this (btw: ~all SMBus devices are I2C devices). With that information I can take a look at the corresponding driver code, as well as any recent changes to it.

FeRChImoNdE, 2024-09-20:
I'll do my best in troubleshooting

[image]

Left side is --debug with Kernel 6.11, right Kernel 6.10
I don't know if it breaks in i2c-6 (last line before error), or i2c-4 (the next line that do not appear)
If is i2c-6, then the driver is i801_smbus, if is i2c-4, then the driver is nvidia (Zotac GeForce 3xxx's have an i2c bus apparently for rgb, never had run well on linux)

```
00:1f.4 SMBus: Intel Corporation Raptor Lake-S PCH SMBus Controller (rev 11)
        DeviceName: SMBus Controller
        Subsystem: ASUSTeK Computer Inc. Device 8882
        Flags: medium devsel, IRQ 18, IOMMU group 15
        Memory at 420622c000 (64-bit, non-prefetchable) [size=256]
        I/O ports at efa0 [size=32]
        Kernel driver in use: i801_smbus
        Kernel modules: i2c_i801

01:00.0 VGA compatible controller: NVIDIA Corporation GA104 [GeForce RTX 3070 Ti] (rev a1) (prog-if 00 [VGA controller])
        Subsystem: ZOTAC International (MCO) Ltd. Device 1653
        Flags: bus master, fast devsel, latency 0, IRQ 130, IOMMU group 16
        Memory at 42000000 (32-bit, non-prefetchable) [size=16M]
        Memory at 4000000000 (64-bit, prefetchable) [size=8G]
        Memory at 4200000000 (64-bit, prefetchable) [size=32M]
        I/O ports at 6000 [size=128]
        Expansion ROM at 43000000 [virtual] [disabled] [size=512K]
        Capabilities: [60] Power Management version 3
        Capabilities: [68] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [78] Express Legacy Endpoint, IntMsgNum 0
        Capabilities: [b4] Vendor Specific Information: Len=14 <?>
        Capabilities: [100] Virtual Channel
        Capabilities: [250] Latency Tolerance Reporting
        Capabilities: [258] L1 PM Substates
        Capabilities: [128] Power Budgeting <?>
        Capabilities: [420] Advanced Error Reporting
        Capabilities: [600] Vendor Specific Information: ID=0001 Rev=1 Len=024 <?>
        Capabilities: [900] Secondary PCI Express
        Capabilities: [bb0] Physical Resizable BAR
        Capabilities: [c1c] Physical Layer 16.0 GT/s <?>
        Capabilities: [d00] Lane Margining at the Receiver
        Capabilities: [e00] Data Link Feature <?>
        Kernel driver in use: nvidia
        Kernel modules: nouveau, nvidia_drm, nvidia
```
I hope it helps, if you need more information, do not hesitate to ask.

jonasmalacofilho, 2024-09-20:
Thanks, @FeRChImoNdE. It makes sense to be related to the `i801_smbus` (bus).

Can you run and paste the output from `sudo decode-dimms`?

jonasmalacofilho, 2024-09-20:
Also, I'm guessing that's a DDR5 system, right?

And can you also paste the output from `sudo lsmod`?

FeRChImoNdE, 2024-09-20:
> Thanks, @FeRChImoNdE. It makes sense to be related to the `i801_smbus` (bus).
> 
> Can you run and paste the output from `sudo decode-dimms`?

```
⫸ sudo decode-dimms
No EEPROM found, try loading the eeprom, at24 or ee1004 module
⫸ sudo modprobe eeprom
modprobe: FATAL: Module eeprom not found in directory /lib/modules/6.11.0-2-cachyos
⫸ sudo modprobe at24
⫸ sudo decode-dimms
# decode-dimms version 4.3

Memory Serial Presence Detect Decoder
By Philip Edelbrock, Christian Zuckschwerdt, Burkart Lingner,
Jean Delvare, Trent Piepho and others

Number of SDRAM DIMMs detected and decoded: 0
⫸ sudo modprobe -r at24
⫸ sudo modprobe ee1004
⫸ sudo decode-dimms
# decode-dimms version 4.3

Memory Serial Presence Detect Decoder
By Philip Edelbrock, Christian Zuckschwerdt, Burkart Lingner,
Jean Delvare, Trent Piepho and others

Number of SDRAM DIMMs detected and decoded: 0
⫸ sudo modprobe -r ee1004
```

FeRChImoNdE, 2024-09-20:
> Also, I'm guessing that's a DDR5 system, right?
> 
> And can you also paste the output from `sudo lsmod`?

```
⫸ sudo lsmod
Module                  Size  Used by
snd_seq_dummy          12288  0
snd_hrtimer            12288  1
rfcomm                102400  6
snd_seq               135168  7 snd_seq_dummy
nfsv3                  90112  1
nfs_acl                12288  1 nfsv3
cmac                   12288  2
algif_hash             12288  1
algif_skcipher         12288  1
af_alg                 32768  6 algif_hash,algif_skcipher
rpcsec_gss_krb5        40960  0
auth_rpcgss           192512  1 rpcsec_gss_krb5
nfsv4                1351680  0
dns_resolver           16384  1 nfsv4
nfs                   638976  4 nfsv4,nfsv3
lockd                 208896  2 nfsv3,nfs
grace                  12288  1 lockd
sunrpc                892928  13 nfsv4,auth_rpcgss,lockd,nfsv3,rpcsec_gss_krb5,nfs_acl,nfs
netfs                 581632  1 nfs
mousedev               24576  0
joydev                 24576  0
bnep                   36864  2
hid_generic            12288  0
btusb                  77824  0
btrtl                  36864  1 btusb
btintel                69632  1 btusb
snd_usb_audio         565248  5
btbcm                  24576  1 btusb
btmtk                  36864  1 btusb
snd_usbmidi_lib        53248  1 snd_usb_audio
bluetooth            1146880  34 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
snd_ump                40960  1 snd_usb_audio
snd_rawmidi            61440  2 snd_usbmidi_lib,snd_ump
usbhid                 90112  0
snd_seq_device         16384  3 snd_seq,snd_ump,snd_rawmidi
nct6775                45056  0
nct6775_core           86016  1 nct6775
hwmon_vid              12288  1 nct6775
vfat                   28672  1
fat                   110592  1 vfat
ext4                 1200128  1
crc16                  12288  2 bluetooth,ext4
mbcache                16384  1 ext4
jbd2                  212992  1 ext4
snd_sof_pci_intel_tgl    16384  0
snd_sof_pci_intel_cnl    20480  1 snd_sof_pci_intel_tgl
snd_sof_intel_hda_generic    40960  2 snd_sof_pci_intel_tgl,snd_sof_pci_intel_cnl
soundwire_intel        81920  1 snd_sof_intel_hda_generic
soundwire_cadence      49152  1 soundwire_intel
intel_rapl_msr         20480  0
snd_sof_intel_hda_common   204800  3 snd_sof_intel_hda_generic,snd_sof_pci_intel_tgl,snd_sof_pci_intel_cnl
intel_rapl_common      57344  1 intel_rapl_msr
snd_sof_intel_hda_mlink    36864  3 soundwire_intel,snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_sof_intel_hda      20480  2 snd_sof_intel_hda_common,snd_sof_intel_hda_generic
intel_uncore_frequency    12288  0
intel_uncore_frequency_common    16384  1 intel_uncore_frequency
snd_sof_pci            24576  3 snd_sof_intel_hda_generic,snd_sof_pci_intel_tgl,snd_sof_pci_intel_cnl
intel_tcc_cooling      12288  0
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_generic
snd_sof               479232  5 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_hda_generic,snd_sof_intel_hda,snd_sof_pci_intel_cnl
x86_pkg_temp_thermal    16384  0
intel_powerclamp       20480  0
snd_sof_utils          16384  1 snd_sof
[... 100 lines cut ...]
x_tables               65536  1 ip_tables
btrfs                2293760  1
blake2b_generic        24576  0
libcrc32c              12288  1 btrfs
crc32c_generic         12288  0
xor                    20480  1 btrfs
raid6_pq              122880  1 btrfs
nvme                   69632  3
nvme_core             262144  4 nvme
spi_intel_pci          12288  0
spi_intel              36864  1 spi_intel_pci
sha256_ssse3           40960  0
nvme_auth              24576  1 nvme_core
xhci_pci               24576  0
xhci_pci_renesas       24576  1 xhci_pci
vmd                    28672  0
nvidia_drm            131072  64
nvidia_uvm           6721536  2
nvidia_modeset       1650688  27 nvidia_drm
video                  81920  3 asus_wmi,asus_nb_wmi,nvidia_modeset
wmi                    36864  3 video,asus_wmi,wmi_bmof
nvidia              72572928  464 nvidia_uvm,nvidia_modeset
crc32c_intel           16384  4

```

FeRChImoNdE, 2024-09-20:
> Also, I'm guessing that's a DDR5 system, right?
> 
> And can you also paste the output from `sudo lsmod`?

I believe that spd5118 module is necessary for DDR5 that is not available as default (need to build from sources)

jonasmalacofilho, 2024-09-20:
spd5118 is the hwmon driver, but the issue here is with the eeprom driver for those memory modules.

`decode-dimms` not being able to report anything is also unexpected to me: do you know if it worked with kernel 6.10, or does it simply not support DDR5?

jonasmalacofilho, 2024-09-20:
@FeRChImoNdE 

Can you execute:

```
cat /sys/bus/i2c/devices/i2c-6/0-005*/name
ls -l /sys/bus/i2c/devices/i2c-6/0-005*/driver
```

FeRChImoNdE, 2024-09-20:
> @FeRChImoNdE
> 
> Can you execute:
> 
> ```
> cat /sys/bus/i2c/devices/i2c-6/0-005*/name
> ls -l /sys/bus/i2c/devices/i2c-6/0-005*/driver
> ```

Had changed 0-005* for 6-005* in both commands (no 0-* folders):

```
⫸ cat /sys/bus/i2c/devices/i2c-6/6-005*/name
spd5118
spd5118
spd5118
spd5118
⫸ ls -l /sys/bus/i2c/devices/i2c-6/6-005*/driver
lrwxrwxrwx 1 root root 0 sep 20 17:15 /sys/bus/i2c/devices/i2c-6/6-0050/driver -> ../../../../../bus/i2c/drivers/spd5118
lrwxrwxrwx 1 root root 0 sep 20 17:15 /sys/bus/i2c/devices/i2c-6/6-0051/driver -> ../../../../../bus/i2c/drivers/spd5118
lrwxrwxrwx 1 root root 0 sep 20 17:15 /sys/bus/i2c/devices/i2c-6/6-0052/driver -> ../../../../../bus/i2c/drivers/spd5118
lrwxrwxrwx 1 root root 0 sep 20 17:15 /sys/bus/i2c/devices/i2c-6/6-0053/driver -> ../../../../../bus/i2c/drivers/spd5118

```

jonasmalacofilho, 2024-09-20:
Thanks, I'll take a look. But you were right that the driver we're dealing with is spd5118.

jonasmalacofilho, 2024-09-20:
The problem might be related to torvalds/linux@79d0df36b541 ([lore](https://lore.kernel.org/all/20240625063459.429953-2-W_Armin@gmx.de/)).

If that's the case, it should affect me too, as soon as 6.11 lands in ArchLinux. So I'll wait for that, as it will make it easier for me to confirm the hypothesis, see if other userspace programs are affected, as well as check if there's anything that can be done to make the `eeprom` attribute readable.

(The commit message suggests that the _custom_ attribute has been removed... but an `eeprom` attribute is still there, so maybe it's now exposed by NVMEM, and we might simply need to learn how to make it readable again).

FeRChImoNdE, 2024-09-21:
Since I don't need it in my system, meanwhile I've disabled SMBus search (and therefore DDR4) in order to run liquidctl with the latest kernel. I'll keep waiting

bezirg, 2024-10-04:
I am running into the same problem since I updated to 6.11.

The problem may be related to ddr5 dimms, which suddenly appeared to my `sensors`: spd5118-i2c-21-53 .

`decode-dimms` fails with:
`No EEPROM found, try loading the eeprom, at24 or ee1004 module`

`liquidctl` fails with:
OSError: [Errno 6] No such device or address

Downgrading to 6.10 fixes the issue.

sgskinner, 2024-10-06:
I've got this same issue on archlinux kernel 6.11.2-arch1-1, and it seems  to have started with the `spd5118` module  loading itself (I did not manually introduce this module, no hardware changes either).

To get back to a functioning state with my CoolerControl install, I blacklisted via:
```
sudo echo "blacklist spd5118" > /etc/modprobe.d/spd5118-blacklist.conf
```

For reference, here's the [linux-hardware probe](https://linux-hardware.org/?probe=51cfa806cd) of the machine in its current working state.

jonasmalacofilho, 2024-10-13:
I've been taking another look at this.

Can you check if the `eeprom` file is readable (both directly or during liquidctl probing) once the spd5118 is unloaded?

drizt, 2024-11-05:
`blacklist spd5118` works for me on Fedora 41 with 

```
$ uname -a
Linux lix 6.11.5-300.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Oct 22 20:11:15 UTC 2024 x86_64 GNU/Linux
```

jonasmalacofilho, 2024-11-15:
The problem has been mitigated in b3913d0.

I originally intended to also check whether exposing an unreadable and therefore useless attribute could be considered a kernel bug, and propose a fix upstream, but it's low priority for me and unlikely to happen any time soon...

So I'm closing this issue as fixed/mitigated.

closed 2024-11-15 by jonasmalacofilho