liquidctl/liquidctl #739: Add Support for NZXT KRAKEN 240

zvdy, 2024-10-06:
### Device type

AIO liquid cooler

### Product page

https://nzxt.com/product/kraken-240

### First-party software

NZXT CAM

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles, voltages, currents or power measurements

### Physical connection

USB Header

### Connection protocol

USB

### Additional information about the device

Bus 001 Device 003: ID 1e71:300e NZXT NZXT Kraken Base
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x300e NZXT Kraken Base
  bcdDevice            2.00
  iManufacturer           1 NZXT Inc.
  iProduct                2 NZXT Kraken Base
  iSerial                 3 [.................]
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     619
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

aleksamagicka, 2024-10-06:
`300e` is [supported in Git](https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/kraken3.py#L587). You'll need to [install liquidctl from Git](https://github.com/liquidctl/liquidctl?tab=readme-ov-file#installing-from-pypi-or-github) for the time being.

zvdy, 2024-10-08:
Came across Issue [#624](https://github.com/liquidctl/liquidctl/issues/624) 

NZXT product nomenclature is somewhat confusing, thanks for the help, got it working now

```
NZXT Kraken 2023
├── Firmware version    2.0.0  
├── LCD Brightness        100  %
└── LCD Orientation         0  °
```

- Fourth-generation NZXT liquid coolers [doc](https://github.com/liquidctl/liquidctl/blob/main/docs/kraken-x3-z3-guide.md)

closed 2024-10-08 by zvdy