liquidctl/liquidctl #748: Unraid / Slackware support

bmartino1, 2024-12-03:
More for expanding audience... with lattest issues with kerneal 6... https://forums.unraid.net/topic/181244-corsair-commander-pro-suposed-to-be-supported-according-to-current-kernel-versions/

Not sure if you ahve a guide or way to run a python script to run install and run liquidctl to access and read data from a Corsair Commander Pro.

OR 
any way to add slackware linux support?
https://forum.linuxfoundation.org/discussion/6201/how-to-build-a-slackware-package

jonasmalacofilho, 2024-12-16:
I can't take the time to read  the linked forum thread right now. Can you describe which specific "issues with kernel 6" you're facing?

---

> run a python script to run install and run liquidctl to access and read data from a Corsair Commander Pro.

Check our README.

> any way to add slackware linux support?

It's more up to distributions or the community around them (maybe yourself) to build distro packages, based on our upstream package.

bmartino1, 2024-12-16:
The readme wasn't clear on how to call the liquidctl nor configure it. Python implementation would be better here... 

Unraid is a mutable OS off a Flash device running in ram. Limetech precompiles a linux 6 kernel version running on top of slackware linux. Your project seems to be a go to for some hardware and data access and controls. had repose for other OS. Thought to ask if a binary was made for slackware linux.

WIth a unraid plugin I can make a python env and build this per teh readme. but there is no information on this past that...
With slackware I would need a simlar precompiled binary to install this applicaiton on unraid directly. and opens this up for use on slackware linux. Making the bianry is out of my whele house.

I have no other in for this issues outside of that forum post and other missing side in the redme on further python implementation.

I would go furtherm but I don't have nor run a fan controler nor need this software myself and was more a side puzzle to assist another client.

Kerneal 6 issues as they dropped support for the hardware driver. wnated confirmation that this would still work if seen in lsusb and if other python things need to be done.

jonasmalacofilho, 2024-12-16:
> The readme wasn't clear on how to call the liquidctl nor configure it. Python implementation would be better here...

I'm not sure what else you feel should be implemented in Python: liquidctl is entirely written in Python, and `pip` is the bog standard way of installing Python packages and libraries...

> Unraid is a mutable OS off a Flash device running in ram. Limetech precompiles a linux 6 kernel version running on top of slackware linux. Your project seems to be a go to for some hardware and data [...]

In Linux, any script with a shebang line can be an executable, and installing liquidctl with a tool like Python will set up an executable `liquidctl` entry point for the CLI in the appropriate location (depending on the user calling `pip` or similar, the use of virtual environments, etc.).

> WIth a unraid plugin I can make a python env and build this per teh readme. but there is no information on this past that... With slackware I would need a simlar precompiled binary to install this [...]

In that case (contract work), I'll be happy to assist you or the client directly for a fee. Please get in touch at your or the client's convenience. My business email address is listed on my GitHub profile page.

> Kerneal 6 issues as they dropped support for the hardware driver. wnated confirmation that this would still work if seen in lsusb and if other python things need to be done.

Again, which driver was dropped, exactly?

closed 2024-12-16 by jonasmalacofilho

jonasmalacofilho, 2024-12-17:
Even though I really didn't have the time to do it, I checked the unraid forum thread.

- kernel 5.9 **added** a HWMON driver for the Commander Pro;
- which for some reason is not (yet) **enabled** in the Unraid kernel configuration.

(Plus the whole discussion doesn't seem to involve contract work at all; it looks like just another fellow forum member asking for help).

---

With this in mind, your fellow forum member has a few options:

- liquidctl can control the Commander Pro *with or without* a compatible HWMON driver (provided HIDRAW or LibUSB access to the device is possible);
- they could also build the Commander Pro HWMON kernel driver as a module separately and install/insert it in the kernel;
- finally, they could ask the Unraid people to enable the Commander Pro (and maybe other) HWMON drivers in their kernel configuration.